Pi4J/pi4j #12: Build profile `javadoc` fails to compile.

savageautomate, 2020-06-11:
When attempting to Maven compile the project with build profile `javadoc`, the build fails.

Build command:
```
 mvn clean install -Pjavadoc
```

Error message:
```
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary for Pi4J :: Parent POM 2.0-SNAPSHOT:
[INFO] 
[INFO] Pi4J :: Parent POM ................................. SUCCESS [  1.493 s]
[INFO] Pi4J :: DOCKER   :: Docker Parent POM .............. SUCCESS [  0.292 s]
[INFO] Pi4J :: LIBRARY  :: Libraries Parent POM ........... SUCCESS [  0.032 s]
[INFO] Pi4J :: LIBRARY  :: JNI Wrapper for PIGPIO Library . FAILURE [  5.232 s]
[INFO] Pi4J :: LIBRARY  :: Java Library (CORE) ............ SKIPPED
[INFO] Pi4J :: PLUGIN   :: Plugins Parent POM ............. SKIPPED
[INFO] Pi4J :: PLUGIN   :: Mock Platform & Providers ...... SKIPPED
[INFO] Pi4J :: PLUGIN   :: PIGPIO I/O Providers ........... SKIPPED
[INFO] Pi4J :: PLUGIN   :: RaspberryPi Platform & Providers SKIPPED
[INFO] Pi4J :: TESTING  :: Unit/Integration Tests ......... SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  8.986 s
[INFO] Finished at: 2020-06-11T15:30:35-04:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-javadoc-plugin:3.2.0:jar (attach-javadocs) on project pi4j-library-pigpio: MavenReportException: Error while generating Javadoc: 
[ERROR] Exit code: 1 - /Users/roberts/dev/pi4j-v2/libraries/pi4j-library-pigpio/src/main/java/module-info.java:32: error: module not found: org.slf4j
[ERROR]     requires org.slf4j;
[ERROR]                 ^
[ERROR] 
[ERROR] Command line was: /Library/Java/JavaVirtualMachines/jdk-11.0.4.jdk/Contents/Home/bin/javadoc @options @packages @argfile
[ERROR] 
[ERROR] Refer to the generated Javadoc files in '/Users/roberts/dev/pi4j-v2/libraries/pi4j-library-pigpio/target/apidocs' dir.
[ERROR] 

```

This Javadoc build profile is part of the release build steps to generate JavaDoc for all project JARs and attach them as artifacts to deploy to Maven repositories.  

**This issue will need to get resolved prior to a release build.**

PS:  Apart from this initial error,  there are also several JavaDoc specific code issues that will also need to get addressed.

FDelporte, 2020-06-11:
Perfect one for me, will fix

savageautomate, 2020-06-11:
OK, feel free to dig in.  I spent a few minutes trying to figure it out, but was not successful.  I found several issues reported against both the JDK and the Maven Javadoc plugin, but in both cases I have updated/tests the latest versions but the error did not go away. I was going to test it on my main development computer where I originally created these projects just to see if it worked there.

FDelporte, 2020-06-11:
checking with "sdk use java 11.0.7.j9-adpt", already fixed some, ongoing... ;-)

savageautomate, 2020-06-11:
@FDelporte 

Thanks for fixing the JavaDoc errors.

**NOTE:**  Users must use the OpenJDK rather than the OracleJDK to get past the JavaDoc build error.
See: https://bugs.openjdk.java.net/browse/JDK-8208269

- Tested using Adopt OpenJDK 11.0.7 - JAVADOC COMPILED SUCCESSFULLY
- Tested using OracleJDK 11 & 14 - JAVADOC COMPILE FAILED

closed 2020-06-11 by savageautomate