Pi4J/pi4j #13: Crete BUILD.INSTRUCTIONS

savageautomate, 2020-06-11:
Add a `BUILD.INSTRUCTIONS` file to the project to document how users and contributors can build the project sources.

Make sure to include information about the following:
 - Build Profiles (native, javadoc, docker, etc)
 - Native Builds (32-bit, 64-bit, Cross-compiler, Docker-compiler, etc)
 - Release Builds
 - Custom Maven `settings.xml` profile settings for Pi4J developers/contributrors
 - Environment variables
 - Requirements (JDK Version, Maven version, Docker, etc)

savageautomate, 2020-06-11:
Include build notes about JavaDoc requiring OpenJDK 11.0.7 (or later)
See https://github.com/Pi4J/pi4j-v2/issues/12

FDelporte, 2020-06-18:
I started creating pages on the V2-site with info about the internal code, building, etc. 
Wouldn't it be better to include this info on https://v2.pi4j.com/core-code-internals/cross-compiling and others and let BUILD.INSTRUCTIONS refer to those pages?

eitch, 2020-06-18:
+1 from me =)

savageautomate, 2020-06-18:
I'm good with that.

savageautomate, 2020-06-30:
I have updated the project's README to include a limited section on _Build Instructions_ and includes a link to the new build instructions page creates at: https://v2.pi4j.com/build

This documentation will be an ongoing effort to maintain up to date/correct info and argument additional details/nuance.  However, the essential information is captured and thus closing this ticket.

closed 2020-06-30 by savageautomate