Pi4J/pi4j #15: Use of signal handlers needs to be improved

FDelporte, 2020-06-12:
As reported by Jim Darby:
_If you're running as root (never a good idea) and using pigpio directly then pigpio steals all the signal handlers. The JVM doesn't like this as it uses them in running the JVM so it ends up with the program crashing. I'm working on this by getting pi4j-v2 to put back the signal handlers that the JVM uses._

hackerjimbo, 2020-06-21:
As a first step I printed out the signal handled at the very start of Java_com_pi4j_library_pigpio_internal_PIGPIO_gpioInitialise to see what it was life before gpioInitialise is called. It looks like quite a lot.

Then I looked at the code in pigpio and found something very interesting. There is actually a compile-time #define, EMBEDDED_IN_VM, which explicitly turns off pigpio's use of signals!

Furthermore, there is a run-time flag as well. The code looks like:

#ifndef EMBEDDED_IN_VM
   if (!(gpioCfg.internals & PI_CFG_NOSIGHANDLER))
      sigSetHandler();
#endif

So, we can avoid the entire problem and use the OS-supplied pigpio by setting that flag!

This looks like the footprints of someone being here before. I'm going to guess that it's Python. Now all we have to do is to figure out how to set that flag.

hackerjimbo, 2020-07-12:
Brace yourselves. Here's the fix (I hope). In the native C bindings we have the function `Java_com_pi4j_library_pigpio_internal_PIGPIO_gpioInitialise`. This contains the single line ` return gpioInitialise();`.

Immediately before that insert `gpioCfgSetInternals (gpioCfgGetInternals () | PI_CFG_NOSIGHANDLER);`

I can't actually test this as I can't build an armv6 native library right now, but I strongly believe it should work.

savageautomate, 2020-07-13:
Here is the compiled JAR with the [changes you suggested](https://github.com/Pi4J/pi4j-v2/commit/b3e40b7370d14d3149a98a3c15c8ee47257b813d) if you would like to test it out:
 - https://www.dropbox.com/s/8cm176mbknk2s0m/pi4j-library-pigpio-2.0-SNAPSHOT.jar

https://github.com/Pi4J/pi4j-v2/blob/b3e40b7370d14d3149a98a3c15c8ee47257b813d/libraries/pi4j-library-pigpio/src/main/native/com_pi4j_library_pigpio_internal_PIGPIO.c#L155-L166

savageautomate, 2020-07-13:
@hackerjimbo ,

While applying the fix you suggested, I also came across these lines of code:

https://github.com/Pi4J/pi4j-v2/blob/5eb404d33cad9f276bd5095bf7821e9ac26aaf53/libraries/pi4j-library-pigpio/src/main/java/com/pi4j/library/pigpio/impl/PiGpioNativeImpl.java#L90-L128

It looks like I was originally registering signal handlers with PiGPIO to try and capture exit/terminate signals and then terminate the PiGPIO library as well as terminate the JVM application with `System.exit(signum);`.    This must be here because I was failing to get the Java application to exit cleanly when PiGPIO  was stealing/handling the signals.  I suspect this code could now be removed.  What do you think?

savageautomate, 2020-07-13:
**REOPENING** only to address the question above (https://github.com/Pi4J/pi4j-v2/issues/15#issuecomment-657541714) about getting rid of Java-side signal handlers registered with PiGPIO.  I suspect these can be removed now?  Especially if the change @hackerjimbo suggested completely disables signals in PiGPIO.

hackerjimbo, 2020-07-13:
@savageautomate Those signal handlers can now never be called as we no longer intercept the signals with PiGPIO. Hopefully :-)

savageautomate, 2020-07-13:
Removed these signal callbacks and deployed update SNAPSHOT JARs.  
CLOSED.

closed 2020-07-13 by savageautomate