Pi4J/pi4j #18: Extend the Context with an ExecutorService

FDelporte, 2020-06-12:
Different approaches should be possible
* User can inject his own executor service into the Context 
* The Context needs to create its own ExecutorService if none is provided

This ExecutorService is needed for
* DigitalOut Blink
* DigitalOut Pulse

https://www.baeldung.com/java-executor-service-tutorial

tcfurrer, 2021-12-12:
Multi-threaded capability and implementations of the asynchronous APIs would be a terrific step forward!

I assume this would require some non-trivial changes to provide the necessary thread safety, correct?

FDelporte, 2021-12-13:
Indeed @tcfurrer will have some impact. I have been looking into this in the past.

It would all start with defining the ExecutorService in the context
https://github.com/Pi4J/pi4j-v2/blob/develop/pi4j-core/src/main/java/com/pi4j/context/Context.java

And could be used to handle for instance thread actions like pulsing a LED
https://github.com/Pi4J/pi4j-v2/blob/develop/pi4j-core/src/main/java/com/pi4j/io/gpio/digital/DigitalOutputBase.java#L102

These are only a few first steps and to be further investigated where and how such multi-threading should be applied.

Here would be an ideal place to discuss this functionality.