Pi4J/pi4j #21: libpi4j-pigpio.so: to include or not to include, that's the question...

FDelporte, 2020-06-15:
At this moment libpi4j-pigpio.so is packaged inside a jar 

Pro
* will be available when needed
* no separate installation needed

Contra
* when building Pi4J on another device, cross-compiling needed
* problem 32 versus 64 bit?

To be decided
* Include - not include
* If not included can we check at program startup if necesarry file can be found?
* If not found, can we show clear message about the missing file?
* If not found, can we provide easy-to-use install script?

eitch, 2020-06-16:
In my opinion cross-compilation is the way to go. The 32-bit and 64-bit problem is solved by using a special loader for the library which detects the architecture, as i have done in develop/1.4 Thus i don't see much contra for still keeping it in a jar, as this makes using it in Java quite trivial.

Contrary to all this, i don't know what the library is for, and would argue that maybe its functionality should be in the main shared library? Or is this the name for the main shared library?

savageautomate, 2020-06-30:
As of this commit:  https://github.com/Pi4J/pi4j-v2/commit/a7c29d098920a11d99217fe2b4a24c596709e9cf

The new native library build logic now **requires** cross-compiling.  We now include building for both 32-bit and 64-bit version of the libraries and a cross-compiling environment is necessary to generate both artifacts.  The artifacts are embedded inside the JAR and extracted at runtime into a temporary folder.  In the rare cases where this is not acceptable, V2 supports a system property (`pi4j.library.path`) to override the embedded behavior and allows the user to specify a location to resolve the native library.  (See: https://github.com/Pi4J/pi4j-v2/blob/master/libraries/pi4j-library-pigpio/src/main/java/com/pi4j/library/pigpio/util/NativeLibraryLoader.java#L72)
 
A user can use either their own host to cross compile the native libraries for `armhf` and `aarch64` or use our new P4J Docker Builder images and perform the build task inside a docker container. 
            
### CROSS-COMPILER

 Cross-compiling is the default compiler option defined in the Maven build.  _(Unless running on Windows or MacOS, then an OS-targeted profile will default to Docker-compiler)_.  The host must be an Intel/AMD or ARM 64-bit Linux-based machine to perform the cross-compile.  

The cross-compiler build will use GCC cross-compiling toolchains installed on your host with a cross compiler environment for 64-bit (aarch64) and 32-bit (armhf) platforms.  The host can be a RaspberryPi 3B or 4B that is running Pi OS 64-bit. 
 _(Unable to cross-compile for 64-bit libraries on 32-bit hosts.)_

Here are the options you can use to force the build to use the native CROSS-COMPILER if needed.

* You can define  the following property in a custom active profile in your `~/.m2/settings.xml` file:
```
<pi4j.native.compiler>CROSS-COMPILER</pi4j.native.compiler>
```
* Or you can define the preferred compiler property using the `-D` flag on the Maven command:
```
mvn clean install -Pnative -Dpi4j.native.compiler=CROSS-COMPILER
```
* Or you can use the associated profiles using the `-P` flag on the Maven command:
```
mvn clean install -Pnative,cross-compile
```

### DOCKER-COMPILER

Docker-compiling is the default compiler option when running on Windows or MacOS host.  The host must be an Intel/AMD or ARM 64-bit Linux-based machine with [Docker](https://www.docker.com/) or[ Docker Desktop](https://www.docker.com/products/docker-desktop) installed to perform the native build using the Pi4J Docker Builder Images.   

This build option will perform all native cross-compiling inside a Docker container.  This is the best option when compiling on MacOS or Windows host.  This option is also suitable for a Linux-based system where you would prefer not to install ARM-based cross-compilers and other build environment requirements like Java/JDK and Maven. 

Here are the options you can use to force the build to use the native DOCKER-COMPILER if needed.

* You can define  the following property in a custom active profile in your `~/.m2/settings.xml` file:
```
<pi4j.native.compiler>DOCKER-COMPILER</pi4j.native.compiler>
```
* Or you can define the preferred compiler property using the `-D` flag on the Maven command:
```
mvn clean install -Pnative -Dpi4j.native.compiler=DOCKER-COMPILER
```
* Or you can use the associated profiles using the `-P` flag on the Maven command:
```
mvn clean install -Pnative,docker
```

savageautomate, 2020-06-30:
To be decided

* _Include - not include_
**Included to keep most use cases simple -- we also provide a workaround for the few cases where this is not working --- we may need to explicitly document this in the V2 docs.**

* _If not included can we check at program startup if necessary file can be found?_
**This should already be the case.  https://github.com/Pi4J/pi4j-v2/blob/master/libraries/pi4j-library-pigpio/src/main/java/com/pi4j/library/pigpio/util/NativeLibraryLoader.java#L115**

* If not found, can we show clear message about the missing file?
**I have updated here to provide more/clear messaging.  https://github.com/Pi4J/pi4j-v2/blob/master/libraries/pi4j-library-pigpio/src/main/java/com/pi4j/library/pigpio/util/NativeLibraryLoader.java#L115**

* If not found, can we provide easy-to-use install script?
**I think documenting this should suffice.  If we get into managing the library installations we will also have to ensure the installed versions are in sync with the accompanying JAR file.  At least when embedded the JAR is portable and it always carries the correct in-sync version of the native library.**

savageautomate, 2020-06-30:
> Contrary to all this, i don't know what the library is for, and would argue that maybe it's functionality should be in the main shared library? Or is this the name for the main shared library?

In V2 the Pi4J core is designed to be independent of implementation.  So in the case of V2, the current native library exists and accompanies the Pi4J-PIGPIO plugin JAR.   This native library exposes the PIGPIO functionality via JNI so Pi4J and Java can communicate with PIGPIO.  There could be other cases for additional native libraries depending on what is needed for implementation.  For example, if we also creates (UNLIKELY) a WiringPi I/O plugin for Pi4J V2, then it too would need a native library to work with the underlying WiringPi lib.

savageautomate, 2020-06-30:
In the event of a failure to extract and load the embedded native library from the JAR file at runtime, a user can define system property `pi4j.library.path`

If `pi4j.library.path-system=system`, then the default system defined library path for the JVM will be used to resolve the appropriate `libpi4j-xxx.so` library.

Alternatively `pi4j.library.path-system` can define an explicit absolute file system path, and Pi4J will use that user defined library path to resolve the appropriate `libpi4j-xxx.so` library.

**NOTE:**
Failure to load this native library can occur on systems that do not support a writable temporary directory.   Users attempting to use Pi4J under Android may also experience this error.

closed 2020-06-30 by savageautomate

FDelporte, 2020-06-30:
Great work @savageautomate !