Pi4J/pi4j #27: I2CRegister.write ignores length value

FDelporte, 2020-07-07:
As reported by @hackerjimbo 

There is an issue with I2CRegister.write (data, offset, length). It seems to ignore the length and fire the whole amount of data in one go.

Starting points for research
* https://github.com/Pi4J/pi4j-v2/blob/master/pi4j-core/src/main/java/com/pi4j/io/i2c/impl/DefaultI2CRegister.java#L84
* https://github.com/Pi4J/pi4j-v2/blob/75e1909006ef8fae8ad02d44f48096b368c634fd/pi4j-core/src/main/java/com/pi4j/io/i2c/I2CRegisterDataWriter.java#L111

savageautomate, 2020-07-07:
If the error is not in the Pi4J core library, then its possible its either in the PiGpio Provider Plugin or the PiGpio JNI (wrapper) Library code.

- Are you using the native or remote (TCP socket) implementation of PiGpio?

 ---

See if you can trace the variables here and make sure all the input parameters are still correct and `length` has not been overwritten/modified.

### Pi4J PiGpio Provider Plugin
https://github.com/Pi4J/pi4j-v2/blob/a54d626c7a7d3750e31a63210c2abd1c983c04df/plugins/pi4j-plugin-pigpio/src/main/java/com/pi4j/plugin/pigpio/provider/i2c/PiGpioI2C.java#L117

```
    public int write(byte[] data, int offset, int length) throws IOException {
        Objects.checkFromIndexSize(offset, length, data.length);
        piGpio.i2cWriteDevice(this.handle, data, offset, length);
        return length;
    }
```

### Pi4J PiGpio JNI (Wrapper) Library
https://github.com/Pi4J/pi4j-v2/blob/a54d626c7a7d3750e31a63210c2abd1c983c04df/libraries/pi4j-library-pigpio/src/main/java/com/pi4j/library/pigpio/PiGpio_I2C.java#L1428

Then .. depending on if using native PiGpio or remote via TCP  ...

**_TCP/SOCKET IMPL:_**
https://github.com/Pi4J/pi4j-v2/blob/a54d626c7a7d3750e31a63210c2abd1c983c04df/libraries/pi4j-library-pigpio/src/main/java/com/pi4j/library/pigpio/impl/PiGpioSocketImpl.java#L1117
```
    public int i2cWriteDevice(int handle, byte[] data, int offset, int length) throws IOException {
        logger.trace("[I2C::WRITE] -> [{}]; I2C Raw Write [{} bytes]", handle, data.length);
        validateReady();
        validateHandle(handle);
        PiGpioPacket tx = new PiGpioPacket(I2CWD, handle).data(data, offset, length);
        PiGpioPacket rx = sendPacket(tx);
        logger.trace("[I2C::WRITE] <- HANDLE={}; SUCCESS={}", handle, rx.success());
        validateResult(rx, false);
        return rx.result();
    }
```

**_NATIVE (JNI WRAPPER):_**
https://github.com/Pi4J/pi4j-v2/blob/a54d626c7a7d3750e31a63210c2abd1c983c04df/libraries/pi4j-library-pigpio/src/main/java/com/pi4j/library/pigpio/impl/PiGpioNativeImpl.java#L1170
```
    public int i2cWriteDevice(int handle, byte[] data, int offset, int length) throws IOException {
        logger.trace("[I2C::WRITE] -> [{}]; I2C Raw Write [{} bytes]", handle, data.length);
        validateReady();
        validateHandle(handle);
        Objects.checkFromIndexSize(offset, length, data.length);
        // write data array to I2C device
        int result = PIGPIO.i2cWriteDevice(handle, data, offset, length);
        logger.trace("[I2C::WRITE] <- HANDLE={}; SUCCESS={}", handle, (result>=0));
        validateResult(result, false);
        return result;
    }
```

**_NATIVE C IMPL_**
https://github.com/Pi4J/pi4j-v2/blob/a54d626c7a7d3750e31a63210c2abd1c983c04df/libraries/pi4j-library-pigpio/src/main/native/com_pi4j_library_pigpio_internal_PIGPIO.c#L1611
```
JNIEXPORT jint JNICALL Java_com_pi4j_library_pigpio_internal_PIGPIO_i2cWriteDevice
  (JNIEnv *env, jclass class, jint handle, jbyteArray data, jint offset, jint count)
{
	// obtain a pointer to the elements of the array and pin the memory
    jbyte *buffer = (*env)->GetByteArrayElements(env, data, 0);

    // get the maximum size of the Java data array; subtract any offset value
    jsize max_length = (*env)->GetArrayLength(env, data) - offset;

    // bounds check to make sure byte count does not exceed max array length (minus offset)
    int length = (count > max_length) ? max_length : count;

    // create a new buffer pointer using the offset
    jbyte *offsetBuffer = buffer + offset;

    // perform the actual I2C write operation into the native buffer array
    jint result = i2cWriteDevice((unsigned)handle, (char *)offsetBuffer, (unsigned)length);

	// unpin the reserved memory for 'data'; abort preserving any changes back to the Java array
	(*env)->ReleaseByteArrayElements(env, data, buffer, JNI_ABORT);

    // return the result
	return result;
}
```

hackerjimbo, 2020-07-07:
It's in the native library rather than running over TCP. Looks like the JNI wrapper is the place to enable tracing and see what happens. I'm not that familiar with how to set the debugging though. If I recall correctly you need some sort of library (slf4j-simple rings a bell) and to set a Java property. Could someone remind me of the details?

savageautomate, 2020-07-07:
Well it looks like the instrumented trace info is not really printing out the actual length and offset so you may have to add you own messages to print out the `length` and `offset` values.

Yes, include slf4j-simple JAR in your dependencies and set this system property to print log/debug/trace message to the console.
```java
        // configure default lolling level, accept a log level as the fist program argument
        System.setProperty("org.slf4j.simpleLogger.defaultLogLevel", "TRACE");
```

**_Example:_**

https://github.com/Pi4J/pi4j-example-telegraph/blob/master/pom.xml#L32-L37

https://github.com/Pi4J/pi4j-example-telegraph/blob/master/src/main/java/com/pi4j/demo/telegraph/Telegraph.java#L60-L61

hackerjimbo, 2020-07-07:
Well this is exciting! I had to download and then rebuild all of pi4j-v2 because of updates in the system (not that I blame anyone, it's q work in progress). However, once it was all rebuilt I couldn't get the result to run.

Interesting issue and probably worth raising elsewhere. I was building on a 64-bit Pi 4 running the 64-bit OS. It _did_ produce the 32-bit binaries for the Pi zero I'm running on _but_ it wasn't compiled for the v6 architecture so it didn't work! This is probably a bug that needs fixing. In the meantime are there any workarounds anyone can suggest?

savageautomate, 2020-07-07:
> Interesting issue and probably worth raising elsewhere. I was building on a 64-bit Pi 4 running the 64-bit OS. It _did_ produce the 32-bit binaries for the Pi zero I'm running on _but_ it wasn't [...]

@hackerjimbo ,

I was afraid that might happen with the new cross-compiler stuff I was working on -- I'll have to revisit the new compiler stuff and get it building for ARMv6 as well.

In the meantime, try to clone this older [commit](https://github.com/Pi4J/pi4j-v2/tree/ec2815015bf269b2365d90be000f67024fe9013a) (using the previous native compiler stuff) which should build for all 32-bit RaspberryPis :  https://github.com/Pi4J/pi4j-v2/tree/ec2815015bf269b2365d90be000f67024fe9013a

Thanks, Robert

hackerjimbo, 2020-07-07:
Well… It loads the cross-compiler on the 64-bit machine and produces a 32-bit library. So that's not too bad. All I think it needs is he right compiler flags and we're sorted. If I had to guess, and wit a bit of poking around, I'd support setting -march=armv6+fp -mfloat-abi=hard -mfpu=vfp will work just fine as it's GCC's default options on the Pi.

A little more poking around reveals that the default for the cross-compiler on the 64-bit Raspberry OS is  -mfloat-abi=hard -mfpu=vfpv3-d16 -mthumb -march=armv7-a+fp which explains why it doesn't work (for so many reasons).

My knowledge of maven is amazingly poor. The same applies to the build scripts for pi4j-v2's native libraries. However, using -Pnative I can at least get it to build the 32-bit libraries (if wrongly). All I need is to know the tweaks to add those compiler flags in (-march=armv6+fp -mfloat-abi=hard -mfpu=vfp). The probably need to be in the shell scripts that build the libraries.

I think having the 64-bit and 32-bit (as just v6) is a good mix. I see little value in adding other 32-bit architectures as almost all of the rest of the 32-bit OS is compiled for v6.

savageautomate, 2020-07-07:
@hackerjimbo 

I've read that compiler flags alone are not enough.  
Granted I have not tested it myself, but I suspect a custom compiler toolchain is needed.  

See:
- https://github.com/Pi4J/pi4j-v2/issues/28
- https://github.com/Pro/raspi-toolchain

The [older commit](https://github.com/Pi4J/pi4j-v2/tree/ec2815015bf269b2365d90be000f67024fe9013a) that I referenced will build using an older Docker cross-compiler image that I created that uses the [RaspberryPi Tools](https://github.com/raspberrypi/tools) GCC v4 cross compiler toolchain.   From this older commit, you can build directly on a 32-bit RaspberryPi or build with the Docker image and it **should** produce ARMv6 compatible binaries.

As for the latest build logic in the `master` branch, this all uses newer builder docker images that I recently created and those are not using the custom compiler toolchains.

---

If you want to experiment with the compiler flags and try out  `march=armv6+fp -mfloat-abi=hard -mfpu=vfp` I believe you can simply ... 

Based on this line the Makefile:

https://github.com/Pi4J/pi4j-v2/blob/75e1909006ef8fae8ad02d44f48096b368c634fd/libraries/pi4j-library-pigpio/src/main/native/Makefile#L29

I think you can export your own `CARGS` with whatever custom arguments you want to include.

The best place to add that export is here in the `build.sh` script where the 32-bit build is located:
https://github.com/Pi4J/pi4j-v2/blob/75e1909006ef8fae8ad02d44f48096b368c634fd/libraries/pi4j-library-pigpio/src/main/native/build.sh#L102-L109

hackerjimbo, 2020-07-08:
Well, I've totally failed to get the libraries built for the v6 architecture. On the plus side that's _not this bug_!

It turns out that the bug is _not_ about getting the length wrong, ironically it's about _checking_ the length. In the i2cWriteI2CBlockData method of PiGpioNativeImpl then length check it wrong! Line 1134 is:

`validateI2cBlockLength(data.length);`

when it should be:

`validateI2cBlockLength(length);`

so that means we have a fix for _this_ bug. Now all we need to do is to get the system working again on the v6 architecture!

FDelporte, 2020-07-08:
@hackerjimbo what JDK or you using on v6 architecture? I've not been successful on that side... 
@savageautomate mentioned this one before but didn't try it out yet: https://www.azul.com/downloads/zulu-community/?version=java-11-lts&os=linux&architecture=arm-32-bit-hf&package=jdk

hackerjimbo, 2020-07-08:
@FDelporte that's _exactly_ the one I'm using. It seems to work very well. It appears to probe what the hardware can actually do at run time and then compile for that. In fact, part of its probing is what's causing issues with pigpio's stealing of signals (another bug I'm close to fixing).

savageautomate, 2020-07-08:
@hackerjimbo,  I created a pull request (https://github.com/Pi4J/pi4j-v2/pull/29) for you to review and test out that includes the fixes you mentioned.

savageautomate, 2020-07-10:
@hackerjimbo 

I have sent you a JAR to test with that should be compatible with ARMv6 and include the fixes from the commits in branch: https://github.com/Pi4J/pi4j-v2/tree/issue/%2327

Let me know if the issue is fully resolved of if there are any lingering bugs.

Thanks, Robert

savageautomate, 2020-07-10:
Received confirmation from @hackerjimbo  the fix is working as expected.
Merged pull request into master.
Issue closed.

closed 2020-07-10 by savageautomate