Pi4J/pi4j #31: No hardware PWM's on all GPIO

DanielMartensson, 2020-07-10:
Hi!

Just started with Pi4J 2.0 and I got an error.

```
2020-07-10 15:44:02 gpioHardwarePWM: bad gpio for PWM (3)
[http-nio-8080-exec-5] WARN com.pi4j.library.pigpio.impl.PiGpioNativeImpl - PIGPIO ERROR: PI_NOT_HPWM_GPIO; GPIO has no hardware PWM
```

And the Java code:

  ```
     // This will turn on/off the FQP30N06L MOSFET
	private Pwm createDigitalPWMOutput(int pinOutput, Context pi4j) {
        try {
            // use try-with-resources to auto-close I2C when complete
        	PwmConfig config = Pwm.newConfigBuilder(pi4j)
                    .id("my-pwm-pin")
                    .name("My Test PWM Pin")
                    .address(pinOutput)
                    .pwmType(PwmType.HARDWARE)
                    .frequency(pwmFrequency)   // optionally pre-configure the desired frequency to 1KHz
                    .dutyCycle(0)     // optionally pre-configure the desired duty-cycle (50%)
                    .shutdown(0)       // optionally pre-configure a shutdown duty-cycle value (on terminate)
                    .initial(0)     // optionally pre-configure an initial duty-cycle value (on startup)
                    .build();
        	Pwm pwm = pi4j.providers().get(PiGpioPwmProvider.class).create(config);
        	pwm.on();
			return pwm;
		} catch (Exception e) {
			e.printStackTrace();
		}
		return null;
	}
```

I call that function with:

```
Context pi4j = Pi4J.newAutoContext();
Pwm pwm0 = createDigitalPWMOutput(3, pi4j);
```

savageautomate, 2020-07-10:
I have used hardware PWM in a live demo that I did for CodeOne 2019.  So it's at least been tested before to some degree.
Have you installed the latest version of PIGPIO from their site or are you using the version that comes with Raspbian?

If you have not already, I would suggest getting and installing he latest version:
http://abyz.me.uk/rpi/pigpio/download.html

savageautomate, 2020-07-10:
@DanielMartensson 

It looks like I used different PWM channels in my demo project:
https://github.com/Pi4J/pi4j-example-telegraph/blob/c07f4a92f50f60ecdc2229621b8cc57445e67330/src/main/java/com/pi4j/demo/telegraph/Telegraph.java#L48-L49

If it possible for you to try pin 18 or 19?

I wonder if there is a mode not getting applied to the requested PWM GPIO pin.

DanielMartensson, 2020-07-10:
> @DanielMartensson
> 
> It looks like I used different PWM channels in my demo project:
> https://github.com/Pi4J/pi4j-example-telegraph/blob/c07f4a92f50f60ecdc2229621b8cc57445e67330/src/main/java/com/pi4j/demo/telegraph/Telegraph.java#L48-L49
> 
> If it possible for you to try pin 18 or 19?
> 
> I wonder if there is a mode not getting applied to the requested PWM GPIO pin.
> 
> This looks suspicious:
> 
> https://github.com/Pi4J/pi4j-v2/blob/5edf5a8a9d65f1dce1ce38397b14835c18030d40/plugins/pi4j-plugin-pigpio/src/main/java/com/pi4j/plugin/pigpio/provider/pwm/PiGpioPwmHardware.java#L77-L99

I can try 4, 5, 6 and 13. That's clean GPIO's. No I2C or SPI.

DanielMartensson, 2020-07-10:
> I have used hardware PWM in a live demo that I did for CodeOne 2019. So it's at least been tested before to some degree.
> Have you installed the latest version of PIGPIO from their site or are you [...]

I have

```
pi@raspberrypi:~/Documents $ pigpiod -version
71
pi@raspberrypi:~/Documents $ 
```

DanielMartensson, 2020-07-10:
I post the whole error log and also the Java code:

Java code:
```

import org.springframework.beans.factory.annotation.Value;
import org.springframework.context.annotation.PropertySource;

import com.pi4j.Pi4J;
import com.pi4j.context.Context;
import com.pi4j.exception.Pi4JException;
import com.pi4j.io.gpio.digital.DigitalInput;
import com.pi4j.io.gpio.digital.DigitalInputConfig;
import com.pi4j.io.gpio.digital.DigitalInputProvider;
import com.pi4j.io.gpio.digital.PullResistance;
import com.pi4j.io.pwm.Pwm;
import com.pi4j.io.pwm.PwmConfig;
import com.pi4j.io.pwm.PwmType;
import com.pi4j.plugin.pigpio.provider.pwm.PiGpioPwmProvider;

import lombok.Getter;

@PropertySource("classpath:application.properties")
@Getter
public class IO {
	
	// Frequency for the PWM 
	@Value("${pi4j.pwmFrequency}")
	private int pwmFrequency;
	
	// PWM outputs
	private Pwm pwm0;
	private Pwm pwm1;
	private Pwm pwm2;
	private Pwm pwm3;

	// Digital inputs
	private DigitalInput pulseOn;
	private DigitalInput stopSignalOn;
	
	// ADS1115 16-Bit ADC
	private ADS1115_ADS1015 ads;
	
	public IO() {
		try {
			Context pi4j = Pi4J.newAutoContext();
			pwm0 = createDigitalPWMOutput(4, pi4j);
			pwm1 = createDigitalPWMOutput(5, pi4j);
			pwm2 = createDigitalPWMOutput(6, pi4j);
			pwm3 = createDigitalPWMOutput(13, pi4j);
			pulseOn = createDigitalInput(23, pi4j);
			stopSignalOn = createDigitalInput(24, pi4j);
			ads = new ADS1115_ADS1015(pi4j, 1, ADS1115_ADS1015.ADS_ADDR_GND); // I2C bus = 1
			ads.useADS1115();
			
		} catch (Pi4JException e) {
			// TODO Auto-generated catch block
			e.printStackTrace();
		}
	}
	

	// This will read the pulse input and the stop signal
	private DigitalInput createDigitalInput(int pinInput, Context pi4j) {
		try {
			DigitalInputConfig config = DigitalInput.newConfigBuilder(pi4j)
					.id("Digial input")
					.name("Digital input")
					.address(pinInput)
					.pull(PullResistance.PULL_DOWN)
[... 1163 lines cut ...]
	at org.apache.catalina.core.StandardWrapperValve.invoke(StandardWrapperValve.java:202)
	at org.apache.catalina.core.StandardContextValve.invoke(StandardContextValve.java:96)
	at org.apache.catalina.authenticator.AuthenticatorBase.invoke(AuthenticatorBase.java:526)
	at org.apache.catalina.core.StandardHostValve.invoke(StandardHostValve.java:139)
	at org.apache.catalina.valves.ErrorReportValve.invoke(ErrorReportValve.java:92)
	at org.apache.catalina.core.StandardEngineValve.invoke(StandardEngineValve.java:74)
	at org.apache.catalina.connector.CoyoteAdapter.service(CoyoteAdapter.java:343)
	at org.apache.coyote.http11.Http11Processor.service(Http11Processor.java:408)
	at org.apache.coyote.AbstractProcessorLight.process(AbstractProcessorLight.java:66)
	at org.apache.coyote.AbstractProtocol$ConnectionHandler.process(AbstractProtocol.java:861)
	at org.apache.tomcat.util.net.NioEndpoint$SocketProcessor.doRun(NioEndpoint.java:1579)
	at org.apache.tomcat.util.net.SocketProcessorBase.run(SocketProcessorBase.java:49)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at org.apache.tomcat.util.threads.TaskThread$WrappingRunnable.run(TaskThread.java:61)
	at java.base/java.lang.Thread.run(Thread.java:834)
2020-07-10 16:43:15 sigHandler: Unhandled signal 11, terminating

2020-07-10 16:43:15 sigHandler: Unhandled signal 4, terminating

2020-07-10 16:43:15 sigHandler: Unhandled signal 11, terminating

pi@raspberrypi:~/Documents $ 

```

savageautomate, 2020-07-10:
So reading that PIPGIO supported hardware PWM on pins 0-31 was a bit confusing to me.  
From WiringPi, Pi4J V1 and all the previous work I had done with PWM ... I was only aware of **two** GPIO pins that supported hardware PWM  --- _ore more accurately_ two hardware PWM channels that could be exposed to select GPIO pins via the ALT modes:   `GPIO PINS 12,13,18,19`

| GPIO (BCM#) | PWM Channel | ALT Function | URL |  
| --- | --- | --- | --- |
| 12 | PWM0 | ALT0 | https://pinout.xyz/pinout/pin32_gpio12 |
| 13 | PWM1 | ALT0 | https://pinout.xyz/pinout/pin32_gpio12 |
| 18 | PWM0 | ALT5 | https://pinout.xyz/pinout/pin12_gpio18 |
| 19 | PWM1 | ALT5 | https://pinout.xyz/pinout/pin35_gpio19 |

If you read further in the actual function description on PiGpio:
http://abyz.me.uk/rpi/pigpio/cif.html#gpioHardwarePWM

You will see that it says: "***The GPIO must be one of the following ...***" 

```
The same PWM channel is available on multiple GPIO. 
The latest frequency and dutycycle setting will be used by all GPIO which share a PWM channel.

The GPIO must be one of the following:

12  PWM channel 0  All models but A and B
13  PWM channel 1  All models but A and B
18  PWM channel 0  All models
19  PWM channel 1  All models but A and B

40  PWM channel 0  Compute module only
41  PWM channel 1  Compute module only
45  PWM channel 1  Compute module only
52  PWM channel 0  Compute module only
53  PWM channel 1  Compute module only
```

So I think that the RaspberryPi hardware only supports 2 hardware PWM channel and they can be routed to the GPIO pins listed above --- but in the end you only get 2 hardware PWM channels.

So I'm not sure if the documentation on the PIPGIO site that says: "_hardware timed PWM on all of GPIO 0-31_" ... really just means that if you use the software PWM functions that the library is using some hardware timing elements to emulate PWM in software or if this just means that the library supports the hardware PWM functions on all pins, but its up to the actual hardware as to which pins can accept hardware PWM 

So .. I guess try BCM pin 13.

DanielMartensson, 2020-07-11:
> So reading that PIPGIO supported hardware PWM on pins 0-31 was a bit confusing to me.
> From WiringPi, Pi4J V1 and all the previous work I had done with PWM ... I was only aware of **two** GPIO pins [...]

So the conclusion is that there are 0-31 hardware PWM pins, bit raspberry pi can only use two of them?

I need 4 PWM pins - hardware.

DanielMartensson, 2020-07-11:
Ok! It's seems to work now. But I have another issue you might want to look at. 

I have selected the pins you suggested. It's seems to...not giving any errors. Also I removed the id-label. Not sure if they are needed.
```

public class IO {
	
	// Frequency for the PWM 
	@Value("${pi4j.pwmFrequency}")
	private int pwmFrequency;
	
	// PWM outputs
	private Pwm pwm0;
	private Pwm pwm1;
	private Pwm pwm2;
	private Pwm pwm3;

	// Digital inputs
	private DigitalInput pulseOn;
	private DigitalInput stopSignalOn;
	
	// ADS1115 16-Bit ADC
	private ADS1115_ADS1015 ads;
	
	public IO() {
		try {
			Context pi4j = Pi4J.newAutoContext();
			pwm0 = createDigitalPWMOutput(12, pi4j);
			pwm1 = createDigitalPWMOutput(13, pi4j);
			pwm2 = createDigitalPWMOutput(18, pi4j);
			pwm3 = createDigitalPWMOutput(19, pi4j);
			pulseOn = createDigitalInput(23, pi4j);
			stopSignalOn = createDigitalInput(24, pi4j);
			ads = new ADS1115_ADS1015(pi4j, 1, ADS1115_ADS1015.ADS_ADDR_GND); // I2C bus = 1
			ads.useADS1115();
			
		} catch (Pi4JException e) {
			// TODO Auto-generated catch block
			e.printStackTrace();
		}
	}
	

	// This will read the pulse input and the stop signal
	private DigitalInput createDigitalInput(int pinInput, Context pi4j) {
		try {
			DigitalInputConfig config = DigitalInput.newConfigBuilder(pi4j)
					.name("Digital input")
					.address(pinInput)
					.pull(PullResistance.PULL_DOWN)
					.debounce(3000L)
					.build();

	        // get a Digital Input I/O provider from the Pi4J context
	        DigitalInputProvider digitalInputProvider = pi4j.provider("pigpio-digital-input");
	        return digitalInputProvider.create(config);
		} catch (Exception e) {
			e.printStackTrace();
		}
		return null;

	}

	// This will turn on/off the FQP30N06L MOSFET
	private Pwm createDigitalPWMOutput(int pinOutput, Context pi4j) {
        try {
            // use try-with-resources to auto-close I2C when complete
        	PwmConfig config = Pwm.newConfigBuilder(pi4j)
                    .name("My Test PWM Pin")
                    .address(pinOutput)
                    .pwmType(PwmType.HARDWARE)
                    .frequency(pwmFrequency)   // optionally pre-configure the desired frequency to 1KHz
                    .dutyCycle(0)     // optionally pre-configure the desired duty-cycle (50%)
                    .shutdown(0)       // optionally pre-configure a shutdown duty-cycle value (on terminate)
                    .initial(0)     // optionally pre-configure an initial duty-cycle value (on startup)
                    .build();
        	Pwm pwm = pi4j.providers().get(PiGpioPwmProvider.class).create(config);
        	pwm.on();
			return pwm;
		} catch (Exception e) {
			e.printStackTrace();
		}
		return null;
	}

}
```

And now to the error. Not sure what it means. But it close the spring boot application.
```

2020-07-11 12:32:34 sigHandler: Unhandled signal 11, terminating

2020-07-11 12:32:34 sigHandler: Unhandled signal 11, terminating
```

hackerjimbo, 2020-07-11:
On 11/07/2020 11:35, Daniel Mårtensson wrote:

> ||
>
> And now to the error. Not sure what it means. But it close the spring 
> boot application.
>
> |2020-07-11 12:32:34 sigHandler: Unhandled signal 11, terminating 
> 2020-07-11 12:32:34 sigHandler: Unhandled signal 11, terminating |

I think this is an issue that I uncovered a while back. Essentially 
PiGpio decides it wants to control all the signal handlers. The Java 
runtime has different ideas. Chaos ensues.

I'll see if I can get a fix sorted this weekend.

Regards,

Jim.

DanielMartensson, 2020-07-11:
> On 11/07/2020 11:35, Daniel Mårtensson wrote: || And now to the error. Not sure what it means. But it close the spring boot application. |2020-07-11 12:32:34 sigHandler: Unhandled signal 11, [...]

Ok! I tried to add id-labels to all of them now. Separately id:s. Not I only got this error `2020-07-11 12:40:14 sigHandler: Unhandled signal 4, terminating
`

savageautomate, 2020-07-12:
@DanielMartensson 

> So the conclusion is that there are 0-31 hardware PWM pins, bit raspberry pi can only use two of them?
> 
> I need 4 PWM pins - hardware.

No, I don't think that accurately reflects the situation.  

See this post in the Raspberry Pi forums:
https://www.raspberrypi.org/forums/viewtopic.php?t=150254#p987511

> Re: Which pin(s) on RPi 3B is PWM capable
> Quote
> Fri Jun 03, 2016 7:51 am
>
> pigpio (and others) will provide hardware timed PWM on all the GPIO connected to the expansion header. Once started the > PWM uses no CPU time. It has less resolution and restricted frequencies compared to the two hardware PWM channels.  >The default of 800Hz 250 steps is pretty much the same as an Arduino.

The Raspberry Pi supports 2 hardware based PWM channels.  You can access these two channels via 2 separate sets of 4 GPIO header pins.  But still limited to only 2 channels (_2 unique PWM timing configurations_).  

The PIGPIO library is providing additional PWM support to any of the GPIO pins (0-31) and its using some hardware timing technique to optimize performance --- but its not the same as the actual hardware PWM pins natively on the RaspberryPi.  In the Pi4J API, we would call this "_Software_" PWM and you would need to set `.pwmType(PwmType.SOFTWARE)`.  We consider this software-based PWM because its being provided at a software layer, in this case by the PIGPIO library.  

If you need 4 PWM pins, then I would suggest trying to use the software PWM functionality, it may be perfectly fine for your application.   If they are not good enough, then you will probably need a PWM expander board/chip (controlled by I2C/SPI) to provide additional PWM support.  

Thanks, Robert

savageautomate, 2020-07-12:
> Ok! It's seems to work now. But I have another issue you might want to look at.
> 
> I have selected the pins you suggested. It's seems to...not giving any errors. Also I removed the id-label. Not sure if they are needed.

The `ID` is not required, it will get auto-assigned an identifier string if you don't provide one.  If you do provide one, they must be unique.  

As @hackerjimbo mentioned, the signals issues was recently uncovered and is being tracked via issue: #15

DanielMartensson, 2020-07-12:
> @DanielMartensson
>
>
> No, I don't think that accurately reflects the situation.
>
> See this post in the Raspberry Pi forums:
> https://www.raspberrypi.org/forums/viewtopic.php?t=150254#p987511
>
>
> The [...]

Hmm.. Still got the error, even if I replaced all hardware enums with software enums.

```
2020-07-12 19:56:03 sigHandler: Unhandled signal 4, terminating

```

hackerjimbo, 2020-07-12:
That's because it's got nothing to do with the `enum`s, it's all about signal handling. Signal 4 is illegal instruction rather than your earlier signal 11 (segmentation fault).

@DanielMartensson, what hardware are you running on?

DanielMartensson, 2020-07-12:
> That's because it's got nothing to do with the `enum`s, it's all about signal handling. Signal 4 is illegal instruction rather than your earlier signal 11 (segmentation fault).
> 
> @DanielMartensson, what hardware are you running on?

I'm using Raspberry Pi 4B

hackerjimbo, 2020-07-12:
OK, I now have no idea why it's not work. It _should_ get this right.

DanielMartensson, 2020-07-12:
> OK, I now have no idea why it's not work. It _should_ get this right.

Perhaps the Pi4J project is to complex? 
I personally would avoid OOP when talking to hardware. Only use functions.

hackerjimbo, 2020-07-12:
I make a lot of use of OOP on my hardware drivers. A great (and very recent) one was talking to the SH1106 display driver. This is available via both I2C and SPI. The _vast_ majority of the code is common but I have a few abstract methods that allow me to drive it one the specific device in use. See also the MCP27x17.

Are you running 32 or 64 bit on the Pi 4? I know for certain that the 32-bit one probes the exact hardware in use by checking what works and what doesn't (trapping the illegal instructions and from that deducing what the functionality available). This allows the JIT compilation to be perfectly tuned for the hardware.

I wouldn't describe Pi4J as too complex. Software only gets complex when it hits 1 000 000 lines! :-) 

Right now we're shaking down the pigpio drivers (which I think we've fixed now) and getting Pi4J to talk nicely to them even on a Pi Zero (getting cross-compilation right, which I also think we've done). Not bad for about a week's work!

DanielMartensson, 2020-07-12:
> I make a lot of use of OOP on my hardware drivers. A great (and very recent) one was talking to the SH1106 display driver. This is available via both I2C and SPI. The _vast_ majority of the code is [...]

I'm using ARMv7 Processor, which is 32 bit. 

```
pi@raspberrypi:~/Documents $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

Hardware	: BCM2835
Revision	: b03112
Serial		: 1000000020b891af
Model		: Raspberry Pi 4 Model B Rev 1.2
pi@raspberrypi:~/Documents $
```

hackerjimbo, 2020-07-12:
The `v7l` means that it's a 64-bit processor operating in 32-bit mode. You've also got a more recent model than I because I have the 1.1 version. Although the _other_ Pi 4 is a 1.4! :-)

However, knowing that you're running in 32-bit mode is helpful. Which JVM are you using?

DanielMartensson, 2020-07-12:
> The `v7l` means that it's a 64-bit processor operating in 32-bit mode. You've also got a more recent model than I because I have the 1.1 version. Although the _other_ Pi 4 is a 1.4! :-)
> 
> However, knowing that you're running in 32-bit mode is helpful. Which JVM are you using?

```
pi@raspberrypi:~/Documents $ java --version
openjdk 11.0.7 2020-04-14
OpenJDK Runtime Environment (build 11.0.7+10-post-Raspbian-3deb10u1)
OpenJDK Server VM (build 11.0.7+10-post-Raspbian-3deb10u1, mixed mode)
pi@raspberrypi:~/Documents $ 
```

hackerjimbo, 2020-07-12:
Just for reference for @savageautomate and others, on the Pi Zeros I'm using:

`openjdk 11.0.7 2020-04-14 LTS
OpenJDK Runtime Environment Zulu11.39+61-CA (build 11.0.7+10-LTS)
OpenJDK Client VM Zulu11.39+61-CA (build 11.0.7+10-LTS, mixed mode)`

DanielMartensson, 2020-07-12:
> Just for reference for @savageautomate and others, on the Pi Zeros I'm using:
> 
> > `openjdk 11.0.7 2020-04-14 LTS OpenJDK Runtime Environment Zulu11.39+61-CA (build 11.0.7+10-LTS) OpenJDK Client VM Zulu11.39+61-CA (build 11.0.7+10-LTS, mixed mode)`

Do you think Java 8 will solve this issue temporary?

hackerjimbo, 2020-07-12:
No. Pi4J-v2 is based on Java 11 and uses the extra features of that. The issues aren't with the languages but with the binding to PiGPIO. I _think_ we've got that more-or-less understood and fixed now.

DanielMartensson, 2020-07-12:
> No. Pi4J-v2 is based on Java 11 and uses the extra features of that. The issues aren't with the language version but with the binding to PiGPIO. I _think_ we've got that more-or-less understood and fixed now.

If you want! I can upload my web application software so you can really see what issues there are inside my setup?

hackerjimbo, 2020-07-12:
I'd like to get the known issues fixed and publicly available. Then we'll see which of yours remain.

DanielMartensson, 2020-07-12:
> I'd like to get the known issues fixed and publicly available. Then we'll see which of yours remain.

This will help. 
https://github.com/DanielMartensson/OpenSourceLogger

FDelporte, 2020-07-13:
Thanks for all the info shared in this topic, I used it for the initial content of the PWM page on the new site --> https://v2.pi4j.com/getting-started/io-examples/pwm

DanielMartensson, 2020-07-13:
> Thanks for all the info shared in this topic, I used it for the initial content of the PWM page on the new site --> https://v2.pi4j.com/getting-started/io-examples/pwm

Did you try my software? Perhaps it will work on your Pi?

FDelporte, 2020-07-13:
@DanielMartensson no sorry, not yet, I'm a bit behind with testing on the Pi itself...

savageautomate, 2020-07-13:
@DanielMartensson 

Here is a compiled from (`pi4j-library-pigpio-2.0-SNAPSHOT.jar`) JAR with @hackerjimbo 's suggested fix for the signals issue if you want to give it a try:
https://www.dropbox.com/s/8cm176mbknk2s0m/pi4j-library-pigpio-2.0-SNAPSHOT.jar

DanielMartensson, 2020-07-13:
> @DanielMartensson
> 
> Here is a compiled from (`pi4j-library-pigpio-2.0-SNAPSHOT.jar`) JAR with @hackerjimbo 's suggested fix for the signals issue if you want to give it a try:
> https://www.dropbox.com/s/8cm176mbknk2s0m/pi4j-library-pigpio-2.0-SNAPSHOT.jar

Is is available for `pom.xml`?

savageautomate, 2020-07-13:
@DanielMartensson 

As soon as @hackerjimbo confirms the fix [here](https://github.com/Pi4J/pi4j-v2/issues/15), I'll deploy it to the maven repositories.

DanielMartensson, 2020-07-13:
> @DanielMartensson
> 
> As soon as @hackerjimbo confirms the fix [here](https://github.com/Pi4J/pi4j-v2/issues/15), I'll deploy it to the maven repositories.
> 
> In the meantime, you could manually copy the JAR file into the directory where you are running it on the Pi.

Do I have to remove the Pi4J dependencies in Pom.xml?

hackerjimbo, 2020-07-13:
@savageautomate — it works! Tested on a Pi Zero and a Pi 1. Both crash with the old version of the java code and the system pigpio and now work with the OS-supplied pigpio and the fix you posted. Great work!

DanielMartensson, 2020-07-13:
Have a look at this code:
https://github.com/DanielMartensson/OpenSourceLogger/blob/master/src/main/java/se/danielmartensson/views/components/threads/ControlThread.java

```
       public ControlThread(IO io) {
		pwm0 = io.getPwm0();
		pwm1 = io.getPwm1();
		pwm2 = io.getPwm2();
		pwm3 = io.getPwm3();
		pulseOn = io.getPulseOn();
		stopSignalOn = io.getStopSignalOn();
		ads = io.getAds();
	}
```

https://github.com/DanielMartensson/OpenSourceLogger/blob/master/src/main/java/se/danielmartensson/views/ControlView.java

```
                @Autowired
	        private IO io;

                // Start the tread forcontrol
		if(control == null) {
			control = new ControlThread(io);
			control.start();
		}
```
https://github.com/DanielMartensson/OpenSourceLogger/blob/master/src/main/java/se/danielmartensson/pi4j/IO.java

            ```
public IO() {
		try {
			Context pi4j = Pi4J.newAutoContext();
			pwm0 = createDigitalPWMOutput(12, pi4j, "pwm0");
			pwm1 = createDigitalPWMOutput(13, pi4j, "pwm1");
			pwm2 = createDigitalPWMOutput(18, pi4j, "pwm2");
			pwm3 = createDigitalPWMOutput(19, pi4j, "pwm3");
			pulseOn = createDigitalInput(23, pi4j, "di0");
			stopSignalOn = createDigitalInput(24, pi4j, "di1");
			ads = new ADS1115_ADS1015(pi4j, 1, ADS1115_ADS1015.ADS_ADDR_GND); // I2C bus = 1
			ads.useADS1115();
			
		} catch (Pi4JException e) {
			// TODO Auto-generated catch block
			e.printStackTrace();
		}
	}

```

DanielMartensson, 2020-07-13:
> @savageautomate — it works! Tested on a Pi Zero and a Pi 1. Both crash with the old version of the java code and the system pigpio and now work with the OS-supplied pigpio and the fix you posted. Great work!

Why did it not work for me?

savageautomate, 2020-07-13:
@DanielMartensson ,

> Why did it not work for me?

Please try again.  I have deployed the latest version to the Maven SNAPSHOT repository now. 

You may want to first delete the directory and files on your local system where this JAR file could be cached to ensure the new version is picked up by Maven.

```
~/.m2/repository/com/pi4j/pi4j-library-pigpio/
```

DanielMartensson, 2020-07-14:
> @DanielMartensson ,
> 
> > Why did it not work for me?
> 
> Please try again. I have deployed the latest version (_with the signal fix_) to the Maven SNAPSHOT repository now.
> 
> You may want to first delete the directory and files on your local system where this JAR file could be cached to ensure the new version is picked up by Maven.
> 
> ```
> ~/.m2/repository/com/pi4j/pi4j-library-pigpio/
> ```

Ok! Seems that there are no errors now :) Need to check if the PWM's can pulse too.

savageautomate, 2020-07-17:
@DanielMartensson 

Just checking in to see if this issue can now be closed. 

Thanks, Robert

DanielMartensson, 2020-07-17:
> @DanielMartensson
> 
> Just checking in to see if this issue can now be closed.
> 
> Thanks, Robert

I haven't test my PWM's yet. I just have created my project and I will soon test it.
https://github.com/DanielMartensson/OpenSourceLogger

DanielMartensson, 2020-07-18:
@savageautomate 

Nope. No activity from the GPIO's here:

https://github.com/DanielMartensson/OpenSourceLogger/blob/master/src/main/java/se/danielmartensson/views/components/threads/ControlThread.java

Row 134