Pi4J/pi4j #34: Create ADC and DAC analog IO provider examples

savageautomate, 2020-07-17:
The Pi4J V2 API supports analog input and output providers and interfaces.  
However, since the RaspberryPi does not include native analog support we should create some examples analog input and analog output providers as reference examples for others how wish to create their own IO providers.
 
See: https://github.com/DanielMartensson/OpenSourceLogger/issues/2

DanielMartensson, 2020-07-17:
I would recommend rewriting STM32 C libraries to Java. I have a repository about C libraries for STM31.

DanielMartensson, 2020-10-07:
My ADS1115 library works good with Pi4J 2.0

FDelporte, 2020-10-07:
you have a link?

DanielMartensson, 2020-10-07:
> Daniel, you have a link?

https://github.com/DanielMartensson/OpenSourceLogger/blob/master/src/main/java/se/danielmartensson/pi4j/ADS1115_ADS1015.java

DanielMartensson, 2020-10-07:
@FDelporte 

By the way! Do you have an example for PWM? Mine won't work for me in OpenSourceLogger repository. Else, the OpenSourceLogger works perfect. Excellent tool for logging 4-20 mA data into a database quick.