Pi4J/pi4j #36: Support for camera

DanielMartensson, 2020-07-28:
Hi!

I want to give a suggestion for this project. Pi4J 2.0 should have access to a camera.

I can help with picture identification/classification in pure Java if you want implement PCA + LDA methods for image classification. It's the same methods being used in OpenCV.

mstahv, 2026-02-10:
Hi! I know this issue is quite old, but I recently created a library that might be relevant for anyone still looking for camera support on Raspberry Pi from Java.

**[libcamera-4j](https://github.com/mstahv/heisala-jetty/tree/main/libcamera-4j)** is a Java library that provides access to Raspberry Pi cameras via the modern libcamera stack. The native C++ code was essentially vibe coded based on the Python wrappers for libcamera (picamera2).

Features:
- Capture JPEG images at various resolutions
- Full resolution and raw (DNG) capture support
- Camera controls: autofocus, manual focus, exposure, gain/ISO, rotation
- Async capture API with CompletableFuture
- Image metadata (exposure time, ISO, white balance, etc.)

Example usage:
```java
byte[] jpeg = CameraCapture.captureJpeg(1920, 1080);

// Or with settings
CameraSettings settings = CameraSettings.builder()
    .rotation(180)
    .manualFocus(0.0f)  // infinity
    .build();
CaptureResult result = CameraCapture.captureWithMetadata(1920, 1080, settings);
```

The library is currently a module in my [heisala-jetty](https://github.com/mstahv/heisala-jetty) hobby project (a jetty monitoring system), but could be moved to a separate repository or contributed to Pi4J if there's interest from others. The `heisala-jetty-server` module serves as the most advanced demo (see screenshots below, sorry about the dirty window glass, there was quite a storm here during the holidays) - it's a Quarkus-based web application with camera controls UI that should work on any Raspberry Pi if you have a camera module connected.

So far I've only tested it with **Camera Module 3** on a **Raspberry Pi Zero 2**, but it should work with other libcamera-compatible cameras.

Requirements:
- Raspberry Pi with libcamera installed
- Native library built on the Pi (CMake + libcamera-dev) or Docker on e.g. ARM Mac like I did (My raspberry pi zero 2W was updated to latest, there was a bit of fiddling to make the native library match with the 0.7 of libcamera)

For the PCA/LDA facial recognition use case mentioned in this issue, libcamera-4j could handle the image acquisition while your classification code processes the captured `BufferedImage`.

If there's interest, I can also publish it to Maven Central. Feedback and contributions welcome!

[image]
[image]

FDelporte, 2026-02-12:
Thanks @mstahv for sharing your work! Would be willing to provide more content (or a pull request in https://github.com/Pi4J/pi4j.github.io) to add this info to https://www.pi4j.com/blog/ or another place on the website where this info would be findable as we want the Pi4J website to be "the source of info for all java-on-pi related info" :-)

mstahv, 2026-02-12:
BTW. There is also now a timelapse feature, but that uses ffmpg directly (quite straightforward call from java app, hardware accelerated). Bit slow on longer video on the Zero 2 W I'm using it for testing. Should probably fetch 5 with some memory for this purpose, although torturing myself with these low end computers is half of the fun 😂

FDelporte, 2026-02-12:
@mstahv moving to a Pi4J GitHub repo would make sense as a separate project, not inside the core library. Do you agree? Do you want to commit to be the lifetime maintainer? ;-)