Pi4J/pi4j #38: Collaboration - diozero

mattjlewis, 2020-10-17:
I've just come across this rewrite of Pi4j having been quite familiar with Pi4j v1. I'm the author of [diozero](www.diozero.com) as well as [pigpioj](https://github.com/mattjlewis/pigpioj) and can see a huge overlap between pi4j v2 and diozero/pigpioj. diozero is completely device independent, supports dynamic loading of providers, has its own built-in provider - a goal that I see Pi4j v2 is also trying to achieve. I notice that you are also implementing pigpioj support - my pigpioj library is very stable now and supports local as well as remote interfacing via sockets.
It's seems a shame to duplicate so much effort. I'd be interested in your thoughts on how we could collaborate.

Regards,
Matt Lewis

FDelporte, 2020-10-17:
Thanks Matt for reaching out! Indeed would be a shame to have two similar projects if we could collaborate! Can we get in contact through email to exchange more info - > frank@webtechie.be

FDelporte, 2020-10-21:
Has been further discussed through email

closed 2020-10-21 by FDelporte