Pi4J/pi4j #39: Cannot extract native libraries from jar on Pi 4B

lukehutch, 2020-10-22:
Hi, I get this error message when trying to use Pi4J v2 on a Raspberry Pi 4B from a "fat jar":

UnsatisfiedLinkError: Pi4J was unable to extract and load the native library [/lib/armhf/libpi4j-pigpio.so] from the embedded resources inside this JAR [/home/pi/myjar.jar]. to a temporary location on this system.  You can alternatively define the 'pi4j.library.path' system property to override this behavior and specify the library path.
	at com.pi4j.library.pigpio.util.NativeLibraryLoader.load(NativeLibraryLoader.java:172)
	at com.pi4j.library.pigpio.internal.PIGPIO.<clinit>(PIGPIO.java:76)
	at com.pi4j.library.pigpio.impl.PiGpioNativeImpl.gpioInitialise(PiGpioNativeImpl.java:97)
	at com.pi4j.library.pigpio.PiGpio.initialize(PiGpio.java:155)
	at com.pi4j.plugin.pigpio.provider.gpio.digital.PiGpioDigitalInputProviderImpl.create(PiGpioDigitalInputProviderImpl.java:64)
	at com.pi4j.plugin.pigpio.provider.gpio.digital.PiGpioDigitalInputProviderImpl.create(PiGpioDigitalInputProviderImpl.java:45)

The /tmp directory is writeable, and the library is in the correct path within the jar [/lib/armhf/libpi4j-pigpio.so], so I don't know why this is failing.

How do I enable log4j logging in Pi4j v2? I looked at the code in NativeLibraryLoader, but I can't see the debug messages on the console to determine what is going wrong here. I checked the docs but didn't see a suggestion about how to enable logging.

lukehutch, 2020-11-21:
If I extract `libpi4j-pigpio.so` to `/home/pi` then launch with `-Dpi4j.library.path=/home/pi`, then I get:

```
java.lang.UnsatisfiedLinkError: /home/pi/libpi4j-pigpio.so: libpigpio.so.1: cannot open shared object file: No such file or directory
```

I don't see `libpigpio.so.1` in the pi4j-v2 jar...

Can anyone help with this please?

lukehutch, 2020-11-22:
I managed to manually extract `libpi4j-pigpio.so` from the fat jar and store it in `/home/pi`, then was able to successfully launch with `-Dpi4j.library.path=/home/pi`. I still don't know why the automatic extraction doesn't work.

savageautomate, 2021-03-17:
@lukehutch ,

Sorry for the delay ...
Now that I have completed Pi4J v1.3 and Pi4J v1.4 and released them, I'm focusing on v2. now.  

Have you tried any recent builds?  We now publish the Pi4J-2.0-SNAPSHOT JARs to the OSS Maven Repository as well as here:  https://github.com/Pi4J/download via automated CI/CD scripts when anything is committed/merged into the `main` branch.

Are you running 32-bit Raspberry Pi OS or 64-bit Raspberry Pi OS?

Thanks, Robert

savageautomate, 2021-03-17:
@lukehutch 

> How do I enable log4j logging in Pi4j v2? I looked at the code in NativeLibraryLoader, but I can't see the debug messages on the console to determine what is going wrong here. I checked the docs but didn't see a suggestion about how to enable logging.

It uses the SPF4J API.  You can include the `slf4j-simple` JAR file in your project to print the log message to the console. 
Next, you will need to enable `DEBUG` level logging with code like this.
```java  
        // configure default lolling level, accept a log level as the fist program argument
        System.setProperty("org.slf4j.simpleLogger.defaultLogLevel", "DEBUG");
```

In my case, I see log message like these on my RPi4B (32-bit `armv7l`)

```bash
[main] DEBUG com.pi4j.library.pigpio.util.NativeLibraryLoader - Attempting to load library [libpi4j-pigpio.so] using path: [/lib/armhf/libpi4j-pigpio.so]
[main] DEBUG com.pi4j.library.pigpio.util.NativeLibraryLoader - Library [libpi4j-pigpio.so] loaded successfully using embedded resource file: [/lib/armhf/libpi4j-pigpio.so]
```

lukehutch, 2021-03-17:
> Have you tried any recent builds? We now publish the Pi4J-2.0-SNAPSHOT JARs to the OSS Maven Repository as well as here: https://github.com/Pi4J/download via automated CI/CD scripts when anything is committed/merged into the `main` branch.

I was using the most recent SNAPSHOT build at the time I left the last comment, yes. 

> Are you running 32-bit Raspberry Pi OS or 64-bit Raspberry Pi OS?

64-bit Raspbian on a Raspberry Pi 4B. 

If you want to see the problem, `mvn install` this:

https://github.com/lukehutch/Adafruit-OLED-Bonnet-Toolkit

Then `mvn package` this:

https://github.com/lukehutch/usb-copier

When the resulting fat jar is run on the Pi, it can't find the Pi4J library inside the fat jar, even though it's at the right location. You can see how I had to manually copy the library file out of the jar here, in order to launch the jar:

https://github.com/lukehutch/usb-copier/issues/1

SeanCheatham, 2021-08-06:
I ran into issues loading the native library as well.  In my particular case, a different error indicated it was unable to find the libgpio.so library.  I followed the installation instructions for that [here](http://abyz.me.uk/rpi/pigpio/download.html).

lukehutch, 2021-08-19:
I found a library that solves this problem:

https://github.com/ibmsoe/jniloader

> The static method in `JniLoader` enables developers to ship native libraries in jar files which are extracted and loaded at runtime, making native loading transparent for the end-user (as long as they are on a platform supported by you).

bennymeg, 2024-07-17:
For me simply [installing pigpio](https://stackoverflow.com/a/71034611/4215944) directly on the pi did the trick.