Pi4J/pi4j #40: Unmapping/remapping pins

svnhub, 2020-11-17:
If I create an IO config for a pin with 
```
config = DigitalInput.newConfigBuilder(rpi4j).provider("pigpio-digital-input").id("pin-"+pinNumber).address(pinNumber)....
pin = rpi4j.create(config);
```
the name "pinX" is registered in the Registry and the pin works accordingly.

If I then want to disconnect the pin I call 
`pin.shutdown(rpi4j);`
which seems to forward the disconnect to pigpio but doesn't unregister the IO in the Registry.

Any attempt to then create a new config with the same id creates an exception:
```
com.pi4j.io.exception.IOAlreadyExistsException: IO instance [pin14] already exists in the Pi4J runtime context; unable to create a new instance using this reserved id.
	at com.pi4j.provider.impl.ProviderProxyHandler.invoke(ProviderProxyHandler.java:99)
	at com.sun.proxy.$Proxy2.create(Unknown Source)
	at com.pi4j.context.Context.create(Context.java:309)
	at com.pi4j.internal.IOCreator.create(IOCreator.java:60)
	at com.pi4j.internal.IOCreator.create(IOCreator.java:101)
	at com.pi4j.internal.IOCreator.create(IOCreator.java:189)
        ...
```

The registry on the autoContext doesn't seem to have a way to remove things from it (it is a DefaultRegistry not RuntimeRegistry)

eitch, 2021-01-19:
This is a good point. I just ran into it myself.

beulenbilly, 2022-01-30:
Is there is a possibility to support you to solve the issue?

touchhome, 2022-11-17:
Does this thread alive?

FDelporte, 2022-11-17:
I see @eitch did a commit https://github.com/Pi4J/pi4j-v2/pull/198/files related to this topic.

eitch, 2022-11-17:
I merged a PR, and i think it should be fine now since v2.2.0

closed 2022-11-17 by eitch

stefanhaustein, 2025-01-19:
I am still running into this issue with 2.7 and Context.shutdown(id) doesn't seem to be called from anywhere (I'd expect it to be called from (e.g.) DigitalBase.shutdown(context)?

Btw: Given that a context is required for creation (at least for DIO), isn't requiring a context for shutdown an (unnecessary?) opportunity for inconsistencies?

eitch, 2025-03-07:
Hi @stefanhaustein 

Please check this test: com.pi4j.test.registry.RegistryTest#testShutdownAndRecreate()

This shows that one should ask the context to shutdown the input, and not shutdown the input directly.