Pi4J/pi4j #43: Pi4j should not have its own IOException class which clashes with java.io.IOException

eitch, 2021-01-18:
Pi4j should not have its own IOException class which clashes with java.io.IOException

This makes client needlessly complicated, as one might have both being thrown.

FDelporte, 2021-01-21:
agree this indeed can cause confusion... but we need to find a matching replacement....

From the documentation:

 _* This exception is thrown if a platform assignment is attempted when a
 * platform instance has already been assigned._

Maybe IOOccupiedException would be a good fit?

eitch, 2021-01-22:
If you ask me, then i would say the less exceptions the better. I am more on the side of those that say the less checked exceptions the better =). 

The question to ask here: Will a user really care about a specific exception? Will they handle it specially and thus call a different routine in their code, or will it just cause them to "catch(Exception e)"? If this doesn't require special handling then i suggest throwing the base Pi4j Exception for this platform instance assignment error

FDelporte, 2021-01-22:
@savageautomate Eitch makes a good point here, Robert what do you think?

eitch, 2021-02-22:
This has now been closed due to #66

closed 2021-02-22 by eitch