Pi4J/pi4j #48: Context should be renamed

eitch, 2021-01-19:
I think we should rename Pi4j class to Pi4jProvider and Context to Pi4j.

Currently in all the code examples we use the following code:

    var pi4j = Pi4J.newAutoContext();

Some people like to use these vars, but others like to use the actual class. And in this case the class is Context, but the variable is pi4j. IMHO this does not make sense.

So if we rename as described, we get this:

    var pi4j = Pi4JProvider.newAutoContext();

or this:

    Pi4j pi4j = Pi4JProvider.newAutoContext();

which looks much better and makes more sense for newbies.

savageautomate, 2021-01-20:
I'll have to think on this.  I really like the `Pi4J` moniker being the static helper.  If we just change `Context` to `Pi4JContext` then we still end up with this:

```java
var pi4j = Pi4J.newAutoContext();
```

The `Pi4J` static helper only serves to create a context today, but we could add more helper methods to it as needed.

eitch, 2021-01-21:
True, but then the suffic Provider would fit those contexts nicely, as it provides contexts and builders. Those builders are used in initialization and then the actual context will become the main object on which actions are performed.