Pi4J/pi4j #56: Will V2 support I2C clock stretching?

FDelporte, 2021-01-26:
Info provided by @thegreystone 

More info on https://github.com/raspberrypi/linux/issues/254 describing a bug in the I2C master.

Maybe could be fixed with i2c-gpio overlay, see https://github.com/raspberrypi/linux/blob/rpi-4.4.y/arch/arm/boot/dts/overlays/i2c-gpio-overlay.dts

eitch, 2021-01-26:
I had this issue myself when writing to an eeprom, and my solution was to set a new clkt_tout which was set in /etc/rc.local after startup. I found a howto here: https://github.com/raspihats/raspihats/blob/master/clk_stretch/i2c1_get_clkt_tout.c

Binaries are compiled and available here: https://github.com/eitch/raspi-builder/tree/develop/bin-files64

But yes we can try the overlay and see if that solves the issue too.

eitch, 2021-06-09:
@FDelporte With the linuxfs-i2c plugin, I2C clock stretching will remain an issue that needs to be solved by setting register in memory. This will require root access, but is something that could be set at OS startup, and does not need to be solved by pi4j.

FDelporte, 2021-06-13:
@eitch is this something that can (easily) be documented or described on the site on
https://pi4j.com/documentation/io-examples/i2c/ ?

eitch, 2021-07-14:
@FDelporte https://github.com/Pi4J/pi4j.github.io/commit/18fa2cd309a7d9aed44a57343ce1e66d37cd22a0 this fine?

FDelporte, 2021-07-19:
perfect @eitch  thanks a lot

closed 2021-07-19 by FDelporte