Pi4J/pi4j #59: To merge or not to merge, that's the question

FDelporte, 2021-02-03:
After release of V1.4 what to do with this repository?

* merge into V1 and remove this one?
* archive V1 and rename this one to pi4j?

Some food for thought: https://github.community/t/merging-two-repositories/118538

When merging V2 into V1, issues could be moved like this: https://docs.github.com/en/github/managing-your-work-on-github/transferring-an-issue-to-another-repository

eitch, 2021-02-04:
Since the history of this repo is not compatible with v1 i think we should keep both v1 and v2 repos. Maybe rename v1 repo to pi4j-v1. This way people using the old version realize this has changed.

FDelporte, 2021-03-04:
Will not do merge.
Probably best solution after the first release of V2: 
- rename pi4j to pi4j-v1 and archive it
- rename pi4j-v2 to pi4j

closed 2021-03-04 by FDelporte