Pi4J/pi4j #60: pigpio initialisation failed

jahlmann, 2021-02-06:
I get the following error:

_WARN com.pi4j.library.pigpio.impl.PiGpioNativeImpl - PIGPIO ERROR: PI_INIT_FAILED; pigpio initialisation failed
java.lang.reflect.InvocationTargetException_

when I try:

var pi4j = Pi4J.newAutoContext();

            var ledConfig = DigitalOutput.newConfigBuilder(pi4j)
                    .id("led")
                    .name("LED Flasher")
                    .address(PIN_LED)
                    .shutdown(DigitalState.LOW)
                    .initial(DigitalState.LOW)
                    .provider("pigpio-digital-output");

I have turned off any pigpio daemons before running the program, please also 
[Led1.pdf](https://github.com/Pi4J/pi4j-v2/files/5936775/Led1.pdf)
see attached.

hackerjimbo, 2021-02-06:
That looks like a warning, rather than an error. Does it actually work or, if it doesn't what other error messages are you getting?

jahlmann, 2021-02-06:
Thank you for replying.
I only get this message:[qtp20093513-13] WARN com.pi4j.library.pigpio.impl.PiGpioNativeImpl - PIGPIO ERROR: PI_INIT_FAILED; pigpio initialisation failed
java.lang.reflect.InvocationTargetException

Venlig Hilsen

Jens Ahlmann Hansen, ph.d.

Bromarksvej 8 B
Ollerup
5762 Vester Skerninge

Mobil: 29 46 43 22

FDelporte, 2021-02-06:
Seems to point in the direction of a problem with your PiGpio.
What OS are you using?
I just retested the minimal example on https://v2.pi4j.com/getting-started/minimal-example-application/ with a new Pi4 and Raspberry Pi OS Full edition and this works without problem. Can you test the same?

g-gunther, 2021-02-06:
I got the same error recently on my Raspberry Pi 3 Model B Rev 1.2
I installed pigpio by following this: http://abyz.me.uk/rpi/pigpio/download.html and all the tests were OK but I got the same error: 
_WARN com.pi4j.library.pigpio.impl.PiGpioNativeImpl - PIGPIO ERROR: PI_INIT_FAILED; pigpio initialisation failed
java.lang.reflect.InvocationTargetException_

I finally make it work by updating the native library method: Java_com_pi4j_library_pigpio_internal_PIGPIO_gpioInitialise
I replaced 
`gpioCfgSetInternals (gpioCfgGetInternals () | PI_CFG_NOSIGHANDLER); `
by 
`gpioCfgSetInternals (PI_CFG_NOSIGHANDLER);`

No idea why but now it works fine

jahlmann, 2021-02-07:
Thank you for both replies.

I will try your suggestions - and post the results.

jahlmann, 2021-02-17:
We are currently using 10 Raspberry Pi 3 Model B Rev 1.2 for an IOT course. For now, we will continue to use WPI, which works for our purposes.
Excuse my ignorance, but exactly where can you update:

native library method: Java_com_pi4j_library_pigpio_internal_PIGPIO_gpioInitialise:
and replace
gpioCfgSetInternals (gpioCfgGetInternals () | PI_CFG_NOSIGHANDLER);
by
gpioCfgSetInternals (PI_CFG_NOSIGHANDLER);_

Thank you

g-gunther, 2021-02-18:
You have to extract the file libpi4j-pigpio.so which is in the jar pi4j-library-pigpio-2.0.jar (folder: /lib/armhf)
After editing you can either repackage the jar file or put the .so file somewhere on your filesystem and use the -Dpi4j.library.path option to point to that file

jahlmann, 2021-02-18:
Many thanks - I will try that.

tatery, 2021-03-11:
Hi,

I've got the same issue when running my code as maven test:

```java
mvn clean test -Dtestng.suite=testng.xml

PLATFORMS: [1] "Pi4J Runtime Platforms" <com.pi4j.platform.impl.DefaultPlatforms>
└─PLATFORM: "RaspberryPi Platform" {raspberrypi} <com.pi4j.plugin.raspberrypi.platform.RaspberryPiPlatform> {Pi4J Platform for the RaspberryPi series of products.}
[ERROR] Tests run: 1, Failures: 1, Errors: 0, Skipped: 0, Time elapsed: 3.891 s <<< FAILURE! - in TestSuite
[ERROR] test(rpi.DigitalIoTest)  Time elapsed: 0.199 s  <<< FAILURE!
java.lang.reflect.InvocationTargetException
        at rpi.DigitalIoTest.test(DigitalIoTest.java:48)
Caused by: java.io.IOException: PIGPIO ERROR: PI_INIT_FAILED; pigpio initialisation failed
        at rpi.DigitalIoTest.test(DigitalIoTest.java:48) 
```

However minimal example is working correctly:

```
sudo ./run.sh
[main] INFO com.pi4j.util.Console -

[main] INFO com.pi4j.util.Console - ************************************************************
[main] INFO com.pi4j.util.Console - ************************************************************
[main] INFO com.pi4j.util.Console -
[main] INFO com.pi4j.util.Console -                   <-- The Pi4J Project -->
[main] INFO com.pi4j.util.Console -                    Minimal Example project
[main] INFO com.pi4j.util.Console -
[main] INFO com.pi4j.util.Console - ************************************************************
[main] INFO com.pi4j.util.Console - ************************************************************
[main] INFO com.pi4j.util.Console -
[main] INFO com.pi4j.Pi4J - New auto context
[main] INFO com.pi4j.Pi4J - New context builder
[main] INFO com.pi4j.util.Console - --------------------
[main] INFO com.pi4j.util.Console - |  Pi4J PLATFORMS  |
[main] INFO com.pi4j.util.Console - --------------------
[main] INFO com.pi4j.util.Console -
PLATFORMS: [1] "Pi4J Runtime Platforms" <com.pi4j.platform.impl.DefaultPlatforms>
└─PLATFORM: "RaspberryPi Platform" {raspberrypi} <com.pi4j.plugin.raspberrypi.platform.RaspberryPiPlatform> {Pi4J Platform for the RaspberryPi series of products.}
[main] INFO com.pi4j.util.Console -
[main] INFO com.pi4j.util.Console - ---------------------------
[main] INFO com.pi4j.util.Console - |  Pi4J DEFAULT PLATFORM  |
[main] INFO com.pi4j.util.Console - ---------------------------
[main] INFO com.pi4j.util.Console -
PLATFORM: "RaspberryPi Platform" {raspberrypi} <com.pi4j.plugin.raspberrypi.platform.RaspberryPiPlatform> {Pi4J Platform for the RaspberryPi series of products.}
[main] INFO com.pi4j.util.Console -
[main] INFO com.pi4j.util.Console - --------------------
[main] INFO com.pi4j.util.Console - |  Pi4J PROVIDERS  |
[main] INFO com.pi4j.util.Console - --------------------
[main] INFO com.pi4j.util.Console -
PROVIDERS: [12] "I/O Providers" <com.pi4j.provider.impl.DefaultProviders>
├─SERIAL: [2] <com.pi4j.io.serial.SerialProvider>
│ ├─PROVIDER: "PiGpio Serial Provider" {pigpio-serial} <com.pi4j.plugin.pigpio.provider.serial.PiGpioSerialProviderImpl> {com.pi4j.plugin.pigpio.provider.serial.PiGpioSerialProviderImpl}
│ └─PROVIDER: "RaspberryPi Serial Provider" {raspberrypi-serial} <com.pi4j.plugin.raspberrypi.provider.serial.RpiSerialProviderImpl> {com.pi4j.plugin.raspberrypi.provider.serial.RpiSerialProviderImpl}
├─DIGITAL_OUTPUT: [2] <com.pi4j.io.gpio.digital.DigitalOutputProvider>
│ ├─PROVIDER: "RaspberryPi Digital Output (GPIO) Provider" {raspberrypi-digital-output} <com.pi4j.plugin.raspberrypi.provider.gpio.digital.RpiDigitalOutputProviderImpl> {com.pi4j.plugin.raspberrypi.provider.gpio.digital.RpiDigitalOutputProviderImpl}
│ └─PROVIDER: "PiGpio Digital Output (GPIO) Provider" {pigpio-digital-output} <com.pi4j.plugin.pigpio.provider.gpio.digital.PiGpioDigitalOutputProviderImpl> {com.pi4j.plugin.pigpio.provider.gpio.digital.PiGpioDigitalOutputProviderImpl}
├─ANALOG_INPUT: [0] <com.pi4j.io.gpio.analog.AnalogInputProvider>
├─PWM: [2] <com.pi4j.io.pwm.PwmProvider>
│ ├─PROVIDER: "PiGpio PWM Provider" {pigpio-pwm} <com.pi4j.plugin.pigpio.provider.pwm.PiGpioPwmProviderImpl> {com.pi4j.plugin.pigpio.provider.pwm.PiGpioPwmProviderImpl}
│ └─PROVIDER: "RaspberryPi PWM Provider" {raspberrypi-pwm} <com.pi4j.plugin.raspberrypi.provider.pwm.RpiPwmProviderImpl> {com.pi4j.plugin.raspberrypi.provider.pwm.RpiPwmProviderImpl}
├─DIGITAL_INPUT: [2] <com.pi4j.io.gpio.digital.DigitalInputProvider>
│ ├─PROVIDER: "RaspberryPi Digital Input (GPIO) Provider" {raspberrypi-digital-input} <com.pi4j.plugin.raspberrypi.provider.gpio.digital.RpiDigitalInputProviderImpl> {com.pi4j.plugin.raspberrypi.provider.gpio.digital.RpiDigitalInputProviderImpl}
│ └─PROVIDER: "PiGpio Digital Input (GPIO) Provider" {pigpio-digital-input} <com.pi4j.plugin.pigpio.provider.gpio.digital.PiGpioDigitalInputProviderImpl> {com.pi4j.plugin.pigpio.provider.gpio.digital.PiGpioDigitalInputProviderImpl}
├─I2C: [2] <com.pi4j.io.i2c.I2CProvider>
│ ├─PROVIDER: "PiGpio I2C Provider" {pigpio-i2c} <com.pi4j.plugin.pigpio.provider.i2c.PiGpioI2CProviderImpl> {com.pi4j.plugin.pigpio.provider.i2c.PiGpioI2CProviderImpl}
│ └─PROVIDER: "RaspberryPi I2C Provider" {raspberrypi-i2c} <com.pi4j.plugin.raspberrypi.provider.i2c.RpiI2CProviderImpl> {com.pi4j.plugin.raspberrypi.provider.i2c.RpiI2CProviderImpl}
├─SPI: [2] <com.pi4j.io.spi.SpiProvider>
│ ├─PROVIDER: "PiGpio SPI Provider" {pigpio-spi} <com.pi4j.plugin.pigpio.provider.spi.PiGpioSpiProviderImpl> {com.pi4j.plugin.pigpio.provider.spi.PiGpioSpiProviderImpl}
[... 96 lines cut ...]
```

[... cut ...]
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2711
Revision        : b03111
Serial          : 10000000blablabla
Model           : Raspberry Pi 4 Model B Rev 1.1

cat /proc/version
Linux version 5.10.17-v7l+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1403 SMP Mon Feb 22 11:33:35 GMT 2021
```

tatery, 2021-03-11:
Small update to my previous post.
When I switched to root ("sudo su") and then run my application (mvn clean test -Dtestng.suite=testng.xml) then it is working. So issue seems to be related to privileges. Any thoughts how to solve this issue?

eitch, 2021-03-11:
@tatery We know of the issue about requiring root privileges. Till now the developers of pigpio don't care about this issue and say one should use the pigpio daemon. The issue is only with I2C and SPI. We are looking to solve this issue.

tatery, 2021-03-11:
@eitch Thanks for response, however in my code I neither use i2c nor spi and an error appears. You wrote; "one should use the pigpio daemon", could you please clarify how to use this daemon with java maven project. Many thanks in advance.

eitch, 2021-03-11:
@tatery currently there is no way to use the daemon from java.

The pi4j-v2 code requires sudo to run the tests. Don't run the tests and you should be fine.

tatery, 2021-03-11:
@eitch Thanks for clarification but not running tests is not possible in my case.
BTW: do you know how to specify pin number used as CS for SPI device (I'd like to use SPI1 with BCM17 as CS)?

savageautomate, 2021-03-11:
@tatery & @eitch 

I believe you can use Pi4J V2 with the `pipgio` daemon.  I have not tried this in quite some time so YMMV, but the support for communicating with the `pipgio` daemon via TCP/IP was originally included in Pi4J v2.   

> **Note**: The `pipgio` daemon also supports communication via pipes, but Pi4J does not support the pipe interface at this time.

I believe all you need to do is to set a couple of system properties that instruct the `pi4j-plugin-pigpio` plugin to use the socket daemon rather than native bindings.    

```
        // PIGPIO which remote Raspberry Pi to connect to
        System.setProperty("pi4j.host", "192.168.1.12");
        System.setProperty("pi4j.remote", "true");
```

This will add some latency and could result in communication issues with high-speed I2C/SPI devices, but it might be worth a try.  You will have to install the latest PIGPIO and follow the instructions to start the `pipgio` daemon.  http://abyz.me.uk/rpi/pigpio/download.html

These properties are not well documented yet as this is not the intended final solution but rather an interim workaround.

Here is an example where they are commented out but were used:
 - https://github.com/Pi4J/pi4j-example-telegraph/blob/master/src/main/java/com/pi4j/demo/telegraph/Telegraph.java#L63-L65

Thanks, Robert

tatery, 2021-03-12:
@savageautomate thanks for hint. I will give it a try it in spare time. However, in my application I'm going to use few i2c and SPI devices (CAN, DAC, ADC, display and others, 15-20 devices in total) so I'm a little bit aware about latency.

savageautomate, 2021-03-17:
FYI, I just tested this today with the latest Pi4J-2.0-SNAPHOT build and it seems to be working on my RPi 4B (32-bit OS) system.  

I'm going to close this issue for now, but please re-open if any problems are encountered.

**Key notes:**
- Make sure you have latest PiGPIO native library installed. (http://abyz.me.uk/rpi/pigpio/download.html)
- Make sure PiGPIO Daemon is not running. (http://abyz.me.uk/rpi/pigpio/faq.html#Cant_initialise_pigpio_library)
- Make sure you run your Java program using `sudo`  (we are looking into alternatives to eliminate `sudo`)

In my case .. I have:

```
> pi@rpi4b-1g:~ $ pi4j --version

--------------------------------------------
            THE Pi4J PROJECT
           (https://pi4j.com)
--------------------------------------------
   PI4J.VERSION     : 2.0-SNAPSHOT
   PI4J.TIMESTAMP   : 2021-03-17 03:08:37
--------------------------------------------

> pi@rpi4b-1g:~ $ pi4j --pigpio

--------------------------------------------
Pi4J - PIGPIO LIBRARY
--------------------------------------------
   PIGPIO.PATH      : /usr/lib/libpigpio.so /usr/local/lib/libpigpio.so
   PIGPIO.VERSION   : 79
--------------------------------------------

```

closed 2021-03-17 by savageautomate

lukehutch, 2021-10-02:
Any updates on this? I still need to run with `sudo` on Pi4J-2.0.

lukehutch, 2021-10-02:
One thing I found helped some (but I think not all) the memory access issues was to edit `/etc/cmdline.txt` and add the kernel option: `iomem=relaxed` . Without this, the GPIO handler cannot access `/dev/mem` .

But I still don't understand why Python libraries don't have a problem accessing GPIO, without any special configuration.

MRROOX, 2021-10-04:
I have the same error, in my case I develop a service with springboot, without sudo it doesn't work.My command : sudo java -jar target/service.jar (works perfect)

dwymark, 2021-12-28:
I ran into this error when using pi4j on a custom image made with Buildroot. The reason for me was that I had to disable the pipio service (`systemctl disable pigpio`).

curiosity4809, 2022-01-23:
any updates?

I try to create an java-app as tomcat-container, which produce the problem with the privileges

ugrepel, 2022-03-20:
+1 using 2.1.1, same as curiosity4809 I'm trying to create a .war file based java-app for a tomcat container, which does (and should) use the user "tomcat", not the user "root".

Is there anything that could be made accessible to other users than just root to allow the user tomcat as well?

Zhuoli, 2023-02-26:
> FYI, I just tested this today with the latest Pi4J-2.0-SNAPHOT build and it seems to be working on my RPi 4B (32-bit OS) system.
>
> I'm going to close this issue for now, but please re-open if any [...]

These 3 key notes help resolved my failure.

yannickkirschen, 2023-09-01:
Any updates on this? I encounter this issue by running the following simple code:

```java
Context pi4j = Pi4J.newAutoContext();

DigitalOutput led19 = pi4j.create(
    DigitalOutput
        .newConfigBuilder(pi4j)
        .id("pin-19")
        .name("Pin 19")
        .address(19)
        .shutdown(DigitalState.LOW)
        .initial(DigitalState.LOW)
        .provider("pigpio-digital-output"));
led19.high();
```

I set up my project according to https://github.com/Pi4J/pi4j-example-minimal and by running the `run.sh` script as sudo it works just fine.

MatthewUtzig, 2024-04-27:
Here is a workaround written in Kotlin to force load the PiGpio providers as long as they are present in the classpath. It works by calling private methods and fields and should only be used as a last resort. Thankfully, it only runs if the providers failed to successfully load in the first place.
```kotlin
fun Context.forceLoadPiGpio() {
    if(providers().all().keys.any { it.startsWith("pigpio") }) return
    val providers = this.providers() as DefaultProviders
    val runtimeProvidersField = providers.javaClass.getDeclaredField("providers")
    assert(runtimeProvidersField.trySetAccessible())
    val runtimeProviders = runtimeProvidersField.get(providers) as DefaultRuntimeProviders
    val addMethod = runtimeProviders.javaClass.getDeclaredMethod("add", java.util.Collection::class.java)
    assert(addMethod.trySetAccessible())
    val pigpio = PiGpioNativeImpl.newInstance()
    addMethod.invoke(runtimeProviders, listOf(
        PiGpioDigitalInputProviderImpl(pigpio),
        PiGpioSerialProviderImpl(pigpio),
        PiGpioPwmProviderImpl(pigpio),
        PiGpioSpiProviderImpl(pigpio),
        PiGpioDigitalOutputProviderImpl(pigpio),
        PiGpioI2CProviderImpl(pigpio)
    ))
}
```

eitch, 2024-04-29:
@yannickkirschen Since Version 2.5.1 we have a new provider for the GPIOs, called GpioD. This provider should now be autoselected and started and does not require sudo anymore. This new provider does not use PIGPIOD, as it is a rewrite using the standard Linux GPIO interface.

See here: https://pi4j.com/documentation/io-examples/digital-output/

Or here: https://pi4j.com/getting-started/minimal-example-application/

eitch, 2024-04-29:
@MatthewUtzig I don't see how your snippet will help, as if the application is not run with sudo, if the pigpiod provider is used, then access to the memory for the GPIOs is denied and your code will fail as well.
Your workaround is to add the provider to pi4j at runtime if it wasn't autoloaded. There are better ways of doing that, instead of reflection: 
https://pi4j.com/documentation/create-context/
https://pi4j.com/documentation/providers/
etc.

yannickkirschen, 2024-04-30:
Thank you @eitch! I tried out GpioD and it works perfectly 🚀