Pi4J/pi4j #71: I2C does not work

eitch, 2021-02-22:
With the following setup:

JARs built from main:

    pi4j-core-2.0-SNAPSHOT.jar
    pi4j-library-pigpio-2.0-SNAPSHOT.jar
    pi4j-plugin-pigpio-2.0-SNAPSHOT.jar
    pi4j-plugin-raspberrypi-2.0-SNAPSHOT.jar
    slf4j-api-1.7.30.jar
    slf4j-simple-1.7.30.jar

Script:

    import com.pi4j.Pi4J;
    import com.pi4j.context.Context;
    import com.pi4j.io.i2c.I2C;
    import com.pi4j.io.i2c.I2CConfig;
    import com.pi4j.io.i2c.I2CProvider;

    public class Tca9534Test {

      private static final byte TCA9534_REG_ADDR_OUT_PORT = 0x01;
      private static final byte TCA9534_REG_ADDR_CFG = 0x03;

      public static void main(String[] args) throws Exception {

        Context pi4j = Pi4J.newAutoContext();
        I2CProvider i2CProvider = pi4j.provider("pigpio-i2c");
        I2CConfig i2cConfig = I2C.newConfigBuilder(pi4j).id("TCA9534").bus(1).device(0x3f).build();
        try (I2C tca9534Dev = i2CProvider.create(i2cConfig)) {

          int config = tca9534Dev.readRegister(TCA9534_REG_ADDR_CFG);
          if (config < 0)
            throw new IllegalStateException(
                "Failed to read configuration from address 0x" + String.format("%02x", TCA9534_REG_ADDR_CFG));

          byte currentState = (byte) tca9534Dev.readRegister(TCA9534_REG_ADDR_OUT_PORT);

          if (config != 0x00) {
            System.out.println("TCA9534 is not configured as OUTPUT, setting register 0x" + String
                .format("%02x", TCA9534_REG_ADDR_CFG) + " to 0x00");
            currentState = 0x00;
            tca9534Dev.writeRegister(TCA9534_REG_ADDR_OUT_PORT, currentState);
            tca9534Dev.writeRegister(TCA9534_REG_ADDR_CFG, (byte) 0x00);
          }

          currentState = setPin(currentState, 8, tca9534Dev, true);
          Thread.sleep(500L);
          currentState = setPin(currentState, 8, tca9534Dev, false);
          Thread.sleep(500L);

          currentState = setPin(currentState, 7, tca9534Dev, true);
          Thread.sleep(500L);
          currentState = setPin(currentState, 7, tca9534Dev, false);
          Thread.sleep(500L);
        }
      }

      public static byte setPin(byte currentState, int pin, I2C tca9534Dev, boolean high) {
        byte newState;
        if (high)
          newState = (byte) (currentState | (1 << pin));
        else
          newState = (byte) (currentState & ~(1 << pin));

        System.out.println("Setting TCA9534 to new state " + asBinary(newState));
        tca9534Dev.writeRegister(TCA9534_REG_ADDR_OUT_PORT, newState);
        return newState;
      }

      public static String asBinary(byte b) {

        StringBuilder sb = new StringBuilder();

        sb.append(((b >>> 7) & 1));
        sb.append(((b >>> 6) & 1));
        sb.append(((b >>> 5) & 1));
        sb.append(((b >>> 4) & 1));
        sb.append(((b >>> 3) & 1));
        sb.append(((b >>> 2) & 1));
        sb.append(((b >>> 1) & 1));
        sb.append(((b >>> 0) & 1));

        return sb.toString();
      }
    }

Running:

    java -cp pi4j-core-2.0-SNAPSHOT.jar:pi4j-library-pigpio-2.0-SNAPSHOT.jar:pi4j-plugin-pigpio-2.0-SNAPSHOT.jar:pi4j-plugin-raspberrypi-2.0-SNAPSHOT.jar:slf4j-api-1.7.30.jar:slf4j-simple-1.7.30.jar Tca9534Test.java

The pi4j context fails with the following output:

    [main] INFO com.pi4j.Pi4J - New auto context
    [main] INFO com.pi4j.Pi4J - New context builder
    [main] WARN com.pi4j.library.pigpio.impl.PiGpioNativeImpl - PIGPIO ERROR: PI_INIT_FAILED; pigpio initialisation failed
    Exception in thread "main" java.lang.reflect.UndeclaredThrowableException
        at com.sun.proxy.$Proxy3.create(Unknown Source)
        at Tca9534Test.main(Tca9534Test.java:17)
    Caused by: java.lang.reflect.InvocationTargetException
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.base/java.lang.reflect.Method.invoke(Method.java:566)
        at com.pi4j.provider.impl.ProviderProxyHandler.invoke(ProviderProxyHandler.java:102)
        at com.sun.proxy.$Proxy3.create(Unknown Source)
        at Tca9534Test.main(Tca9534Test.java:17)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
        at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
        at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
        at java.base/java.lang.reflect.Method.invoke(Method.java:566)
        at jdk.compiler/com.sun.tools.javac.launcher.Main.execute(Main.java:404)
        at jdk.compiler/com.sun.tools.javac.launcher.Main.run(Main.java:179)
        at jdk.compiler/com.sun.tools.javac.launcher.Main.main(Main.java:119)
    Caused by: com.pi4j.library.pigpio.PiGpioException: PIGPIO ERROR: PI_INIT_FAILED; pigpio initialisation failed
        at com.pi4j.library.pigpio.impl.PiGpioBase.validateResult(PiGpioBase.java:265)
        at com.pi4j.library.pigpio.impl.PiGpioBase.validateResult(PiGpioBase.java:251)
        at com.pi4j.library.pigpio.impl.PiGpioNativeImpl.gpioInitialise(PiGpioNativeImpl.java:97)
        at com.pi4j.library.pigpio.PiGpio.initialize(PiGpio.java:146)
        at com.pi4j.plugin.pigpio.provider.i2c.PiGpioI2CProviderImpl.create(PiGpioI2CProviderImpl.java:62)
        at com.pi4j.plugin.pigpio.provider.i2c.PiGpioI2CProviderImpl.create(PiGpioI2CProviderImpl.java:43)
        ... 14 more

eitch, 2021-02-22:
I did this on the latest Raspberry Pi OS from 11. January 2020. Java:

    $ java -version
    openjdk version "11.0.10" 2021-01-19 LTS
    OpenJDK Runtime Environment (build 11.0.10+9-LTS)
    OpenJDK 32-Bit Server VM (build 11.0.10+9-LTS, mixed mode)

OS: 

    $ uname -a
    Linux raspberrypi 5.4.83-v7+ #1379 SMP Mon Dec 14 13:08:57 GMT 2020 armv7l GNU/Linux

eitch, 2021-02-22:
pigpio was installed as described on the website: http://abyz.me.uk/rpi/pigpio/download.html

eitch, 2021-02-22:
Dammit! Found the issue: The code needs to run as root...

How can this be changed? The device file has group i2c, so does my user:

    $ ls -la /dev/i2c-1 
    crw-rw---- 1 root i2c 89, 1 Feb 22 09:55 /dev/i2c-1

    $ groups
    pi adm dialout cdrom sudo audio video plugdev games users input netdev gpio i2c spi

hackerjimbo, 2021-02-22:
Hi Robert,

I'd recommend installing pigpio by using apt-get install pigpio. That 
way you get the supported version and it's updated along with the rest 
of the operating system by the usual apt-get update && apt-get upgrade.

The other issue may be that I2C hasn't been enabled. Are you seeing 
/dev/i2c-1 as a character special device?

Finally, are you running as root? It's a feature of pigpio that it needs 
to be run as root (something that I think needs to be changed a bit).

Regards,

Jim.

hackerjimbo, 2021-02-22:
Hi Robert,

I'm going to say that I'd view this as a misfeature on pigpio. To use 
/all/ of its features it needs to run as root but as a security 
professional I'm /very/ wary of this. A lot of applications, including 
most if not all of mine, /don't/ need the extra functionality only 
provided by running as root.

WiringPi, as used by pi4j-v1, had the ability to run as any user…

Regards,

Jim.

eitch, 2021-02-22:
Well yes i can of course install the version from the packages, but i am just playing around with the software and getting the latest main version might have offered something to mitigate my issues. Anyhow, installing from packages yields the same issue.

In my opinion we must find a solution to use the I2C bus without root permissions. This really doesn't make sense, as the i2c-tools work without root as well.

For this i have created an issue with pigpio: https://github.com/joan2937/pigpio/issues/432

Let's see what the developer says.

Personally i won't be able to upgrade to pi4j-v2 if root is required. I already had to write a workaround for PWM then, but that wasn't critical, so executing a binary with +s did the trick there, but not for the I2C bus.

FDelporte, 2021-02-22:
@eitch BTW this sudo-thing is also making things more complicated when trying to run from VSC on the Pi, see readme in https://github.com/Pi4J/pi4j-example-crowpi/blob/main/README.md

lukehutch, 2021-02-24:
Two things I had to do to get I2C and GPIO working that I don't think were documented (although this doesn't solve the need to run as root):

* `sudo nano /etc/config.txt` => add options: `dtparam=i2c_arm=on` to turn on I2C, and `dtparam=i2c_baudrate=1000000` to raise I2C datarate
* `sudo nano /etc/cmdline.txt` => add kernel option: `iomem=relaxed`, otherwise GPIO handler cannot access `/dev/mem`

eitch, 2021-06-09:
The code now works fine with the new linuxfs-i2c provider. Use the following dependencies:

            <dependency>
                <groupId>com.pi4j</groupId>
                <artifactId>pi4j-core</artifactId>
                <version>${pi4j.version}</version>
            </dependency>
            <dependency>
                <groupId>com.pi4j</groupId>
                <artifactId>pi4j-plugin-raspberrypi</artifactId>
                <version>${pi4j.version}</version>
            </dependency>
            <dependency>
                <groupId>com.pi4j</groupId>
                <artifactId>pi4j-plugin-linuxfs</artifactId>
                <version>${pi4j.version}</version>
            </dependency>

closed 2021-06-09 by eitch