Pi4J/pi4j #77: SPI configuration example

tatery, 2021-03-12:
Hello,

I'm looking for some example of SPI configuration where it is possible to select CS line and spiFlags as it is in pigpio: 
```
int spiOpen(unsigned spiChan, unsigned baud, unsigned spiFlags)
```
Current SPI implementation available as example does not allowed to set even CS line:
```java
        var config = Spi
                .newConfigBuilder(pi4j)
                    .id("my-spi-device")
                    .name("My SPI Device")
                    .address(SPI_CHANNEL)
                    .mode(SpiMode.MODE_0)
                    .baud(Spi.DEFAULT_BAUD)
                    .build();
```
I've found that SPI implementation I'm looking for is available in PiGpioNativeImpl. This is how I'm going to use it:
```java
piGpio = PiGpioNativeImpl.newInstance();
spiHandler = piGpio.spiOpen(1, 15600000, 0x0140);    //0x140 -> SPI AUX with BCM17 as CS

byte[] data = new byte[2];
data[0] = (byte)0x70;
data[1] = (byte)0x3C;
piGpio.spiWrite(spiHandler, data, 2);
```
but I'm not sure how it is connected with Context. Should I create context (var pi4j = Pi4J.newAutoContext()) before using SPI?

Thanks in advance for your support.

BTW: please consider to release pi4j v2 and let people to play with it. Thanks to this you will have users feedback and probably more pull-requests and maybe few more contributors in development.

tatery, 2021-03-15:
Above example with PiGpioNativeImpl is working without running pi4j context. Of course program has to be run as admin.
I'm still not sure if context is needed or not. Could you please give me some advice on this issue?

savageautomate, 2021-03-17:
@tatery 

> BTW: please consider to release pi4j v2 and let people to play with it. Thanks to this you will have users feedback and probably more pull-requests and maybe few more contributors in development.

We are working towards this goal.  We have just completed releasing Pi4J v1.3 and Pi4J v1.4 recently and now all efforts are focused on Pi4J v2.0.  However before an official release we need to complete the minimum viable feature-set and minimum architectural design goals.   Pi4J v2.0 compiled SNAPSHOTS are now available in the OSS Maven Repository and available here: https://github.com/pi4j/download/

savageautomate, 2021-03-17:
> Above example with PiGpioNativeImpl is working without running pi4j context. Of course program has to be run as admin.
I'm still not sure if context is needed or not. Could you please give me some advice on this issue?

If you are using the `pi4j-library-pigpio` directly, then there is no need for `Context`.  The `pi4j-library-pigpio` library is the RAW JNI wrapper library for the PiGPIO native library (http://abyz.me.uk/rpi/pigpio/).  Using the `pi4j-library-pigpio` directly is perfectly fine, but its not really using any of the Pi4J architecture or APIs.  

The ultimate goal is for Pi4J to provide encapsulated APIs that are independent of the implementation layer.  Even allowing you to change implementation providers without significantly modifying your codebase.

savageautomate, 2021-03-17:
> 'm looking for some example of SPI configuration where it is possible to select CS line and spiFlags as it is in pigpio:
`int spiOpen(unsigned spiChan, unsigned baud, unsigned spiFlags)`
Current SPI implementation available as example does not allowed to set even CS line:

        var config = Spi
                .newConfigBuilder(pi4j)
                    .id("my-spi-device")
                    .name("My SPI Device")
                    .address(SPI_CHANNEL)
                    .mode(SpiMode.MODE_0)
                    .baud(Spi.DEFAULT_BAUD)
                    .build();

It does not look like the Pi4J v2.0 APIs support direct configuration of `spiFlags` with the exception of `SPI MODE`.  What flag are you needing to modify?  We just have to make sure that the API we provide can be supported via multiple SPI provider implementations and no unique to just PIGPIO. 

Here are some additional details on `spiOpen(...)`: http://abyz.me.uk/rpi/pigpio/pdif2.html#spi_open

> where it is possible to select CS line

Do you mean you want to change to using the auxiliary SPI chip-select.  Are you needing to change `active-low` / `active-high` states for the chip select?  

I'm definitely not an expert in SPI .. so feel free to help educate me if I'm missing something.

Thanks, Robert

savageautomate, 2021-03-17:
@eitch ,

Adding you to this thread since you are working on the LinuxFS implementation of I2C/SPI providers.

tatery, 2021-03-17:
@savageautomate 

>What flag are you needing to modify?

I need to use auxiliary SPI with BCM17 as CS but current pi4j SPI API doesn't allow to select CS line (SPI0 has 2 CS lines, SPI1 has 3 CS lines). So to select CS line for SPI0/SPI1 Ux flags (3-bits) from pigpio should be available (in my opinion) in pi4j AP

> Are you needing to change active-low / active-high states for the chip select?

I do not need this. I have never seen chip that use CS active high but maybe there are some - I don't know.

tatery, 2021-03-18:
@savageautomate please have a look what is specified on raspberry pi website regarding SPI: https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md.

FDelporte, 2021-10-07:
There is an SPI example in the CrowPi project, see https://pi4j.com/documentation/io-examples/spi/

tuxedo0801, 2021-10-07:
That's only half of the truth. The configuration part is missing.

savageautomate, 2021-11-04:
I do like the idea of adding a bus (SpiBus) configuration parameter.   I was thinking (_perhaps I misunderstood_) that you needed more than just the bus configuration.   I had do do some digging to figure out why SpiBus was missing in the APIs and I think it's just a legacy issue that when porting the code to V2 was missed.  I don't think the V1 code ever supported alternate SPI busses probably due to original lack of support via WiringPi and later no alternate SPIDEV overlays ... way back when.  

I may leave the `flags` configuration parameter in place for someones need of advanced configuration bits.

In looking into the potential API  for a LinuxFS-based SPI implementation provider (using `spidevB.D` interfaces), it looks like the native `ioctl` API also supports a `request` parameter which are essentially flags as well.
```
int ioctl(int fd, unsigned long request, ...);
```
> The second argument is a device-dependent request code. 

REF: 
 - https://man7.org/linux/man-pages/man2/ioctl.2.html
 - https://www.kernel.org/doc/Documentation/spi/spidev

 I'll update PR https://github.com/Pi4J/pi4j-v2/pull/147 to add  SpiBus support.

savageautomate, 2021-11-04:
@tatery ,

OK, please take a look at the lasted changes in PR https://github.com/Pi4J/pi4j-v2/pull/147.  

I added the `bus` config parameter as well as `channel` and `chipSelect` (`channel` and `chipSelect` are both aliases of `address` added for greater clarity/easy code readability.)

```java
 // create SPI config
 var config  = Spi.newConfigBuilder(pi4j)
     .id("my-spi")
     .name("My SPI")
     .bus(SpiBus.BUS_1)                    <----- CONFIGURE SPI BUS
     .chipSelect(SpiChipSelect.CS_0)       <----- CONFIGURE SPI CS/ADDRESS/CHANNEL
     .mode(SpiMode.MODE_3)
     .build();

 // use try-with-resources to auto-close SPI when complete
 try (var spi = pi4j.spi().create(config)) {

     // write data using output stream
     spi.out().write(sample);
 }
```

Also, please check the flags implementation code (_linked below_).  I lifted most of it from your example, but please double check the bitwise operations.  My **bitwise-fu** is severely out of practice.   The goal was to accept the configured `flags` value then apply `mode` and `bus` bits to override/update the `flags` value.  (_just like your code did._)  

SEE:
 - https://github.com/Pi4J/pi4j-v2/pull/147/commits/beea5a556e069c324939635afff72b2e7f263c07#diff-680c211fdf3afe067275438d15d85243b1b28347f7d8d690d54877799c63ea8b

Thanks, Robert

closed 2022-05-01 by savageautomate