Pi4J/pi4j #83: 1.4 libs: java.lang.NoClassDefFoundError: com/pi4j/component/relay/Relay

kurt-lefevre, 2021-03-18:
I use the Pi4J libraries to control 2 relays on a PiFace Digital board. Up till version 1.3 everything worked fine.
With the 1.3 distro I need following libs: pi4j-core.jar, pi4j-device.jar and pi4j-gpio-extension.jar

I downloaded the 1.4 libs (pi4-distribution) from https://search.maven.org/search?q=com.pi4j and replaced the 1.3 libs with ones in the 1.4 zip.

I use JDK 11.

1.4 gives me following exception at startup:
**Exception in thread "main" java.lang.NoClassDefFoundError: com/pi4j/component/relay/Relay**
at be.relaycontrol.RelayControl.(Unknown Source)
at be.accesscontrol.AccessController.runController(Unknown Source)
at be.accesscontrol.AccessController.main(Unknown Source)
**Caused by: java.lang.ClassNotFoundException: com.pi4j.component.relay.Relay**
at java.base/jdk.internal.loader.BuiltinClassLoader.loadClass(BuiltinClassLoader.java:581)
at java.base/jdk.internal.loader.ClassLoaders$AppClassLoader.loadClass(ClassLoaders.java:178)
at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:522)

Any hints what I'm doing wrong?

Tx,
Kurt

savageautomate, 2021-03-18:
As of Pi4J v1.4 support was dropped for the component interfaces to which includes 'Relay'.  See the News/Announcements section on this page:  https://pi4j.com/1.4/

You could technically grab that part of v1.3 and include it in your own fork of v1.4 and compile it yourself ... but it's no longer supported or included by the project.

kurt-lefevre, 2021-03-19:
This obviuosly explains it. I'll have a look at the source code.

Thanks for the answer!