Pi4J/pi4j #108: Setting hardware PWM frequency on Raspberry Pi Zero W

MrAndeos, 2021-06-03:
Hello!
I just started using pi4j 2.0 in my project and came across this. I am using on of the pins on PWM channel 1 to drive a MOSFET that dimms a LED strip, I want it to be running at 100 kHz, but it seems that to achieve so I need to add two extra zeros in the frequency:
```java
displayLed = pi4j.create(Pwm.newConfigBuilder(pi4j)
                  .id("displayLed")
                  .pwmType(PwmType.HARDWARE)
                  .address(13)
                  .provider("pigpio-pwm")
                  .frequency(10000000)// Shouldn't this be 100 000?
                  .build());
displayLed.dutyCycle(80).on();
```
Also, the last line here actually doesn't turn the PWM output on, I need to call `displayLed.on();` later to enable it.

closed 2021-06-07 by MrAndeos

MrAndeos, 2021-06-07:
It turned out that it was the matter of me using a different lib for driving WS281x addressable LEDs that also interfaces with the PWM pins. It's nothing related to the Pi4J, sorry for creating the issue.

FDelporte, 2021-06-07:
@MrAndeos thanks for reporting the fix and using Pi4J! Always nice to see when a reported problem isn't actually caused by the library :-) I guess that is one of the returning challenges in soft+hardware projects: defining/debugging at which level the problem occurs.