Pi4J/pi4j #121: Trouble with I2C

rz259, 2021-07-29:
Hi,

I try to get a stripped-down version of the example for I2C from the PI4J site (https://pi4j.com/documentation/io-examples/i2c/) running.

The main method starts with the following statements:

```
		Context pi4j = Pi4J.newAutoContext();
		I2CProvider i2CProvider = pi4j.provider("linuxfs-i2c");

```
Unfortunately I get teh following exception when trying to get the I2CProvider:

```
Exception in thread "main" com.pi4j.provider.exception.ProviderNotFoundException: Pi4J provider [linuxfs-i2c] could not be found.  Please include this 'provider' JAR in the classpath.
        at com.pi4j@2.0-SNAPSHOT/com.pi4j.provider.impl.DefaultRuntimeProviders.get(DefaultRuntimeProviders.java:238)
        at com.pi4j@2.0-SNAPSHOT/com.pi4j.provider.impl.DefaultProviders.get(DefaultProviders.java:147)
        at com.pi4j@2.0-SNAPSHOT/com.pi4j.context.Context.provider(Context.java:240)
        at de.rziegaus.ATECC608A@1.00/de.rziegaus.SimpleATECC608A.main(SimpleATECC608A.java:18)
```

So I added the dependency for linuxfs-i2c, but now I get a different problem with the program. Now I get an exception saying

```
Error occurred during initialization of boot layer
java.lang.LayerInstantiationException: Package lib.armhf in both module com.pi4j.library.pigpio and module com.pi4j.library.linuxfs
```

If I remove the jar files concerning pigpio (library and plugin) manually everything works fine - at least I don't get any errors when I run the stripped-down version of the program.

What concerns me a bit - do I really have to choose between either pigpio and linuxfs-i2c? Or is there a workaround?

eitch, 2021-07-30:
Hi, did you add the linuxfs-ic2 plugin:

    <dependency>
        <groupId>com.pi4j</groupId>
        <artifactId>pi4j-plugin-linuxfs</artifactId>
        <version>2.0-SNAPSHOT</version>
    </dependency>

eitch, 2021-07-30:
In my project i use the following dependencies:

	<dependency>
	    <groupId>com.pi4j</groupId>
	    <artifactId>pi4j-core</artifactId>
	    <version>${pi4j.version}</version>
	</dependency>
	<dependency>
	    <groupId>com.pi4j</groupId>
	    <artifactId>pi4j-plugin-raspberrypi</artifactId>
	    <version>${pi4j.version}</version>
	</dependency>
	<dependency>
	    <groupId>com.pi4j</groupId>
	    <artifactId>pi4j-plugin-pigpio</artifactId>
	    <version>${pi4j.version}</version>
	</dependency>
	<dependency>
	    <groupId>com.pi4j</groupId>
	    <artifactId>pi4j-plugin-linuxfs</artifactId>
	    <version>${pi4j.version}</version>
	</dependency>

rz259, 2021-07-30:
This are the dependencies I use:

```
        <dependency>
            <groupId>org.slf4j</groupId>
            <artifactId>slf4j-api</artifactId>
            <version>${slf4j.version}</version>
        </dependency>
        <dependency>
            <groupId>org.slf4j</groupId>
            <artifactId>slf4j-simple</artifactId>
            <version>${slf4j.version}</version>
        </dependency>

        <!-- include Pi4J Core -->
        <dependency>
            <groupId>com.pi4j</groupId>
            <artifactId>pi4j-core</artifactId>
            <version>${pi4j.version}</version>
        </dependency>

        <!-- include Pi4J Plugins (Platforms and I/O Providers) -->
        <dependency>
            <groupId>com.pi4j</groupId>
            <artifactId>pi4j-plugin-raspberrypi</artifactId>
            <version>${pi4j.version}</version>
        </dependency>
        
        <!--  
        <dependency>
            <groupId>com.pi4j</groupId>
            <artifactId>pi4j-plugin-pigpio</artifactId>
            <version>${pi4j.version}</version>
        </dependency>
        -->
        
        <!-- include provider for linuxfs-i2c -->
		<dependency>
		    <groupId>com.pi4j</groupId>
		    <artifactId>pi4j-plugin-linuxfs</artifactId>
		    <version>${pi4j.version}</version>
		</dependency>        
```

As you can see, the plugin for pigpio is commented out and this way it works fine. When I remove the comments to include the pigpio-support, then I get the error message I described above

```
Error occurred during initialization of boot layer
java.lang.LayerInstantiationException: Package lib.armhf in both module com.pi4j.library.pigpio and module com.pi4j.library.linuxfs
```
To make the program work again, I have to do the following:

- remove or comment out the dependency in the POM-file
- remove the two jar files for the pigpio from the directory on my raspberry
- remove the two jar files for the pigpio from the target/lib-directory on development machine

As I also need to access gpio-ports I am looking for a working solution. I don't understand why this is not working.
Do I use the wrong modules? Maybe there are newer modules available that circumvent or remedy the problem?

Thanks for all help,

Rudi

eitch, 2021-07-30:
Hi, so you can use both pigpio and linuxfs plugins, but i think there is something wrong with the module configuration of the packages. Are you using modules? Do you need to use modules? I'll have to test this and find out what the problems with the modules is, but i'm not yet very versed in this affair...

eitch, 2021-07-30:
Do you by any change have a larger stack trace?

eitch, 2021-07-30:
@rz259 I've found a problem, where the native libraries are in the same directory in the jars for both plugins. I have now changed the directory for the linuxfs native libraries. Can you try again with the latest snapshot version?

rz259, 2021-07-30:
Hello eitch,

unfortunately I do not have a larger stack trace.

How can I download the latest snapshot version? I deleted the files from the raspberry and from the local computer and thus maven downloaded the versions - but unfortunately I can't tell you if they are the newest snapshot.

At any case the files that got downloaded by Maven do not solve the problems, so it would be nice if you could give me a hint how to download the latest snapshot.

Kind regards,

Rudi

eitch, 2021-08-02:
Hi Rudi. Can you check in your maven repository when 

    .m2/repository/com/pi4j/pi4j-library-linuxfs/2.0-SNAPSHOT/pi4j-library-linuxfs-2.0-SNAPSHOT.jar

was built?

Kind regards
Robert

rz259, 2021-08-02:
Hi Robert,

the jar was built on 2021-07-29 18.35 h (European Summertime).

Thanks for your help,

Rudi

rz259, 2021-08-06:
Ok, the problem is resolved. I have downloaded the latest snapshot and now it works.

Thanks for your help,

Rudi

closed 2021-08-06 by rz259

eitch, 2021-08-06:
Ahh, good to know! I was hoping to find some time this weekend to test it
out again. But now not necessarily.