Pi4J/pi4j #122: Use bnd-maven-plugin für better osgi support

stbischof, 2021-08-31:
* switch from felix to bnd-maven-plugin
* use Annotation support instead of instructions e.g. @Example
* Generate module-info files over annotations
* Add tests for osgi
I will come up with an PR

FDelporte, 2021-08-31:
looking forward to the PR! 
is OSGI still a thing? ;-)

stbischof, 2021-08-31:
oh yeah!!!!! it is!

FDelporte, 2021-08-31:
Just kidding about the OSGI. Really interested in where and how you use it. I used it myself at my last job but later refactored to Spring as it was a better fit for our use-case at that time.

stbischof, 2021-08-31:
is semantic versioning and [baselining](https://bnd.bndtools.org/chapters/180-baselining.html) an option for this project?

FDelporte, 2021-09-01:
Semantic seems OK to me. FYI we are still a bit struggling to fully automate the release cycle, especially the last step with bumping version number etc. As your change involves a lot of additional entries of the version number, can you also look into this issue? Develop branch should get 2.X.X-SNAPSHOT version and master the release version number.

FDelporte, 2021-09-01:
Regarding baselining: no experience, can you advise how/why to use and what it brings as improvements?

stbischof, 2021-09-01:
I added a Version on each exported package (2.0.0). When an other bundle-developer uses your bundle as dependency [bnd](https://github.com/bndtools/bnd) will generate the Import-Package statement in the MANIFEST.MF with an Version-Range from the Given version to the next mayor version. This way it is possible to run different Versions of the same bundle inside one framework.

The [concept](https://blog.osgi.org/2013/09/baselining-semantic-versioning-made-easy.html) of Baseline cares about the the correctness of this version. After building the jar the bnd-baselining-plugin compares the released version with the current build and tells you if the versions are correctly adjusted to the changes.

eitch, 2021-09-03:
@stbischof i still haven't properly understood baselining. Is it possible to auto-generate all these things, so that no fixed versions are in the code? Then i would find this really interesting.

About the semantic versioning. We of course want to use this. I guess i should have created a version 2.0.0, and not a 2.0. For the PR we can then create the 2.0.1 and then we will have a version which should be fine, right?

stbischof, 2021-09-03:
The focus is to use semantic version also on package level.

The baseline plugin will tell you if your chances are not conform with the used versiin of the package or jar.

eitch, 2021-09-03:
@stbischof is there no way to autogenerate the files? We use mvn versions:set to set the version and don't want to modify any files by hand. Is that possible for a release?

eitch, 2021-09-15:
Closing for now, as this conflicts with the release management process

closed 2021-09-15 by eitch