Pi4J/pi4j #127: Serial Port won't open

davideserra, 2021-09-13:
Hi averyone, 
I'm in migration phase of my project from Raspberry PI 3B+ on Raspberry 4B due "low System Memory" error in previous version. I buyed Raspberry PI4 8 Gbytes Version, with a 64 bit Operating System ad I migrate my NetBeans project. The appliance is an Radio Telescope with 1 Raspebrry PI4 as central Unit connected to 4 Arduino Due platform. In the main unit I has written in Java (with NetBeans) a main program that comunicate with the Arduino on COM port "/dev/ttyACMxxx" for read radio data and pilot he azimuth and altitude rotor. The oldest version works fine (until the memory problem). I download 64 bit version of pi4j, then set  the code as follow: 
        Rotorconfig = new SerialConfig();
        Rotorconfig.device(RotorPortName);
        Rotorconfig.baud(Baud._57600);
        Rotorconfig.dataBits(DataBits._8);
        Rotorconfig.parity(Parity.NONE);
        Rotorconfig.stopBits(StopBits._1);
        Rotorconfig.flowControl(FlowControl.NONE);        
        RotorPort.open(Rotorconfig);

I have also a listener to serial eventa but this happen after the port open (as lik as previosu version)
        Rotor = new RotorControl(RotorPortName);
        Rotor.RotorPort.addListener((SerialDataEventListener) new SerialDataEventListener() {
            @Override
            public void dataReceived(SerialDataEvent sde) {
                String Data = "";
                try 
                {
                    Data = sde.getAsciiString();
                    int AziPos = Data.indexOf("AZI");
                    int AltPos = Data.indexOf("ALT");
                    if(AziPos >= 0){
                        try{
                            int EndData = Data.indexOf(";", AziPos);
                            if(EndData > AziPos){
                                String Azimuth = Data.substring(AziPos + 5, EndData);    
                                //jTextField1.setText(Azimuth);
                                jTextField1.setText(Rotor.GetAzimuthValue(Azimuth));
                                saveProperties();
                                if(AzimuthGo){
                                    int Delta = Math.abs(Integer.valueOf(AzimuthToGo) - Integer.valueOf(Azimuth));
                                    if(Delta < 2) {
                                        AzimuthGo = false;
                                        Rotor.AzimuthStop();
                                        loadProperties();
                                    }                            
                                }
                            }
                        }catch(Exception ex){
                            Logger.getLogger(Main.class.getName()).log(Level.SEVERE, null, ex);
                        }
                    }
                    if(AltPos >= 0){
                        try{
                            int EndData = Data.indexOf(";", AltPos);                        
                            if(EndData > AltPos){
                                String Altitude = Data.substring(AltPos + 5, EndData);    
                                //jTextField2.setText(Altitude);
                                jTextField2.setText(Rotor.GetAltitudeValue(Altitude));
                                saveProperties();
                                if(AltitudeGo){
                                    int Delta = Math.abs(Integer.valueOf(AltitudeToGo) - Integer.valueOf(Altitude));
                                    if(Delta < 2) {
                                        AltitudeGo = false;
                                        Rotor.AltitudeStop();
                                        loadProperties();
                                    }
                                }    
                            }
                        }catch(Exception ex){
                            Logger.getLogger(Main.class.getName()).log(Level.SEVERE, null, ex);
                        }
                    }
                    if(AzimuthGo || AltitudeGo){
                        jButton16.setEnabled(false);
                    }
                } catch (Exception ex) {
                    Logger.getLogger(Main.class.getName()).log(Level.SEVERE, null, ex);
                }
            }
        });   

But on the statement "RotorPort.open(Rotorconfig);"  that doesn't work and return this exception: 

Exception in thread "AWT-EventQueue-0" java.lang.UnsatisfiedLinkError: 'int com.pi4j.jni.Serial.open(java.lang.String, int, int, int, int, int)'
	at com.pi4j.jni.Serial.open(Native Method)
	at com.pi4j.io.serial.impl.SerialImpl.open(SerialImpl.java:149)
	at com.pi4j.io.serial.impl.SerialImpl.open(SerialImpl.java:319)
	at skywalker.RotorControl.InizializeRotor(RotorControl.java:55)
	at skywalker.RotorControl.<init>(RotorControl.java:38)
	at skywalker.Main.<init>(Main.java:298)
	at skywalker.Main.lambda$main$1(Main.java:4073)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:740)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

Anyone can help me? Thanks very much!

davideserra, 2021-09-13:
Following the error I found: 
"Exception in thread "AWT-EventQueue-0" java.lang.UnsatisfiedLinkError: Can't load library: /home/pi/NetBeansProject/SkyWalker/lib/**libpi4j.so**"

Someone can tell to me wat's that's libpi4j.so? I doesn't find this library in download package.

I didn't install library with provided installer "pi4j-1.4.deb" due to error: 
./pi4j-1.3-SNAPSHOT.deb: riga 1: errore di sintassi vicino al token non atteso "newline"
./pi4j-1.3-SNAPSHOT.deb: riga 1: `!<arch>'

I download zip file "pi4j-1.4.zip", uncompressed and include in my project NetBeans library. The program compile fine without errors but run time compiler search that library and doesn't found.

FDelporte, 2021-09-13:
which version of pi4j are you using in your project?

davideserra, 2021-09-13:
Thansk for reply: I buy this version:
https://www.amazon.it/gp/product/B0899VXM8F/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1

davideserra, 2021-09-13:
oh sorry: pi4j version: https://pi4j.com/download/pi4j-1.4.zip