Pi4J/pi4j #128: Serial Communication with the PI and Arduino

RobinQuante, 2021-09-14:
Can you add an Exampe for the Arduino and the Pi with a USB oder Serial Bus.
I have read it will work, but dont know how.

FDelporte, 2021-09-14:
You're right, serial has no example yet on https://pi4j.com/documentation/io-examples/serial/
Will try to create something asap...

tuxedo0801, 2021-09-24:
Need the same here... Tried to test it by myself, but serial communication is simply not working. Example would help.

FDelporte, 2021-10-06:
Working on it...

tuxedo0801, 2021-10-06:
Maybe it helps when I provide my experiments? I'm trying to read a temperature value that is provided via serial port, line by line:

```java
        SerialConfig serialConfig = SerialConfigBuilder.newInstance(pi4j)
                .use_115200_N81()
                .name("RS232")
                .id("RS232")
                .device("/dev/ttyUSB0")
                .flowControl(FlowControl.NONE)
                .build();
        temp = pi4j.create(serialConfig);
        // ...
        temp.open();
        // ...
        while (!temp.isOpen()) {
            System.out.println("Waiting for port to open");
            Thread.sleep(1000);
        }
        BufferedReader br = new BufferedReader(new InputStreamReader(temp.getInputStream()));
        
        String line = "";
        while (line != null) {
            line = br.readLine();
            System.out.println("Temp: "+line);
        }                
```

Problem is: Port never opens. while-loop never exits (no error message).

tuxedo0801, 2021-10-06:
Hmm, maybe the "provider" is missing?

So I added 

```java
                .provider("pigpio-serial")
```

to the SerialConfigBuilder execution. 
Now I get a different error:

```
11:33:43 [main] INFO com.pi4j.util.Console - Error: Underlying input stream returned zero bytes
java.io.IOException: Underlying input stream returned zero bytes
	at java.base/sun.nio.cs.StreamDecoder.readBytes(StreamDecoder.java:288)
	at java.base/sun.nio.cs.StreamDecoder.implRead(StreamDecoder.java:326)
	at java.base/sun.nio.cs.StreamDecoder.read(StreamDecoder.java:178)
	at java.base/java.io.InputStreamReader.read(InputStreamReader.java:181)
	at java.base/java.io.BufferedReader.fill(BufferedReader.java:161)
	at java.base/java.io.BufferedReader.readLine(BufferedReader.java:326)
	at java.base/java.io.BufferedReader.readLine(BufferedReader.java:392)
```

Looks like "opening" is now done, but reading fails.

Also tried with 
```
.provider("raspberrypi-serial")
```
..same effect as without a provider.

FDelporte, 2021-10-06:
thanks for the valuable feedback!! will try out myself (but not today)

tuxedo0801, 2021-10-06:
More details:

https://github.com/Pi4J/pi4j-v2/blob/develop/plugins/pi4j-plugin-pigpio/src/main/java/com/pi4j/plugin/pigpio/provider/serial/PiGpioSerial.java#L74

The value of "handle" decides about successful port opening. But value is not checked. See also: http://abyz.me.uk/rpi/pigpio/cif.html#serOpen

I guess there is some issue in my case, but debugging on pi currently works not as expected. Maybe my Netbeans IDE has an issue. I can step through the stack, but not check the source and values at the same time.

Will further debug...

tuxedo0801, 2021-10-06:
Finally it works... but it's a bit different.. It's not a blocking-read... You have to check with "temp.available() > 0" if there is some data to read. 

I'm not finally sure if my code is fail-proof. Will check a bit more, but after that, I can provide a hopefully working example.

FDelporte, 2021-10-06:
Thanks a lot!! This is really great and reduces my work load ;-)
The example project I started is uploaded to https://github.com/Pi4J/pi4j-example-serial now, will use your example to further extend it and add documentation to the website.

FDelporte, 2021-10-07:
@tuxedo0801 thanks for all your feedback!!!

I still need to fully document the example I created, but it works perfectly now to read data from a GPS receiver. I changed a bit the reader in the runnable to process lines as you can see in https://github.com/Pi4J/pi4j-example-serial/blob/main/src/main/java/com/pi4j/example/SerialReader.java#L35

Data is now received like this (GPS antenna inside, so no real coordinates available):

```
[SerialReader] INFO com.pi4j.util.Console - Data: '$GPRMC,,V,,,,,,,,,,N*53'
[SerialReader] INFO com.pi4j.util.Console - Data: '$GPVTG,,,,,,,,,N*30'
[SerialReader] INFO com.pi4j.util.Console - Data: '$GPGGA,,,,,,0,00,99.99,,,,,,*48'
[SerialReader] INFO com.pi4j.util.Console - Data: '$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30'
[SerialReader] INFO com.pi4j.util.Console - Data: '$GPGSV,1,1,01,20,,,28*70'
[SerialReader] INFO com.pi4j.util.Console - Data: '$GPGLL,,,,,,V,N*64'
[SerialReader] INFO com.pi4j.util.Console - Data: '$GPRMC,,V,,,,,,,,,,N*53'
[SerialReader] INFO com.pi4j.util.Console - Data: '$GPVTG,,,,,,,,,N*30'
[SerialReader] INFO com.pi4j.util.Console - Data: '$GPGGA,,,,,,0,00,99.99,,,,,,*48'
[SerialReader] INFO com.pi4j.util.Console - Data: '$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30'
```

FDelporte, 2021-10-18:
thanks @tuxedo0801 for all your feedback, has been used for the example I created and documented on https://pi4j.com/documentation/io-examples/serial/