Pi4J/pi4j #129: Pi4J provider could not be found

tommyzat, 2021-09-18:
Hello!

I'm trying to build a JAR artifact through IntelliJ, but everytime I try to run it in Pi I get this:

```
Exception in thread "main" com.pi4j.provider.exception.ProviderNotFoundException: Pi4J provider [pigpio-digital-input] could not be found.  Please include this 'provider' JAR in the classpath.
	at com.pi4j.provider.impl.DefaultRuntimeProviders.get(DefaultRuntimeProviders.java:238)
	at com.pi4j.provider.impl.DefaultProviders.get(DefaultProviders.java:147)
	at com.pi4j.provider.Providers.get(Providers.java:253)
	at com.pi4j.context.Context.create(Context.java:316)
	at com.pi4j.internal.IOCreator.create(IOCreator.java:58)
	at com.pi4j.internal.IOCreator.create(IOCreator.java:106)
	at com.pi4j.internal.IOCreator.create(IOCreator.java:186)
	at bumtzihouse.pi.gpio.GpioAccess.configureFrontButton(GpioAccess.kt:49)
	at bumtzihouse.pi.gpio.GpioAccess.init(GpioAccess.kt:18)
	at bumtzihouse.pi.Application$main$1.invokeSuspend(Application.kt:18)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at kotlinx.coroutines.EventLoopImplBase.processNextEvent(EventLoop.common.kt:274)
	at kotlinx.coroutines.BlockingCoroutine.joinBlocking(Builders.kt:85)
	at kotlinx.coroutines.BuildersKt__BuildersKt.runBlocking(Builders.kt:59)
	at kotlinx.coroutines.BuildersKt.runBlocking(Unknown Source)
	at kotlinx.coroutines.BuildersKt__BuildersKt.runBlocking$default(Builders.kt:38)
	at kotlinx.coroutines.BuildersKt.runBlocking$default(Unknown Source)
	at bumtzihouse.pi.Application.main(Application.kt:14)
```

FDelporte, 2021-09-18:
Are you trying to create a "fat jar" which includes all dependencies? Are modules as described on https://pi4j.com/getting-started/minimal-example-application/?

tommyzat, 2021-09-18:
https://imgur.com/fjdsAdd

Having those at the bottom surely means the dependencies were added, no?

eitch, 2021-09-19:
Looking at that screenshot, it looks like you are extracting all your dependencies into one large JAR. This currently does not work with pi4j-v2 due to a facility to automatically lookup which services and plugins are available.

You will need to keep your libraries in a separate folder and add that folder to your class or module path

tommyzat, 2021-09-20:
So is it possible to have a fat JAR at all? I'm trying to compile something and then just run it on my RPi. I'm not sure how you suggested that I should do...

FDelporte, 2021-09-20:
@tommyzat can you take a look at the description on https://pi4j.com/getting-started/minimal-example-application/ and the pom file in https://github.com/Pi4J/pi4j-example-minimal/blob/main/pom.xml. This produces all required files to run the application on Raspberry Pi in target/distribution, including a run.sh file which contains the correct java start command.

Regarding the "fat jar" approach: we are looking how we can support this or add more info to the site.

tommyzat, 2021-09-20:
I understand now. Thank you for your explanation - I'm eagerly waiting for the fat jar option!

closed 2021-09-20 by tommyzat

FDelporte, 2021-09-20:
@tommyzat thanks for the feedback!

michal-kroliczek, 2021-12-28:
Hi. Any update on this?
I got new Pi and followed steps I did usually, then I saw Pi4j-2. 
To be honest that is quite disappointing, that having all as maven dependencies it is not possible to use fat jar. 
That should be resolved if you guys want to follow sentence on your site:
"Pi4J version 2.0 brings with it many new features, and an entirely new architecture that focuses on extensibility, simplified integration and a modern Java API including the following... " 

What is exactly the issue with fat jar approach? Having target folder with all the libs seems like building app with Ant 10 years ago :(

FDelporte, 2022-02-22:
FAT Jar example has been described on https://pi4j.com/documentation/building/fat-jar/

tscholze, 2023-11-15:
Has anyone got it working using Kotlin, Gradle and ShadowJar? :) 
It seems the custom module run command from `run.sh` seems not to work with Kotlin.

FDelporte, 2023-11-15:
@tscholze have you taken a look at the Kotlin DSL documentation? 
https://pi4j.com/kotlin/

tscholze, 2023-11-15:
@FDelporte yes and thanks for the Kotlin DSL, this is amazing!
I just have the same problem as the issue creator, I created a fat-tar using Shadow Gradle plugin.

That's why I thought I need to have more than just a `java -jar foo.jar`. :)

FDelporte, 2023-11-15:
Sorry, I can't help you with the Kotlin side, but a fat-jar worked as described on https://pi4j.com/documentation/building/fat-jar/

Regarding Kotlin, all thanks to @mhashim6 who created the Pi4J DSL!

tscholze, 2023-11-15:
@FDelporte because now I get the error `PIGPIO ERROR: PI_INIT_FAILED; pigpio initialisation failed`, I think it works in regards of the fat jar. Or could this mean that this is still not working? Yes I use the pin "6" which should work. :)

The solving part was this:
```
tasks.named("shadowJar", ShadowJar::class.java) {
    mergeServiceFiles() // <--
    manifest.attributes["Main-Class"] = "MainKt"
}
```