Pi4J/pi4j #161: Digital Output state change does not change pin voltage

camueller, 2021-11-13:
Somehow I'm unable to simply change the state of BCM26 to HIGH using the following code (already taking into accoount issue #137):
```
int pin = 26;
Context context = Pi4J.newAutoContext();
DigitalOutputConfig config = DigitalOutput.newConfigBuilder(context)
        .id("BCM" + pin)
        .name("Tester")
        .address(pin)
        .provider(RaspberryPi.DIGITAL_OUTPUT_PROVIDER_ID)
        .build();
DigitalOutput output = context.create(config);
output.state(DigitalState.HIGH);
```
The code runs without error. I even debugged into Pi4J - the state is changed there but voltage between BCM26 and GND remains 0V.

Initially I used a simpler code base on the documentation:

```
int pin = 26;
Context context = Pi4J.newAutoContext();
DigitalOutput output = context.digitalOutput().create(pin);
output.state(DigitalState.HIGH);
```

perryp, 2021-12-19:
I have the same (very fundamental) problem. 
Will there a fix/workaround be available soon?

FDelporte, 2021-12-20:
I will try to reproduce this problem on my side. Can you confirm you have the same problem
* on other pins?
* with `provider("pigpio-digital-input")`?

Would you also be able to provide the output of
```
        PrintInfo.printLoadedPlatforms(console, pi4j);
        PrintInfo.printDefaultPlatform(console, pi4j);
        PrintInfo.printProviders(console, pi4j);
```

See https://github.com/Pi4J/pi4j-example-minimal/blob/main/src/main/java/com/pi4j/example/PrintInfo.java

camueller, 2021-12-20:
BTW: I'm using Pi4J version 2.1.0.

> on other pins?
Tried BCM17 (aka WPI0) and BCM11 (aka WPI14) with same result.

> with provider("pigpio-digital-input")
Produces `com.pi4j.provider.exception.ProviderIOTypeException: Pi4J provider IO type mismatch for [raspberrypi-digital-input(DIGITAL_INPUT)]; provider instance is not of IO type [DIGITAL_OUTPUT]`

> PrintInfo.printLoadedPlatforms(console, pi4j);
```
PLATFORMS: [1] "Pi4J Runtime Platforms" <com.pi4j.platform.impl.DefaultPlatforms>
�~T~T�~T~@PLATFORM: "RaspberryPi Platform" {raspberrypi} <com.pi4j.plugin.raspberrypi.platform.RaspberryPiPlatform> {Pi4J Platform for the RaspberryPi series of products.}
```
> PrintInfo.printDefaultPlatform(console, pi4j);
```
PLATFORM: "RaspberryPi Platform" {raspberrypi} <com.pi4j.plugin.raspberrypi.platform.RaspberryPiPlatform> {Pi4J Platform for the RaspberryPi series of products.}
```
> PrintInfo.printProviders(console, pi4j);
```
PROVIDERS: [6] "I/O Providers" <com.pi4j.provider.impl.DefaultProviders>
�~T~\�~T~@I2C: [1] <com.pi4j.io.i2c.I2CProvider>
�~T~B �~T~T�~T~@PROVIDER: "RaspberryPi I2C Provider" {raspberrypi-i2c} <com.pi4j.plugin.raspberrypi.provider.i2c.RpiI2CProviderImpl> {com.pi4j.plugin.raspberrypi.provider.i2c.RpiI2CProviderImpl}
�~T~\�~T~@DIGITAL_OUTPUT: [1] <com.pi4j.io.gpio.digital.DigitalOutputProvider>
�~T~B �~T~T�~T~@PROVIDER: "RaspberryPi Digital Output (GPIO) Provider" {raspberrypi-digital-output} <com.pi4j.plugin.raspberrypi.provider.gpio.digital.RpiDigitalOutputProviderImpl> {com.pi4j.plugin.raspberrypi.provider.gpio.digital.RpiDigitalOutputProviderImpl}
�~T~\�~T~@SERIAL: [1] <com.pi4j.io.serial.SerialProvider>
�~T~B �~T~T�~T~@PROVIDER: "RaspberryPi Serial Provider" {raspberrypi-serial} <com.pi4j.plugin.raspberrypi.provider.serial.RpiSerialProviderImpl> {com.pi4j.plugin.raspberrypi.provider.serial.RpiSerialProviderImpl}
�~T~\�~T~@SPI: [1] <com.pi4j.io.spi.SpiProvider>
�~T~B �~T~T�~T~@PROVIDER: "RaspberryPi SPI Provider" {raspberrypi-spi} <com.pi4j.plugin.raspberrypi.provider.spi.RpiSpiProviderImpl> {com.pi4j.plugin.raspberrypi.provider.spi.RpiSpiProviderImpl}
�~T~\�~T~@DIGITAL_INPUT: [1] <com.pi4j.io.gpio.digital.DigitalInputProvider>
�~T~B �~T~T�~T~@PROVIDER: "RaspberryPi Digital Input (GPIO) Provider" {raspberrypi-digital-input} <com.pi4j.plugin.raspberrypi.provider.gpio.digital.RpiDigitalInputProviderImpl> {com.pi4j.plugin.raspberrypi.provider.gpio.digital.RpiDigitalInputProviderImpl}
�~T~\�~T~@ANALOG_INPUT: [0] <com.pi4j.io.gpio.analog.AnalogInputProvider>
�~T~\�~T~@ANALOG_OUTPUT: [0] <com.pi4j.io.gpio.analog.AnalogOutputProvider>
�~T~T�~T~@PWM: [1] <com.pi4j.io.pwm.PwmProvider>
  �~T~T�~T~@PROVIDER: "RaspberryPi PWM Provider" {raspberrypi-pwm} <com.pi4j.plugin.raspberrypi.provider.pwm.RpiPwmProviderImpl> {com.pi4j.plugin.raspberrypi.provider.pwm.RpiPwmProviderImpl}
```

perryp, 2021-12-22:
In my case there was a very simple solution: running the app as root with sudo solved the problem.

camueller, 2021-12-22:
That's not really an option and it was not required using Pi4J 1.x with `wiringpi`. If it works with `root` usually the user is missing a group `root` has. I will test if it works with root.

FDelporte, 2021-12-22:
(at this moment) sudo is indeed needed for PiGpio, I added as remark at bottom of page https://pi4j.com/getting-started/minimal-example-application

But in this case, the error message seems to be different...

camueller, 2021-12-22:
Running the Java process as `root` did not change anything. If there are insufficiant privileges this should result in an exception. But as I already stated, the code runs without any error. I simply does not change the pin state.

camueller, 2021-12-22:
I debugged it: In the `EventManager` there are no `listeners` registered - therefore the `forEach` in the code below is simply skipped:
```
   public SOURCE_TYPE dispatch(EVENT_TYPE event) {
        this.listeners.forEach((listener) -> {
            try {
                this.delegate.dispatch(listener, event);
            } catch (Exception var4) {
                this.logger.error(var4.getMessage(), var4);
            }

        });
        return this.source;
    }
}
```

camueller, 2021-12-22:
I did not have `pi4j-library-pigpio` added to my project. However, even with this library added, `listeners` is still empty.
These are the Pi4J dependencies I inluded:
```
		<dependency>
			<groupId>com.pi4j</groupId>
			<artifactId>pi4j-core</artifactId>
			<version>2.1.0</version>
		</dependency>
		<dependency>
			<groupId>com.pi4j</groupId>
			<artifactId>pi4j-plugin-raspberrypi</artifactId>
			<version>2.1.0</version>
		</dependency>
		<dependency>
			<groupId>com.pi4j</groupId>
			<artifactId>pi4j-library-pigpio</artifactId>
			<version>2.1.0</version>
		</dependency>
```

camueller, 2021-12-22:
In the meantime I found your repository `pi4j-example-minimal` - I was hoping for an example like this but I did not find a reference to it anywhere. I built it and transfered all files in `distribution` to the Raspberry Pi. The example runs without errors if started as `root` and BCM 22 changes periodically as announced on the console.

From the example I realized, that it contains **`pi4j-library-pigpio-2.0.jar` - I missed this dependency as well** as you can see from my last post. **I don't think that Pi4J should run without errors if half of its dependencies are missing.** In your reply to my first post you asked for the output of some descriptors but obviously you have been unable to conclude the missing dependencies.

After I added `pi4j-library-pigpio` to the dependencies I copied the digital output related code from `MinimalExample.java` contained in `pi4j-example-minimal` to my code:
```
                    Context pi4j = Pi4J.newAutoContext();
                    DigitalOutputConfigBuilder ledConfig = DigitalOutput.newConfigBuilder(pi4j)
                            .id("led")
                            .name("LED Flasher")
                            .address(pin)
                            .shutdown(DigitalState.LOW)
                            .initial(DigitalState.LOW)
                            .provider("pigpio-digital-output");
                    DigitalOutput led = pi4j.create(ledConfig);
                    led.high();
```
The code changes BCM 22 to HIGH. Obviously something is different in this code compared to the code in my initial post.
It is the **provider which "magically" has to be `pigpio-digital-output`** whereas I used `RaspberryPi.DIGITAL_OUTPUT_PROVIDER_ID`. The latter translates to `raspberrypi-digital-output`. I did not find constant with the value of `pigpio-digital-output` in the whole code base.

**Bottom line: For Pi4J 2.x you pursued a modular approach, but now there are multiple deployment units (jars) and so many things to wire during set-up that even you are unable to point to the error in just 10 lines of code in my initial posting.**

FDelporte, 2021-12-23:
@camueller I find it really a pity you didn't find the minimal example before as it is part of the "getting started with Pi4J" of our website pi4j.com, so I clearly did something wrong with the documentation website...

Your remark about the unclarity of the newAutoContext is correct. We have taken "the modular approach" for V2 to be in sync with how the "Java world" is evolving, but we need to rethink this approach. I started a discussion for this: https://github.com/Pi4J/pi4j-v2/discussions/178. Any valuable feedback, ideas, code improvement proposals are welcome!

_Sidenote: this project is an open-source "pet project" for the three main contributors. As such we can't provide any "real-time" support but do what's possible besides our professional activities to get as many people as possible interested in and started with Java on the Raspberry Pi._