Pi4J/pi4j #162: Create a Spring example application via V2

FDelporte, 2021-11-15:
An example is available with V1 on github.com/FDelporte/JavaOnRaspberryPi/tree/master/Chapter_10_Spring, but not yet with V2

FDelporte, 2022-02-02:
Can be used to help users like in this discussion
https://github.com/Pi4J/pi4j-v2/discussions/192

StefanSchubert, 2022-03-10:
If I got https://github.com/Pi4J/pi4j-v2/discussions/203 solved, I will rewrite also https://github.com/StefanSchubert/aquarium_IoT for v2 which might be listened as spring-boot sample application if you like too.

FDelporte, 2022-03-10:
Would be great! Looking forward to any contribution to illustrate how Pi4J is used. Could even be added to https://pi4j.com/featured-projects/

BTW - and this can't be a coincidence - I'm at this same moment working on a Pi4J + Spring + Vaadin example application and can confirm it works! I hope to have it fully documented in the next week(s): https://github.com/FDelporte/Vaadin-examples

StefanSchubert, 2022-03-11:
I took a look into your pom on the vaadin example, if this is working, then the maven shade plugin is no longer required (from the special fat jar description) as you are not relying on it.

Which would be great as it complains a lot on breaking jigsaw module.infos capsulations.
If we don’t need it anymore I would recommend to throw it from the fat client requirements list ;-)