Pi4J/pi4j #163: Ktor and Pi4j2

tommyzat, 2021-11-20:
Hello, I'm trying to run Pi4j with Ktor, however Ktor does not support modular java builds (https://github.com/ktorio/ktor/issues/1137) and pi4j does not support fat jars.

I am stuck and don't know what to do. How can I have them both run and built on a raspberry pi?

Thank you!

FDelporte, 2021-11-20:
Correct Pi4J doesn't support fat jars, we are still looking how/when this could be achieved.

On the other hand, in the ktor issue it is clear they are "stuck" with an outdated Java version. In my opinion, that's more critical but I understand that's not an easy fix either...

Any ideas or contributions to make Pi4J fat-jar-friendly are welcome!!!

FDelporte, 2022-01-12:
FAT JAR can be created!

Full info on: https://pi4j.com/getting-started/minimal-example-application-fatjar/
More detailed FAT JAR info on: https://pi4j.com/documentation/building/fat-jar/

closed 2022-01-12 by FDelporte

tommyzat, 2022-01-12:
That's great news, thank you!

I'd also like to point out for any lurkers that Ktor will support modules in 2.0
https://youtrack.jetbrains.com/issue/KTOR-619

So you can choose from both methods now when using Ktor.