Pi4J/pi4j #185: No temporary memory allocation in ByteBuffer APIs

FDelporte, 2022-01-06:
Reported by @haumacher

APIs using ByteBuffers are very convenient - much better than (byte[], off, len) APIs. Fortunately, pi4j supports such APIs, but unfortunately, it is currently not possible to create memory-efficient code using these APIs. Internally, temporary byte arrays are allocated for communication (instead of using the ByteBuffers internal array). When communicating with devices such as gyroscopes and accelerometers, a high frequency of calls is necessary to transfer data from the device. If each call allocates temporary memory, this produces a high pressure on the PIs GC, potentially resulting in GC pauses on the device.

The full code will need to be reviewed to make optimal use of ByteBuffers.

haumacher, 2022-02-05:
Ok I'll see whether I can make a more detailed proposal for an optimization.

FDelporte, 2022-06-27:
Dear @haumacher, were you able to further look into this issue?

closed 2022-08-25 by eitch