Pi4J/pi4j #207: Can't Create 2 serial connections

camaro4life18, 2022-03-25:
Hello, I'm new to Pi4J and serial communication. I have a Pi4 connected to 2 different Integrated Circuits for monitoring pH and ORP of a pool. I connected them to UART1 and UART5. I can read from both individually in my Java app, but if I try to create 2 different objects, I get this error....

Exception in thread "main" com.pi4j.io.exception.IOAlreadyExistsException: IO instance [com.pi4j.plugin.pigpio.provider.serial.PiGpioSerial] already exists in the Pi4J runtime context; unable to create a new instance using this reserved id.
        at com.pi4j.registry.impl.DefaultRuntimeRegistry.add(DefaultRuntimeRegistry.java:80)
        at com.pi4j.provider.impl.ProviderProxyHandler.invoke(ProviderProxyHandler.java:106)
        at com.sun.proxy.$Proxy3.create(Unknown Source)
        at com.pi4j.context.Context.create(Context.java:317)
        at com.pi4j.internal.IOCreator.create(IOCreator.java:58)
        at com.pi4j.internal.IOCreator.create(IOCreator.java:146)
        at com.camaro4life18.pool.automation.EzoProbesUart.<init>(EzoProbesUart.java:11)
        at com.camaro4life18.pool.automation.Main.main(Main.java:33)

See my code here where I try to create 2 serial device objects. Lines 32-33
https://github.com/camaro4life18/pool-automation/blob/development/PoolAutomation/src/main/java/com/camaro4life18/pool/automation/Main.java

camaro4life18, 2022-03-28:
I decided to develop a micro-service to handle the Serial communication individually between my 2 devices, and discovered that the Pi4j java libraries are supporting the limitations of the linux pigpio libraries. I ran into exceptions when trying to run 2 different java applications accessing the Serial connections. I'm not sure why there is a limitation of only allowing 1 serial connection, as I can communicate to both devices in Python.

I may explore doing the serial communication myself using other Serial libraries since Pi4j v2 doesn't seem to support it.

camaro4life18, 2022-03-29:
Anyone reading this with the same issue, I ended up abandoning this library for Serial communication. I'm still using the DigitalOutput to control some relays, but for serial communication I had a lot of luck with jSerialComm. It works great on the RPI, without having to install any OS libraries....and I can create more than 1 Serial connection.
https://fazecast.github.io/jSerialComm/

eitch, 2024-09-17:
We recommend using jSerialComm for serial connections

closed 2024-09-17 by eitch