Pi4J/pi4j #211: pigpio does not work under recent RPi OS - Release: April 4th 2022,  64-bit, Kernel: 5.15 Debian: 11 (bullseye)

elmot, 2022-04-13:
I tried to run standard minimalistic example and it did not work for regular user, but works for root.

It looks like the problem is related to IO memory permissions.
I tried two different JDKs, openjdk 11, shipped with the RpiOS, and latest available librca.
I have tried to add `CAP_SYS_RAWIO` capability to my java executable(`sudo setcap CAP_SYS_RAWIO+ep java`), and that did not work either.

elmot, 2022-04-13:
stdout:
```txt
PLATFORMS: [1] "Pi4J Runtime Platforms" <com.pi4j.platform.impl.DefaultPlatforms>
└─PLATFORM: "RaspberryPi Platform" {raspberrypi} <com.pi4j.plugin.raspberrypi.platform.RaspberryPiPlatform> {Pi4J Platform for the RaspberryPi series of products.}
PLATFORM: "RaspberryPi Platform" {raspberrypi} <com.pi4j.plugin.raspberrypi.platform.RaspberryPiPlatform> {Pi4J Platform for the RaspberryPi series of products.}
PROVIDERS: [12] "I/O Providers" <com.pi4j.provider.impl.DefaultProviders>
├─SERIAL: [2] <com.pi4j.io.serial.SerialProvider>
│ ├─PROVIDER: "PiGpio Serial Provider" {pigpio-serial} <com.pi4j.plugin.pigpio.provider.serial.PiGpioSerialProviderImpl> {com.pi4j.plugin.pigpio.provider.serial.PiGpioSerialProviderImpl}
│ └─PROVIDER: "RaspberryPi Serial Provider" {raspberrypi-serial} <com.pi4j.plugin.raspberrypi.provider.serial.RpiSerialProviderImpl> {com.pi4j.plugin.raspberrypi.provider.serial.RpiSerialProviderImpl}
├─DIGITAL_INPUT: [2] <com.pi4j.io.gpio.digital.DigitalInputProvider>
│ ├─PROVIDER: "RaspberryPi Digital Input (GPIO) Provider" {raspberrypi-digital-input} <com.pi4j.plugin.raspberrypi.provider.gpio.digital.RpiDigitalInputProviderImpl> {com.pi4j.plugin.raspberrypi.provider.gpio.digital.RpiDigitalInputProviderImpl}
│ └─PROVIDER: "PiGpio Digital Input (GPIO) Provider" {pigpio-digital-input} <com.pi4j.plugin.pigpio.provider.gpio.digital.PiGpioDigitalInputProviderImpl> {com.pi4j.plugin.pigpio.provider.gpio.digital.PiGpioDigitalInputProviderImpl}
├─PWM: [2] <com.pi4j.io.pwm.PwmProvider>
│ ├─PROVIDER: "PiGpio PWM Provider" {pigpio-pwm} <com.pi4j.plugin.pigpio.provider.pwm.PiGpioPwmProviderImpl> {com.pi4j.plugin.pigpio.provider.pwm.PiGpioPwmProviderImpl}
│ └─PROVIDER: "RaspberryPi PWM Provider" {raspberrypi-pwm} <com.pi4j.plugin.raspberrypi.provider.pwm.RpiPwmProviderImpl> {com.pi4j.plugin.raspberrypi.provider.pwm.RpiPwmProviderImpl}
├─ANALOG_INPUT: [0] <com.pi4j.io.gpio.analog.AnalogInputProvider>
├─I2C: [2] <com.pi4j.io.i2c.I2CProvider>
│ ├─PROVIDER: "RaspberryPi I2C Provider" {raspberrypi-i2c} <com.pi4j.plugin.raspberrypi.provider.i2c.RpiI2CProviderImpl> {com.pi4j.plugin.raspberrypi.provider.i2c.RpiI2CProviderImpl}
│ └─PROVIDER: "PiGpio I2C Provider" {pigpio-i2c} <com.pi4j.plugin.pigpio.provider.i2c.PiGpioI2CProviderImpl> {com.pi4j.plugin.pigpio.provider.i2c.PiGpioI2CProviderImpl}
├─DIGITAL_OUTPUT: [2] <com.pi4j.io.gpio.digital.DigitalOutputProvider>
│ ├─PROVIDER: "PiGpio Digital Output (GPIO) Provider" {pigpio-digital-output} <com.pi4j.plugin.pigpio.provider.gpio.digital.PiGpioDigitalOutputProviderImpl> {com.pi4j.plugin.pigpio.provider.gpio.digital.PiGpioDigitalOutputProviderImpl}
│ └─PROVIDER: "RaspberryPi Digital Output (GPIO) Provider" {raspberrypi-digital-output} <com.pi4j.plugin.raspberrypi.provider.gpio.digital.RpiDigitalOutputProviderImpl> {com.pi4j.plugin.raspberrypi.provider.gpio.digital.RpiDigitalOutputProviderImpl}
├─ANALOG_OUTPUT: [0] <com.pi4j.io.gpio.analog.AnalogOutputProvider>
└─SPI: [2] <com.pi4j.io.spi.SpiProvider>
  ├─PROVIDER: "RaspberryPi SPI Provider" {raspberrypi-spi} <com.pi4j.plugin.raspberrypi.provider.spi.RpiSpiProviderImpl> {com.pi4j.plugin.raspberrypi.provider.spi.RpiSpiProviderImpl}
  └─PROVIDER: "PiGpio SPI Provider" {pigpio-spi} <com.pi4j.plugin.pigpio.provider.spi.PiGpioSpiProviderImpl> {com.pi4j.plugin.pigpio.provider.spi.PiGpioSpiProviderImpl}```
```
stderr:
```txt
[main] INFO com.pi4j.util.Console - ************************************************************
[main] INFO com.pi4j.util.Console - ************************************************************
[main] INFO com.pi4j.util.Console -
[main] INFO com.pi4j.util.Console -                   <-- The Pi4J Project -->
[main] INFO com.pi4j.util.Console -                    Minimal Example project
[main] INFO com.pi4j.util.Console -
[main] INFO com.pi4j.util.Console - ************************************************************
[main] INFO com.pi4j.util.Console - ************************************************************
[main] INFO com.pi4j.util.Console -
[main] INFO com.pi4j.Pi4J - New auto context
[main] INFO com.pi4j.Pi4J - New context builder
[main] INFO com.pi4j.platform.impl.DefaultRuntimePlatforms - adding platform to managed platform map [id=raspberrypi; name=RaspberryPi Platform; priority=5; class=com.pi4j.plugin.raspberrypi.platform.RaspberryPiPlatform]
[main] INFO com.pi4j.util.Console - --------------------
[main] INFO com.pi4j.util.Console - |  Pi4J PLATFORMS  |
[main] INFO com.pi4j.util.Console - --------------------
[main] INFO com.pi4j.util.Console -
[main] INFO com.pi4j.util.Console -
[main] INFO com.pi4j.util.Console - ---------------------------
[main] INFO com.pi4j.util.Console - |  Pi4J DEFAULT PLATFORM  |
[main] INFO com.pi4j.util.Console - ---------------------------
[main] INFO com.pi4j.util.Console -
[main] INFO com.pi4j.util.Console -
[main] INFO com.pi4j.util.Console - --------------------
[main] INFO com.pi4j.util.Console - |  Pi4J PROVIDERS  |
[main] INFO com.pi4j.util.Console - --------------------
[main] INFO com.pi4j.util.Console -
[main] INFO com.pi4j.util.Console -
[main] WARN com.pi4j.library.pigpio.impl.PiGpioNativeImpl - PIGPIO ERROR: PI_INIT_FAILED; pigpio initialisation failed
Exception in thread "main" java.lang.reflect.UndeclaredThrowableException
        at jdk.proxy1/com.sun.proxy.jdk.proxy1.$Proxy2.create(Unknown Source)
        at com.pi4j@2.1.1/com.pi4j.context.Context.create(Context.java:317)
[... cut ...]
        at com.pi4j.library.pigpio@2.1.1/com.pi4j.library.pigpio.impl.PiGpioNativeImpl.gpioInitialise(PiGpioNativeImpl.java:95)
        at com.pi4j.library.pigpio@2.1.1/com.pi4j.library.pigpio.PiGpio.initialize(PiGpio.java:146)
        at com.pi4j.plugin.pigpio@2.1.1/com.pi4j.plugin.pigpio.provider.gpio.digital.PiGpioDigitalOutputProviderImpl.create(PiGpioDigitalOutputProviderImpl.java:60)
        at com.pi4j.plugin.pigpio@2.1.1/com.pi4j.plugin.pigpio.provider.gpio.digital.PiGpioDigitalOutputProviderImpl.create(PiGpioDigitalOutputProviderImpl.java:41)
        ... 11 more
```

FDelporte, 2022-04-18:
Hi @elmot, this problem is not related to Rpi OS, but how Pi4J uses PiGpio "under the hood". For this library indeed sudo needs to be used to have full access to the GPIOs. This is by default and mentioned a few times on the documentation website, but indeed considered as an important improvement point for future versions.

If you need I2C, you can use the LinuxFS provider (https://pi4j.com/documentation/providers/linuxfs/) instead of PiGpio, but this doesn't provide basic digital input/output (yet), see https://github.com/Pi4J/pi4j-v2/issues/117 

As soon as we find time ourselves, or new contributors willing to assist us in this topic, we could push this project in a sudo-free direction...