Pi4J/pi4j #214: java.lang.ClassNotFoundException: com.pi4j.library.pigpio.PiGpio

razorne, 2022-04-23:
Hello, 

i am new to the library and i am trying to make a led blink. I have written a short code, copying from the tutorial page on official website.

```
public static void main(String[] args) throws InterruptedException {
        private int PIN_LED = 22;
        var pi4j = Pi4J.newAutoContext();
        var ledConfig = DigitalOutput.newConfigBuilder(pi4j)
                .id("led")
                .name("LED Flasher")
                .address(PIN_LED)
                .shutdown(DigitalState.LOW)
                .initial(DigitalState.LOW)
                .provider("pigpio-digital-output");

        var led = pi4j.create(ledConfig);

        led.low();
        Thread.sleep(1000);
        led.high();
        Thread.sleep(1000);
        pi4j.shutdown();
}
```

I am not using maven nor gradle to package the program (i am a maven user, but for this project, for reasons that are not relevant, i cannot use it) I run the program directly from the raspberry, using this sh:

`java - cp "/home/rasp/project35/lib/*:/home/rasp/project35/dist/*" com.project35.Main`

And this is the content of those folders. As you can see i am using the version of dependencies that are used in the official getting started page. 

[image]
[image]

I am using JDK 11 since i have installed the Raspberry Pi OS Full (32-bit), just like suggested by the getting started page.

Am i missing something here?

Thanks

savageautomate, 2022-05-02:
I think you may be missing one or more transitive dependencies.  The library JAR `pi4j-library-pigpio-2..0.jar` appears to be missing.  Also you must have the PiGpio library installed on your Rpi.  See: https://abyz.me.uk/rpi/pigpio/download.html

razorne, 2022-05-02:
Thank for your comment. Yes i was missing that dependency, by adding it, and executing my sh as sudo (otherwise i would have received another error) i was able to make my led blink

savageautomate, 2022-05-04:
Yes, `sudo` is required for using the PiGpio provider.  I'm working on Linux GPIO (filesystem) providers now that don't require `sudo` access.

closed 2022-05-04 by savageautomate

razorne, 2022-05-05:
Very nice! Thank you