Pi4J/pi4j #217: Fatal error when trying to run on ARMv6 (Raspberry Pi Zero W)

MrAndeos, 2022-05-02:
Hello!
So I'm building the latest version of the library on my Pi4:
```
git clone https://github.com/Pi4J/pi4j-v2
export JAVA_HOME="/usr/lib/jvm/jdk-16.0.1"
cd pi4j-v2/
mvn clean install -Pnative,cross-compile
```

It builds just fine and I'm able to use the generated JARs together with appropriate (aarch64) libpi4j-pigpio.so library, placed in a custom directory that's set like so: 
`
System.setProperty("pi4j.library.path", "/home/pi/test/lib/");
`
This all works on Pi4, together with the latest pigpio cloned from joan2937's github and compiled on Pi4.

But the problem occurs when I try to run my application on Pi Zero W.
I have this exact same setup. The pigpio native libs are compiled on the Pi Zero, and I'm using the "armhf" version of libpi4j-pigpio.so.
Java is the latest build from azul that supports ARMv6HF (11.0.15+10).
This is what I see in the console:
```
pi@zerow:~/test $ sudo java -jar pi4j-minimal-test-app.jar
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGILL (0x4) at pc=0xacf08b34, pid=1411, tid=1412
#
# JRE version: OpenJDK Runtime Environment Zulu11.56+19-CA (11.0.15+10) (build 11.0.15+10-LTS)
# Java VM: OpenJDK Client VM Zulu11.56+19-CA (11.0.15+10-LTS, mixed mode, serial gc, linux-arm)
# Problematic frame:
# C  [libpi4j-pigpio.so+0x2b34]
#
# No core dump will be written. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /home/pi/test/hs_err_pid1411.log
#
# If you would like to submit a bug report, please visit:
#   http://www.azul.com/support/
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
Aborted
```
And here is the error report as well as my Java code:
[hs_err_pid1411.log](https://github.com/Pi4J/pi4j-v2/files/8606516/hs_err_pid1411.log)
[pi4j-minimal-test-app.zip](https://github.com/Pi4J/pi4j-v2/files/8606518/pi4j-minimal-test-app.zip)

I will appreciate any help in getting this to work :)

taartspi, 2022-05-03:
I do not see your code using any slf4 logger class so I assume the pi4j-core wanted to log something and went down this path.

In my environment i have slf4j-simple-2.0.0-alpha0.jar  to satisfy the logger.   

Reading the following maybe you have an old level of slf4-simple and it still requires a logging implementation .

///////////////////////////////////////////////////////////////////////////
This warning, i.e. not an error, message is reported when no SLF4J providers could be found on the class path. Placing one (and only one) of slf4j-nop.jar slf4j-simple.jar, slf4j-log4j12.jar, slf4j-jdk14.jar or logback-classic.jar on the class path should solve the problem. Note that these providers must target slf4j-api 1.8 or later.
Note that slf4j-api versions 2.0.x and later use the [ServiceLoader](https://docs.oracle.com/javase/8/docs/api/java/util/ServiceLoader.html) mechanism. Backends such as logback 1.3 and later which target slf4j-api 2.x, do not ship with org.slf4j.impl.StaticLoggerBinder. If you place a logging backend which targets slf4j-api 2.0.x, you need slf4j-api-2.x.jar on the classpath. See also [relevant faq](https://www.slf4j.org/faq.html#changesInVersion18) entry.
SINCE 1.6.0 In the absence of a provider, SLF4J will default to a no-operation (NOP) logger provider.
/////////////////////////////////////////

PS are you compiling with java 16 and running java 11.   I think the V2 project assume java 11 for both.

MrAndeos, 2022-05-03:
I compiled Pi4J with Java 11 and added slf4j-simple-1.7.36.jar to my classpath, it works on Pi4 as before and I can see that it uses slf4j-simple for logging, but here on the Pi Zero it's still exactly the same error :(
And before I was using slf4j-api-1.7.32.jar, so it didn't require the implementation to work.

FDelporte, 2022-05-03:
I see you mentioned `aarch64` regarding the build. This means you are creating a 64-bit version of the application, but the Pi Zero W (1) only supports 32-bit version. I think this could be related to this problem.

MrAndeos, 2022-05-03:
I'm building Pi4J on aarch64 system because it specifically states to not try to build it on armv6: https://github.com/Pi4J/pi4j-v2/blob/9763aa3ecbd8564890b1af7e9e367f74289bad39/libraries/pi4j-library-pigpio/src/main/native/build.sh#L56

Then I want to use Pi4J with pigpio on ARMv6, so I take the generated JARs and the library from ``pi4j-v2/libraries/pi4j-library-pigpio/target/lib/armhf`` to the Pi Zero and try to run it there.
In the "lib" directory there are two libraries generated, one in aarch64 directory and one in armhf directory. I use the aarch64 one on Pi4 and the armhf one on Pi Zero.

MrAndeos, 2022-05-03:
I downloaded the latest release ([pi4j-2.1.1.zip](https://github.com/Pi4J/download/raw/main/pi4j-2.1.1.zip)), used the JARs from there in my classpath, extracted libpi4j-pigpio.so from pi4j-library-pigpio.jar/lib/armhf and put it in my explicitly defined path (``/home/pi/test/lib/``).
This now works on Pi Zero. So looks like something that was recently changed causes the native library to breake on ARMv6.

savageautomate, 2022-05-04:
The Pi4J deployed snapshot and release builds are getting cross-compiled using GitHub CI workflows on Intel/AMD x64 machines.  So it sounds like something about the ARM64 cross-compiled build of the ARMHF (ARMv6) native libraries are not getting compiled or linked properly?

MrAndeos, 2022-05-04:
Then this is almost certainly the case here. I remember cross-compiling Pi4J on x86-64 machine about half a year ago and there were no issues running the compiled libraries on Pi Zero.

hwvenancio, 2022-12-14:
Was this issue fixed? I'm getting the following error when testing pi4j 2.x on my Raspberry Pi Zero W (ARMv6):
```
...
Caused by: com.pi4j.library.pigpio.PiGpioException: PIGPIO ERROR: PI_INIT_FAILED; pigpio initialisation failed
...
```
When using pi4j 1.4 it works