Pi4J/pi4j #237: Throwing error while fatJaring using gradle

Reqeique, 2022-08-12:
I was currently experimenting with pi4j but when I start to fat Jar it, it kept throwing exception 
my first attempt was to utilize shadowJar, which wasn't able to jar all the necessary libraries resulting `lib wasn't found exception`
so I put all the necessary `jar` files manually but gave `Method [create([class java.lang.Integer])] could not be resolved in provider [pigpio-digital-output(com.pi4j.plugin.pigpio.provider.gpio.digital.PiGpioDigitalOutputProviderImpl)]; Overridden instances of 'create()' are not supported unless defined as default methods in the provider interface`
following that I tried my third attempts which was to utilize built in jaring method using the following code 
```kt
task("fatJar", type = Jar::class){

    duplicatesStrategy = DuplicatesStrategy.EXCLUDE
    manifest {
        attributes(mapOf("Main-Class" to "myClass"))
    }
    from(configurations.runtimeClasspath.get().map{ if
            (it.isDirectory) it else zipTree(it) })
    with(tasks["jar"] as CopySpec)
}
```
but no luck!
Note: the code i used was from https://pi4j.com/documentation/io-examples/digital-output/ which was then converted to kotlin

taartspi, 2022-08-14:
One thing, use this as the example of code. I believe once your present code creates and runs through the fat, it will not change the LED.  https://github.com/Pi4J/pi4j-example-minimal

I created a clone of this example successfully ran it.

FAT    used info from https://pi4j.com/documentation/building/fat-jar/ and example  https://github.com/Pi4J/pi4j-example-fatjar to copy the needed edits to the pom file.   

I edited the pom file as follows .  Was able to: mvn clean package, cd into target/distribution and sudo java -jar pi4j-example-fatjar.jar 

<?xml version="1.0" encoding="UTF-8"?>
<project xmlns="http://maven.apache.org/POM/4.0.0"
    xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance"
    xsi:schemaLocation="http://maven.apache.org/POM/4.0.0 http://maven.apache.org/xsd/maven-4.0.0.xsd">
    <modelVersion>4.0.0</modelVersion>

    <groupId>com.pi4j</groupId>
    <artifactId>pi4j-example-minimal</artifactId>
    <name>Pi4J :: MINIMAL EXAMPLE :: Sample minimal project</name>
    <description>Pi4J Sample Code</description>
    <version>0.0.1</version>

    
    <properties>
        
        <project.build.sourceEncoding>UTF-8</project.build.sourceEncoding>

    <main.class>com.pi4j.example.MinimalExample</main.class>
        
        <java.version>11</java.version>
        <maven.compiler.source>${java.version}</maven.compiler.source>
        <maven.compiler.target>${java.version}</maven.compiler.target>

        
        <slf4j.version>1.7.32</slf4j.version>
        <pi4j.version>2.2.0-SNAPSHOT</pi4j.version>

        
        <exec-maven-plugin.version>1.6.0</exec-maven-plugin.version>
        <maven-compiler-plugin.version>3.8.1</maven-compiler-plugin.version>

 <maven-shade-plugin.version>3.2.4</maven-shade-plugin.version>
    </properties>

    
    <repositories>
        <repository>
            <id>oss-snapshots-repo</id>
            <name>Sonatype OSS Maven Repository</name>
            <url>https://oss.sonatype.org/content/groups/public</url>
            <releases>
                <enabled>false</enabled>
            </releases>
            <snapshots>
                <enabled>true</enabled>
            </snapshots>
        </repository>

        <repository>
            <id>oss-snapshots-M2-repo</id>
            <name>Sonatype OSS Maven2 Repository</name>
            <url>https://mvnrepository.com/artifact/org.apache.felix/maven-bundle-plugin</url>
        </repository>
        <repository>
            <id>sonatype-oss-staging</id>
            <name>Sonatype Open Source Maven Repository</name>
            <url>https://oss.sonatype.org/service/local/staging/deploy/maven2</url>
        </repository>

    </repositories>

    <dependencies>
[... 140 lines cut ...]
                <groupId>org.apache.maven.plugins</groupId>
                <artifactId>maven-dependency-plugin</artifactId>
                <executions>
                    <execution>
                        <id>copy-dependencies</id>
                        <phase>process-sources</phase>
                        <goals>
                            <goal>copy-dependencies</goal>
                        </goals>
                        <configuration>
                            <outputDirectory>${project.build.directory}/distribution</outputDirectory>
                            <includeScope>runtime</includeScope>
                            <excludeTransitive>false</excludeTransitive>
                            <overWriteReleases>false</overWriteReleases>
                            <overWriteSnapshots>true</overWriteSnapshots>
                            <overWriteIfNewer>true</overWriteIfNewer>
                        </configuration>
                    </execution>
                </executions>
            </plugin>
        </plugins>
    </build>
</project>

hope this help.  Tom

taartspi, 2022-08-14:
[pom.txt](https://github.com/Pi4J/pi4j-v2/files/9333640/pom.txt)

taartspi, 2022-08-20:
I could see you prefer gradle to build.  I used the example testcase from above and added gradle to it.  I see an error. I normally use maven so all this is a matter of digging and I have not found an answer.   I do see that running under gradle build there are fewer providers  loaded.  

Maven these output providers.
├─DIGITAL_OUTPUT: [3] <com.pi4j.io.gpio.digital.DigitalOutputProvider> 
│ ├─PROVIDER: "PiGpio Digital Output (GPIO) Provider" {pigpio-digital-output} <com.pi4j.plugin.pigpio.provider.gpio.digital.PiGpioDigitalOutputProviderImpl> {com.pi4j.plugin.pigpio.provider.gpio.digital.PiGpioDigitalOutputProviderImpl} 
│ ├─PROVIDER: "LinuxFS Digital Output (GPIO) Provider" {linuxfs-digital-output} <com.pi4j.plugin.linuxfs.provider.gpio.digital.LinuxFsDigitalOutputProviderImpl> {com.pi4j.plugin.linuxfs.provider.gpio.digital.LinuxFsDigitalOutputProviderImpl} 
│ └─PROVIDER: "RaspberryPi Digital Output (GPIO) Provider" {raspberrypi-digital-output} <com.pi4j.plugin.raspberrypi.provider.gpio.digital.RpiDigitalOutputProviderImpl> {com.pi4j.plugin.raspberrypi.provider.gpio.digital.RpiDigitalOutputProviderImpl} 
├─I

Using gradle only 
└─PROVIDER: "RaspberryPi Digital Output (GPIO) Provider" {raspberrypi-digital-output} <com.pi4j.plugin.raspberrypi.provider.gpio.digital.RpiDigitalOutputProviderImpl> {com.pi4j.plugin.raspberrypi.provider.gpio.digital.RpiDigitalOutputProviderImpl} 

My error is
 Exception in thread "main" com.pi4j.provider.exception.ProviderNotFoundException: Pi4J provider [pigpio-digital-output] c

Was this your initial error when you said a 'lib' was not found ?

Reqeique, 2022-08-20:
@taartspi that was absolutely the error but could be resolved easily

RLEO94, 2022-09-27:
Hi did you solve this problem? 

I got the same exception. If I run my code on my computer in Intellij IDEA with the Embedded Linux JVM plugin, then the code works well (GPIO Out turns HIGH and Input is detected, ecc..).

If I create a fatjar with gradle, copy the file to my RPI4 and execute it with java -jar, then I got the same exception (Provider not found).
On the webside is described how to create a fat jar with Maven:  https://pi4j.com/documentation/building/fat-jar/.  
So I try now to use this plugin:  https://github.com/dkorotych/gradle-maven-exec-plugin 
with the goal to execute the described maven commands/plugins through gradle, but I'm still struggling

Reqeique, 2022-09-27:
@RLEO94 for `Provider not found` you could easily tweak it by modifying the fatjar to include the the library that wasn't found manually by using tools like win-rar or creating custom jaRing method as above to include all dependencies as above
overall can't guarantee wouldn't throw `Method [create([class java.lang.Integer])] could not be resolved in provider`

RLEO94, 2022-09-28:
Hi
Thanks for your help. I will give it a try :)

bennymeg, 2024-06-06:
This is worth including in the docs alongside the maven Uber Jar document.
More information can be found in _shadow_: [merging-service-descriptor-files](https://imperceptiblethoughts.com/shadow/configuration/merging/#merging-service-descriptor-files).