Pi4J/pi4j #240: Spi flags overwriiten

taartspi, 2022-09-20:
Sometime back I made a fix in the PiGpioSpi and commented a user might supply the flags attribute and that be overwritten.  I said I would look at this and the other pigpio providers.

PiGpioSpi is the only provider that overwrites user data.   I see two ways to fix this.

1  At present the PiGpioSpi does not know if the flags contents was user supplied or the default value. Same applies to the mode and bus.
If it was known the code could validate the user hadn’t set bus or mode to one value in flags, and used a different values with the individual attributes.  If the user confused things we could throw an IOException as done with other violations.   But this requires changes in DefaultSpiConfig to track when these are user supplied and changes in PiGpioSpi to validate.  The validation rules will be a cumbersome reading when you consider the combination of user /default supplied.

2. Make mode and bus required, IOException if not user supplied.  The validation will be compare mode and bus against the flag IF flag was user supplied, IOException for fails.   I think the advantages to this is mode and bus value are made very obvious to the coder,  and validation is simpler.

Your thoughts  
@eitch @FDelporte

taartspi, 2022-10-16:
I created a pull request as an implementation of the above question    #248

eitch, 2022-10-17:
I merged your PR. We will release a new version soon

taartspi, 2023-02-01:
Code change merged

closed 2023-02-01 by taartspi