Pi4J/pi4j #244: Multiple SPI throws IOAlreadyExistsException

mvanassche, 2022-09-22:
When creating multiple spi, an exception IOAlreadyExistsException is thrown.

The config id is not transferred to the instance object.

Unlike other IO, SpiBase inherits from IOBase, which does not transfer the id/name to the instance (GpioBase and I2CBase do transfer).

I suppose there is simply 
```
        this.name = config.name();
        this.id = config.id();
        this.description = config.description();
```
missing in the constructor of SpiBase?

mvanassche, 2022-09-23:
```
diff --git a/pi4j-core/src/main/java/com/pi4j/io/spi/SpiBase.java b/pi4j-core/src/main/java/com/pi4j/io/spi/SpiBase.java
index 80f807e..751bc80 100644
--- a/pi4j-core/src/main/java/com/pi4j/io/spi/SpiBase.java
+++ b/pi4j-core/src/main/java/com/pi4j/io/spi/SpiBase.java
@@ -48,6 +48,9 @@ public abstract class SpiBase extends IOBase<Spi, SpiConfig, SpiProvider> implem
      */
     public SpiBase(SpiProvider provider, SpiConfig config) {
         super(provider, config);
+        this.name = config.name();
+        this.id = config.id();
+        this.description = config.description();
     }
 
     /** {@inheritDoc} */
```

I tested with that diff with my example with two SPI devices on the same bus, different channel, and so far, it works.

FDelporte, 2022-09-23:
Thanks @mvanassche for reporting and proposing a fix!!!
I created a PR for this https://github.com/Pi4J/pi4j-v2/pull/245.