Pi4J/pi4j #249: No Example for using Pi as I2C Slave device.

Monroe07, 2022-10-21:
I'm hoping I 'm just missing the obvious, but how does one go about making the Pi a slave on the i2C bus? The only I2C examples I've found are for when the Pi is a Master and connecting to another device that is the slave.

taartspi, 2022-10-21:
O believe the pi is only a master.  The bcm2711 pi4 specs
ARM Peripherals
Table 24. I2C Address Map
Chapter 3. BSC
3.1. Overview
The Broadcom Serial Control (BSC) controller is a master, fast-mode (400Kb/s) BSC controller. The Broadcom Serial Control bus is a proprietary bus compliant with the Philips® I2C bus/interface version 2.1 January 2000.
• I2C single master only operation (supports clock stretching wait states)

eitch, 2022-10-28:
AFAIK what @taartspi says is correct, so closing this issue.

closed 2022-10-28 by eitch