Pi4J/pi4j #257: IOAlreadyExistsException when reopening Serial-connection

TorHahl, 2022-12-12:
Hello.
I've got a problem when re-opening a serial connection using the class com.pi4j.io.serial.Serial.
Basically I want to write an interface class to receive data over an rs485-to-usb converter. And when the usb device gets disconnected and/or reconnected the software interface should close and reopen the serial connection. (Possibly the usb device could also change meanwhile - e.g. between ttyUSB0 and ttyUSB1, even when using the same usb slot.)
So the serial interface works fine, without problems, after application start. But whenever the interface gets closed and reopened the Context.create-function delivers the following IOAlreadyExistsException:

Stacktrace:
com.pi4j.io.exception.IOAlreadyExistsException: IO instance [com.pi4j.plugin.pigpio.provider.serial.PiGpioSerial] already exists in the Pi4J runtime context; unable to create a new instance using this reserved id.

I closes the serial interface by using the Serial.close() function. I've also tried to use additionally the functions Serial.shutdown() and/or Context.getProvider().shutdown() and/or Context.getProvider().initialize(). But nothing worked for me. So the only way to change the serial device is now for me to restart the application.
I guess therefore that this is a bug and using the Serial.close() function doesn't unregister the used serial provider from the PI4J-Context. So I hope that someone here could verify this problem or has any idea what went wrong.
Alternatively maybe someone has an idea, how I could change the serial device on the fly - without having to close the serial connection.

My specifications:
- Raspberry 4B 4GB
- PI4J-Version: 2.2.1
- OS: Raspberry OS Lite (Debian Bullseye 32-bit)
- Pigpio-Version: 1.79-1+rpt1

One more note:
Using a raspberry-hat with a rs485-interface is unfortunately not really an option. Because some of the necessary I/O-Pins are still in use for other purposes.

FDelporte, 2022-12-12:
Can you give the full stacktrace of this error? I remember we had some missing cleanup in another provider that was fixed recently, but can't immediately find it back.

I checked the code and calling close() on a serial connection should close it...
https://github.com/Pi4J/pi4j-v2/blob/develop/plugins/pi4j-plugin-pigpio/src/main/java/com/pi4j/plugin/pigpio/provider/serial/PiGpioSerial.java#L96

TorHahl, 2022-12-12:
Of course, no problem.

Stacktrace:
com.pi4j.io.exception.IOAlreadyExistsException: IO instance [com.pi4j.plugin.pigpio.provider.serial.PiGpioSerial] already exists in the Pi4J runtime context; unable to create a new instance using this reserved id.
at com.pi4j@2.2.1/com.pi4j.registry.impl.DefaultRuntimeRegistry.add(DefaultRuntimeRegistry.java:80)
at com.pi4j@2.2.1/com.pi4j.provider.impl.ProviderProxyHandler.invoke(ProviderProxyHandler.java:106)
at com.sun.proxy.$Proxy3.create(Unknown Source)
at com.pi4j@2.2.1/com.pi4j.context.Context.create(Context.java:317)
at com.pi4j@2.2.1/com.pi4j.internal.IOCreator.create(IOCreator.java:58)
at com.pi4j@2.2.1/com.pi4j.internal.IOCreator.create(IOCreator.java:146)
at ... (my own connect function)

TorHahl, 2022-12-14:
A short update:
To avoid this problem I followed at least the tip in the neighboring issue thread  [Can't Create 2 serial connections](https://github.com/Pi4J/pi4j-v2/issues/207) and use jSerialComm.
https://fazecast.github.io/jSerialComm/

Especially the handling of usb-to-serial devices is very easy with jSerialComm. Now I don't have to check all usb device connections anymore for myself. Many thanks then again at this point for the tip!

commit 102a9fd3 (2024-04-15): fix(io): IOBase constructor do not update the fields id, name & description

The classes extending IOBase were already updating those fields (except SerialBase).
SerialBase was the only class missing this change.

These changes allow the user to use multiple Serial interface at the same time. It
was impossible before, as Pi4J would throws IOAlreadyExistsException.

This commit "moves" the fixes for the similar issues in the SPI interface.

Ref: #244, #257
Closes: #257
closed 2024-04-15 by eitch