Pi4J/pi4j #265: GPIO digital input doesn't change state as button is pressed/realised

zaze06, 2023-02-08:
I'm trying to update data on a screen when buttons are pressed, but the buttons whont be registered as pressed according to Pi4j
button initialization:
```java
button1 = pi4j.create(DigitalInput.newConfigBuilder(pi4j)
                .id("button")
                .name("button 1")
                .address(GPIOPin.GPIO27)
                .pull(PullResistance.PULL_DOWN)
                .debounce(3000L)
                .provider("linuxfs-digital-input")
        );

        button2 = pi4j.create(DigitalInput.newConfigBuilder(pi4j)
                .id("button1")
                .name("button 2")
                .address(GPIOPin.GPIO22)
                .pull(PullResistance.PULL_DOWN)
                .debounce(3000L)
                .provider("linuxfs-digital-input")
        );

        button3 = pi4j.create(DigitalInput.newConfigBuilder(pi4j)
                .id("button2")
                .name("button 3")
                .address(GPIOPin.GPIO23)
                .pull(PullResistance.PULL_DOWN)
                .debounce(3000L)
                .provider("linuxfs-digital-input")
        );

        home = pi4j.create(DigitalInput.newConfigBuilder(pi4j)
                .id("button3")
                .name("button home")
                .address(GPIOPin.GPIO17)
                .pull(PullResistance.PULL_DOWN)
                .debounce(3000L)
                .provider("linuxfs-digital-input")
        );
```

I read the state of the buttons using the `DigitalInput#isHigh()` and `DigitalInput#isLow()` methods in a `javax.swing.Timer`

The timer is defined as a field and the `java.awt.event.ActionListener` is provided on the definition of the Timer.
The timer is started after the buttons have ben defined.

Build.gradle dependencies include `com.pi4j:pi4j-plugin-linuxfs:2.3.0`

Physical setup:
[image]

links to the code parts:
[button setup](https://github.com/zaze06/menue/blob/master/src/main/java/me/alien/menu/Main.java#L274)
[Timer forn the button reads](https://github.com/zaze06/menue/blob/master/src/main/java/me/alien/menu/Main.java#L178)
[Build.gradle dependencies](https://github.com/zaze06/menue/blob/master/build.gradle#L21)

If more information is needed I will be glad to try to provide it.

taartspi, 2023-02-09:
The wiring detail appears to be some additional board and not the Pi 40 pin connector.  Not knowing how this maps to the pi 40 I can’t  validate which pin numbering scheme you used. But the pi4j code uses the BCM numbering for the address. 
Also I use the linuxfs for i2c work but not gpio. So I can’t say if you encountered a bug so also try your code using the pigpio provider.

FDelporte, 2023-02-09:
Indeed as @taartspi pointed out digital input and output with LinuxFS are under construction...
https://pi4j.com/documentation/providers/linuxfs/

Please try adding the following dependency

```
   <dependency>
        <groupId>com.pi4j</groupId>
        <artifactId>pi4j-plugin-pigpio</artifactId>
        <version>${pi4j.version}</version>
    </dependency>
```

And configure the button pins with the `pigpio-digital-input` provider

```
button1 = pi4j.create(DigitalInput.newConfigBuilder(pi4j)
                .id("button")
                .name("button 1")
                .address(GPIOPin.GPIO27)
                .pull(PullResistance.PULL_DOWN)
                .debounce(3000L)
                .provider("pigpio-digital-input")
        );
```