Pi4J/pi4j #268: Software PWM fails to work on Pi 1

hackerjimbo, 2023-03-03:
Yes, I still have an original Pi 1.

I'm porting the last piece of my Pi4J v1 software to the latest v2.3.0. It all works _apart_ from the software PWM.

I set it up as follows:

```
final PwmConfig config = PwmConfigBuilder.newInstance (ctx)
                    .id ("BCM17")
                    .name ("PWM")
                    .address (17)
                    .pwmType(PwmType.SOFTWARE)
                    .initial (50)
                    .provider ("pigpio-pwm")
                    .shutdown (0)
                    .build ();
            
pwm = ctx.create (config);
            
pwm.on ();
```

And then move from the initial 50% duty cycle. I can print out the values of pwm.actualFrequency(), pwm.address(), pwm.dutyCycle() and pwm.isOn () and they seem to change fine.

However, it appears that the output pin is just left floating!

Any ideas? Setting the id to BCM17 was something else I found in the issues, but that doesn't seem to have helped.

hackerjimbo, 2023-03-03:
Hardware PWM _isn't_ supported on that pin, hence the use of the software one. Perhaps there is an issue with the software PWM?

FDelporte, 2023-03-03:
ah sorry, indeed I missed the PWM type setting you are using ;-)
sorry, can not immediately think of a possible cause.
I guess this page also doesn't provide a solution?
https://pi4j.com/documentation/io-examples/pwm/

taartspi, 2023-03-03:
I believe Pi4j V1 used wiringPi for the native interface.  V2 uses Pigpio native interface.     Pigpio website  says it works with any version of the Pi.     I also verified your code work expectantly on a Pi4.
As you had this working with V1 we can assume your wiring is correct.
The Pi4j PWM software  is not aware of the model/rev you are using, just calls the same native code on all cases.

Is your Pi1 the 26 or 40 pin connector  ?

hackerjimbo, 2023-03-03:
It's the original 26 pin connector. The documentation page si where I got the original code from. I'm still quite surprised at the name having to be BCM17 (in my case). That was reported as a requirement some time ago.

hackerjimbo, 2023-03-03:
Well that took a long time, but I have a solution.

In the non-working code I called `pwm.on ()` and then `pwn.duty (percent)`. Turns out what _actually_ works is just calling `pcm.one (duty, 100)` (100 Hz works for me and it's only a Pi 1).

It would appear that `duty` on it's own doesn't turn it on properly.

FDelporte, 2023-03-03:
Good find @hackerjimbo! Is this a missing part in the documentation, or do you think an error in the implementation?

FDelporte, 2023-03-03:
and as always, thanks for the support @taartspi :-)

hackerjimbo, 2023-03-04:
I would say it's an error in the implementation. In the non-working code I set it to be on and then just changed the duty cycle. It had an initial duty cycle (from the configuration) of 50% so I think most users would expect the `on` to produce an output. Changing the duty cycle using `duty` should continue to work.

Oh, hang on! I've just releasised. The `on` method _isn't_ a request to turn it on but an inspector method to query the status. I should have done an `on (true)` to make it work.. So maybe it's my fault after all!