Pi4J/pi4j #274: pigpiod startup problem

gratefulfrog, 2023-04-17:
Hello,

I've been using this library for sever weeks on an RPI4B+ with no issues whatsoever. It is really a great tool!

I am now trying to run it on an RPI3B+.

When I setup the pigpiod to startup automatically with `sudo systemctl enable pigpiod` it does startup but starts continually blinking certain gpio pins. In particular, the ones I need for my project... GPIO 17, 22, 23, 24, 27

If I run  `sudo systemctl stop pigpiod` then restart, it works fine, the gpios stay at 0.

I am looking for at least a workaround, if possible, to be able to continue my project which requires fully automatic startup.

I am fully available to help debug this issue.

Thanks for any suggestion.
Cheers,
Bob

FDelporte, 2023-04-17:
Hi Pi4J doesn't need pigpiod, so it should not be started at all...
It uses pigpio which is included in the library itself.

gratefulfrog, 2023-04-17:
Hello,
Thank you for your reply!
I am not sure that I understand.
I have an RPI4 4Gbyte ram. Are yous aying I should not start the pigpiod at all?

I do not know what you mean by 'RPI4J' ?  Does J refer to java? I only use the python interface.

But my problem today is getting it to work on the RPI3B+.  Would you  have any thoughts about that?

FDelporte, 2023-04-17:
Ah, if you are you using Python and not a Java project with Pi4J, you are not using the correct GitHub project for your question...
Better to try via https://abyz.me.uk/rpi/pigpio/

closed 2023-04-17 by FDelporte