Pi4J/pi4j #283: Add a callback to the Serial class fire when its open

mbcoder, 2023-07-04:
I've just starting using the Pi4J libs and have to say its a great API with solid documentation which got me going really quickly.

I've used Pi4J in this example GPS data logger [blog](https://www.esri.com/arcgis-blog/products/sdk-java/developers/how-to-use-the-arcgis-maps-sdk-for-java-in-a-raspberry-pi-for-recording-gps-tracks/) and when I had opened a serial port I had the following code to wait until the port was open:

```
            while (!serial.isOpen()) {
                try {
                    Thread.sleep(250);
                } catch (InterruptedException e) {
                    throw new RuntimeException(e);
                }
            }
```

I was thinking a nice enhancement to the API would be to add a listener to the serial class which would fire once the serial port is ready to read from.  If this existed then the above code could be replaced with something like:

```
serial.addOpenedListener(() -> {//do something now the serial port is open});
```

If you think this is a worthwhile addition I'd be happy to make a PR for review.

FDelporte, 2023-07-05:
We had a similar idea in the past, see #118, with a reference to the implementation in V1.
Maybe we should indeed consider extending the serial implementation.

eitch, 2023-07-06:
The idea is great. Do you mind creating a PR? We gladly accept PRs.

DieterHolz, 2023-07-17:
Wouldn't that more or less mean, that your whole application is inside the lambda expression of 'addOpenedListener'?

How about doing the 'wait' inside Pi4J instead.

When calling 

`pi4j.create(Serial.newConfigBuilder(pi4j)...`

it does not return until 'Serial' is open and ready for being used.

If that kind of synchronous call is not acceptable then provide two methods: 'serial.open' will wait until serial is ready and 'serial.asynOpen(onFinishedHandler)' will do the opening asynchronously.