Pi4J/pi4j #289: Requesting Support for GPIO Pin Access on Konstakang AOSP for Raspberry Pi

BierDav, 2023-07-23:
I am currently developing an Android application that runs on the Konstakang AOSP for Raspberry Pi. One of the critical functionalities I need for my app is the ability to access the GPIO (General Purpose Input/Output) pins on the Raspberry Pi board directly from the Android app.

I am aware that accessing GPIO pins directly from an Android app might require root privileges and involve low-level hardware interactions. I'm a experienced full stack developer, but working with a raspberry pi is very new to me. So I don't have a clue how GPIO pins work behind the scenes, I seek guidance and insights from the community to understand if this use case is even feasible.

The problem I'm currently stuck with is that the .so files are not compatible with Android i guess:
`Cause: pi4j-library-pigpio-2.3.0 extracted from path C:\Users\David Bieregger\.gradle\caches\transforms-3\7a8e860f66fe14b2e2add8fde4eee6f6\transformed\pi4j-library-pigpio-2.3.0\aarch64\libpi4j-pigpio.so is not an ABI`

jipspaul, 2024-03-03:
Any news ? did you success doing it ?

taartspi, 2024-03-03:
Yes, those .so are specific to the Pi ARM architecture. I am not familiar with   Konstakang AOSP.  The Pi4j would let you use java to access the GPIOs.       In the pi4 the following site documents the code used to actually touch the hardware :  https://abyz.me.uk/rpi/pigpio/python.html   That web site documents support for C and python if they are usable in your development.                 As far as Pi5 that GPIO support is under development.