Pi4J/pi4j #307: Extend Digital Input and Output implementations in LinuxFS

FDelporte, 2023-10-24:
With version 2.4.0 of Pi4J, the LinuxFS plugin got extended to provide Digital Input and Output implementations. This is a first step, but not the "full package".

For instance, handling of PullUpDown Resistors is not implemented yet.

alex9849, 2023-11-20:
Hello Frank, :)

I just tested the LinuxFS implementations. The do work on a raspberry pi 3, but I get the following error on a raspberry pi 5:
```
java.io.IOException: Invalid argument
        at java.base/sun.nio.ch.FileDispatcherImpl.write0(Native Method) ~[na:na]
        at java.base/sun.nio.ch.FileDispatcherImpl.write(FileDispatcherImpl.java:62) ~[na:na]
        at java.base/sun.nio.ch.IOUtil.writeFromNativeBuffer(IOUtil.java:132) ~[na:na]
        at java.base/sun.nio.ch.IOUtil.write(IOUtil.java:97) ~[na:na]
        at java.base/sun.nio.ch.IOUtil.write(IOUtil.java:67) ~[na:na]
        at java.base/sun.nio.ch.FileChannelImpl.write(FileChannelImpl.java:288) ~[na:na]
        at java.base/java.nio.channels.Channels.writeFullyImpl(Channels.java:74) ~[na:na]
        at java.base/java.nio.channels.Channels.writeFully(Channels.java:96) ~[na:na]
        at java.base/java.nio.channels.Channels$1.write(Channels.java:171) ~[na:na]
        at java.base/java.nio.file.Files.write(Files.java:3517) ~[na:na]
        at java.base/java.nio.file.Files.writeString(Files.java:3734) ~[na:na]
        at java.base/java.nio.file.Files.writeString(Files.java:3674) ~[na:na]
        at com.pi4j.plugin.linuxfs.internal.LinuxGpio.export(LinuxGpio.java:94) ~[pi4j-plugin-linuxfs-2.4.0.jar!/:na]
        at com.pi4j.plugin.linuxfs.provider.gpio.digital.LinuxFsDigitalOutput.initialize(LinuxFsDigitalOutput.java:71) ~[pi4j-plugin-linuxfs-2.4.0.jar!/:na]
        at com.pi4j.plugin.linuxfs.provider.gpio.digital.LinuxFsDigitalOutput.initialize(LinuxFsDigitalOutput.java:46) ~[pi4j-plugin-linuxfs-2.4.0.jar!/:na]
        at com.pi4j.provider.impl.ProviderProxyHandler.invoke(ProviderProxyHandler.java:103) ~[pi4j-core-2.4.0.jar!/:na]
        at jdk.proxy2/jdk.proxy2.$Proxy102.create(Unknown Source) ~[na:na]
        at com.pi4j.context.Context.create(Context.java:325) ~[pi4j-core-2.4.0.jar!/:na]
        at com.pi4j.internal.IOCreator.create(IOCreator.java:58) ~[pi4j-core-2.4.0.jar!/:na]
        at com.pi4j.internal.IOCreator.create(IOCreator.java:96) ~[pi4j-core-2.4.0.jar!/:na]
        at net.alex9849.cocktailpi.utils.PinUtils.getBoardOutputPin(PinUtils.java:41) ~[classes!/:1.0-SNAPSHOT]
```
The file `/sys/class/gpio` does exist.
I initialized the context using: 
```
                ContextBuilder ctxBuilder = Pi4J.newContextBuilder();
                ctxBuilder.add(LinuxFsI2CProvider.newInstance());
                ctxBuilder.add(LinuxFsDigitalOutputProvider.newInstance());
                ctxBuilder.add(LinuxFsDigitalInputProvider.newInstance());
```

I would like to get pi4J running on a raspberry pi 5. Sadly pigpio got incompatible.

eitch, 2023-11-21:
Hi @alex9849 

If you don't have a /sys/class/gpio file, then this means that a kernel module is not loaded. This would be something to do with the Raspberry Pi OS. Further the LinuxFS plugin doesn't use pigpio, just files exported by the kernel.

Sadly i don't yet have a Rasberry Pi 5, so i can't test this out. Maybe you can check the Raspberry Pi forums for the kernel which needs to be loaded? Are you using the latest OS? Maybe you need to enable it in the raspi-config tool?

FDelporte, 2023-11-21:
Hi, I have a 5 but only did a very quick test with the minimal example. It looks like PiGpio doesn't work either - as expected based on some messages on tickets in the project, e.g.
https://github.com/joan2937/pigpio/issues/589

I will try more later, but agenda is still blocked the next days.

alex9849, 2023-11-21:
Yes. pigpio got incompatible sadly. But I found a few things out. 

My kernel it okay. I do have a /sys/class/gpio file. That's not the problem. I just wanted to point out that the error happens even tough that file exists. I tested the DigitalOutput integration with LinuxFS on a raspberry pi 3 and a raspberry pi 5. It works on the Pi3, but not on the Pi5.

The LinuxFS the I2C integration works on both platforms. Maybe the format of the gpio files changed? Is the format documented somewhere?

mores, 2025-08-11:
> The gpiod provider has input and output for the Pi5. Does this answer your requirement

Yes - that meets my needs.