Pi4J/pi4j #312: Unable to obtain data from DHT11

harlanhu, 2023-12-27:
I wrote some code, hoping to get data from DHT 11, after repeated tests, unfortunately I can't get enough 40 bits of data, I think it may be because the high or low level state of 40~100 us can not be accurately obtained, I tried to use a while loop to determine the current pin state, and run the Adafruit DHT.py script, when python runs and reads out the data, Java loops do not have a full output of the correct high and low level states.The time of a loop is about 6~10 microseconds, which is enough to get the current state of the pin.
Can you give me some possible solutions?Thanks.

taartspi, 2023-12-27:
https://github.com/taartspi/pi4j-example-devices/tree/master/src/main/java/com/pi4j/devices/dht22

I believe dht 11 and 22 operate the same.  My implementation may miss a few bits so i use the checksum to ensure the high order between bits were zero so unimportant.   As I can’t  view all your code maybe seeing all of my code will assist you

FDelporte, 2023-12-27:
The students who did the CrowPi implementations, also with a DHT, had a lot of issues because of the timing constraints of these devices...

This is their implementation:
https://github.com/Pi4J/pi4j-example-crowpi/blob/main/src/main/java/com/pi4j/crowpi/applications/HumiTempApp.java

As you can see in the comments of the component, they use a Linux driver:
https://github.com/Pi4J/pi4j-example-crowpi/blob/main/src/main/java/com/pi4j/crowpi/components/HumiTempComponent.java

This is implemented as part of Pi4J OS:
https://github.com/Pi4J/pi4j-os/blob/main/crowpi/resources/system/config.txt#L18