Pi4J/pi4j #317: Does Pi4J support the new PI 5?

ceesgroot, 2024-01-25:
Hi there. Just go my new PI 5 and I am wondering if PI4J supports it?

eitch, 2024-01-25:
Hi @ceesgroot
Pi4j does not yet work on the Pi 5. We are working on it, but it isn't so simple. Keep tabs on this repository for a new version, or subscribe to our social media outlets.

ceesgroot, 2024-01-25:
Hi @eitch
Thanks for your prompt answer. My BME280 sensor is working on I2C. The issue seems to be with my MAX7219 display on the SPI interface. I will check the repo now then.

eitch, 2024-01-25:
Hi @ceesgroot 
I just saw your message. I just did a test with the linuxfs library, which has gpio working, if i see it correctly. I wasn't able to test it properly, since i have nothing hooked up, but the code runs fine.

I'm happy to hear I2C is working. Is this with the linuxfs plugin?

SPI is a pigpio plugin, so that probably won't be working anytime soon, as this library hasn't had any updates in 3 years. Maybe we'll find a solution...

@alex9849 is helping us out here using libgpiod

ceesgroot, 2024-01-25:
hi @eitch 
I2C was indeed with the linuxfs (2.4.0) plugin. Good luck finding a solution. Please let me know if you want me to do some beta testing.

ceesgroot, 2024-01-28:
Hi @eitch
Small update. I tried to light a led on my PI 5 with linuxfs, but although the code ran fine, it didn't come up. I ordered and got a PI 4 in the mean time and the same config and code ran fine there. 
I will send the PI 5 back to the shop and move on with the PI 4 for now. Will check the blog every once in a while and probably order a new PI 5 when the library is working. 
Thanks for your support for now. BR, Cees

FDelporte, 2024-02-03:
See #321

closed 2024-02-03 by FDelporte