Pi4J/pi4j #321: New plugin to support Raspberry Pi 5

FDelporte, 2024-02-03:
The Raspberry Pi 5 has a new chip to interact with the GPIOs: RPI1. 
All info on https://www.raspberrypi.com/documentation/microcontrollers/rp1.html

Pi4J V2 uses LinuxFS and PiGpio to interact with the GPIOs, but these are not (fully) compatible with that RP1.

@alex9849 proposed via Slack on 20240114 the following solution: "I currently work on a plugin that allows Pi4J to use the gpiod library, which is currently the only way to access and control the gpio pins on the new Raspberry Pi 5. I've finished the DigitalOutputProvider and the native library. Everything compiles. (But I would need to modify the builder image for docker as some dependencies are missing). This is what I have written so far. It might work, but it is completely untested: https://github.com/alex9849/pi4j-v2/tree/feature/gpiod

piggy1130, 2024-02-05:
Hi Alex, thank you for your support for Raspberry Pi 5 with Java. I really want to control the GPIO with Java in Pi 5. Could you please let me know the step to install your code to let the GPIO control work in Raspberry Pi 5? I really appreciate any help that I could get. Thank you again!!

alex9849, 2024-02-05:
@piggy1130 Hi. Compiling the current version by yourself is a bit compilcated for now, as the official version of the builder that is automatically downloaded to build the project hasn't been updated yet and therefore can't compile the project. 

But I think the implementation should work very soon.

PR #322 (2024-02-06): Feature/gpiod
This pull requests enables Pi4j to take advantage of the GpioD library, that gets developed together with the linux kernel.

Contents:
- Added GpioD JNI bridge
- Added GpioD DigitalOutputProvider (Tested against a RaspberryPi 4 & 5)
- Added GpioD DigitalInputProvider (Tested against a RaspberryPi 5)
  - Added debouncing

I'm not sure if this will immediately compile in your CI/CD pipeline. It might miss some dependencies that are needed to build the JNI library for gpiod.  These are required packets: https://github.com/Pi4J/pi4j-v2/blob/501a50f1594a9bae2ea2cd794d6998db8d801faf/li [... cut]
closed 2024-02-06 by FDelporte