Pi4J/pi4j #335: java.io.IOException: Remote I/O error java.base/java.io.RandomAccessFile.writeBytes(Native Method)

taartspi, 2024-03-15:
Using linuxfs-i2c, dependent upon i2c  operations this exception can occur.   If the program initially uses  read or write, and later uses  readRegister or writeRegister there is no exception. However if the program  initially uses readRegister or writeRegister subsequent write or read may encounter this exception.                       As a temporary fix, when the program first create the i2c device, a read operation will be performed.     It is possible the attached i2c device always performs an  auto-increment in which case this additional read operation  will increment the device's internal pointer. As a program is not assured the state of any hardware, the program is already responsible to reset the chip or explicitly set such internal registers so this temporary change should show minimal effect.  When the proper fix is understood and implemented this 'extra' read will be removed.

eitch, 2024-03-15:
I currently see this fix as ok

taartspi, 2024-04-20:
Real fix found and implemented       Fix will build in snapHot 2.5.2

closed 2024-04-20 by taartspi