Pi4J/pi4j #342: [HELP] Raspberry PI 5

nomisRev, 2024-04-06:
Hey all,

Sorry to open an issue for this, but I am stuck trying to get a DHT22 to work on a Raspberry PI 5. Afaik as I understand, it should work in 2.5.1, but I am always running into a "provider" issue.

I am using Kotlin & Gradle, so all the examples on the website were not entirely clear to me but I'd be happy to contribute if we come to a conclusion. I.e. there is no section on fatJar with Gradle, which I achieved through the [Gradle shadow plugin](https://github.com/johnrengelman/shadow).

[My example PI5J](https://github.com/nomisRev/Pi5J/tree/main.). I am on the latest Pi5 OS, and am not seeing the `java.lang.UnsatisfiedLinkError` but instead I am getting:

```console
[main] INFO com.pi4j.Pi4J - New auto context
[main] INFO com.pi4j.Pi4J - New context builder
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Initializing Pi4J context/runtime...
[main] INFO com.pi4j.platform.impl.DefaultRuntimePlatforms - adding platform to managed platform map [id=raspberrypi; name=RaspberryPi Platform; priority=5; class=com.pi4j.plugin.raspberrypi.platform.RaspberryPiPlatform]
[main] INFO com.pi4j.runtime.impl.DefaultRuntime - Pi4J context/runtime successfully initialized.
<-- The Pi4J DHT22 Kotlin Project -->
Exception in thread "main" com.pi4j.provider.exception.ProviderNotFoundException: Pi4J provider [gpoid] could not be found.  Please include this 'provider' JAR in the classpath.
        at com.pi4j.provider.impl.DefaultRuntimeProviders.get(DefaultRuntimeProviders.java:227)
        at com.pi4j.provider.impl.DefaultProviders.get(DefaultProviders.java:147)
        at com.pi4j.provider.Providers.get(Providers.java:253)
        at com.pi4j.context.Context.create(Context.java:327)
        at com.pi4j.internal.IOCreator.create(IOCreator.java:58)
        at com.pi4j.internal.IOCreator.create(IOCreator.java:106)
        at io.github.nomisrev.AppKt.main(App.kt:22)
        at io.github.nomisrev.AppKt.main(App.kt)
[pi4j-shutdown] INFO com.pi4j.runtime.impl.DefaultRuntime - Shutting down Pi4J context/runtime...
[pi4j-shutdown] INFO com.pi4j.util.ExecutorPool - Shutting down executor pool Pi4J.RUNTIME
[pi4j-shutdown] INFO com.pi4j.runtime.impl.DefaultRuntime - Pi4J context/runtime successfully shutdown. Dispatching shutdown event.
```

Any ideas, or suggestions?

Thanks in advance!

nomisRev, 2024-04-07:
Hey @taartspi,

Yes, I replaced the `pigpio` with `gpiod`.
Here is my dependency list:

```kotlin
dependencies {
  api("org.slf4j:slf4j-api:2.0.12")
  api("org.slf4j:slf4j-simple:2.0.12")
  api("com.pi4j:pi4j-core:2.5.1")
  api("com.pi4j:pi4j-plugin-raspberrypi:2.5.1")
  api("com.pi4j:pi4j-library-gpiod:2.5.1")
  api("com.pi4j:pi4j-plugin-gpiod:2.5.1")
}
```

Here is my full [Gradle build file](https://github.com/nomisRev/Pi5J/blob/main/build.gradle.kts).

Thank you for reaching out so quickly.

taartspi, 2024-04-07:
are you still hitting the provider not found error ?                      If yes see https://github.com/Pi4J/pi4j-v2/discussions/126   Maybe its information can assist.